espressif/arduino-esp32 #1615: touchRead gives wrong value occasionally

glucee, 2018-07-09:
### Hardware:
Board:							ESP32 Huzzah
Core Installation/update date:		07/2018
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I use touchRead function in the loop to get the real-time value, I found the value is not always correct. When i don't touch it, it also occasionally gives wrong value. 

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
 Serial.begin(9600);
}

void loop() {
Serial.println(touchRead(15));
delay(100);
}
```

me-no-dev, 2018-07-10:
yes, I have some new code that I need to add and it will work. Soon :)

asetyde, 2018-07-10:
Soon .. when ? I also wait a bugfix :D thx

closed 2019-08-15 by stale[bot]

l0rb, 2020-09-11:
This is still an issue, should be opened again.

selman-nus, 2020-11-16:
Agreed, i see random values with 100ms delay.