espressif/arduino-esp32 #1616: SD library uses a lot of heap

themindfactory, 2018-07-09:
When you initialize this library it uses 26K of heap, which is a lot for users that wish only to open 1 file, and do not need high performance (buffers).

Would be nice to be able to call it with buffer and number of open files required to trim this down... in the past I have written FAT32 code that did not use more than about 2K for a few files, and 512 byte FAT table cache.....

Anyone able to do this?

Thanks!

PS. Is there a better place to put such a request??

lbernstone, 2018-07-09:
Try changing this to 1 from 5.  That parameter is the max number of open files.  If it significantly reduces your memory use, I'll make a pull request with that number as a define.
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L728

atanisoft, 2018-07-09:
@ibernstone, instead of a #define can it be exposed as a parameter on SD.begin() ?

Shdwdrgn, 2018-08-06:
Adding a "me too" for the suggestion to add this as a parameter on SD.begin().  My own code has a definable maximum number of simultaneous files that it will use, and it would be handy to simply pass that variable to the SD card setup.

lbernstone, 2018-08-06:
If someone would test, and tell me that it actually reduces the memory usage, I'd be happy to write this, but I don't want to add a control that only causes issues.

Shdwdrgn, 2018-08-10:
I used a simple sketch which issued SD.begin() and then printed ESP.getFreeHeap().  I hope that was appropriate?

I checked values from 1-5 and recompiled after each change.  This yielded the following:

1: 280192
2: 276056
3: 271920
4: 267784
5: 263644

The difference between each value was 4136 bytes, except for the last two where there was a difference of 4140 bytes.

Does that provide the info you were looking for, or is there something else I should check?

closed 2019-08-15 by stale[bot]