espressif/arduino-esp32 #1621: AsyncUDPServer example

jorgearroyop, 2018-07-11:
Board: Adafruit ESP32 Feather
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed:115200

### Description:
When I run the example AsyncUDPServer I receive successfully packages but every time a package is received it throws the message below and reboots. 
Any idea of how can I solve this?
Thanks!

### Debug Messages:
```
UDP Packet Type: Unicast, From: 192.168.1.103:62580, To: 192.168.1.132:1234, Length: 8, Data: test msg
assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/pbuf.c", line 694, function: pbuf_free
abort() was called at PC 0x400d67d3 on core 1

Backtrace: 0x40091788:0x3ffba950 0x4009198b:0x3ffba970 0x400d67d3:0x3ffba990 0x400f100b:0x3ffba9c0 0x400d2390:0x3ffba9e0 0x400d2899:0x3ffbaa00 0x400d2902:0x3ffbaa80

Rebooting...
 ```

me-no-dev, 2018-07-11:
done :) https://github.com/espressif/arduino-esp32/commit/ddfeae90d0f537805ed4d1ca87618f41f8032359

I have no clue why it happened now...

closed 2018-07-11 by me-no-dev

jorgearroyop, 2018-07-12:
It works great now, thank you guys!

I'm new on the Arduino platform but I just want to mention that I tried to update through the Boards Manager of the Arduino IDE and the error continued on the same line. Then I went and look into my AsyncUDP.cpp file and I saw that the new line was not there, so I added it manually and everything works fine now.

jorgearroyop, 2018-07-12:
Thanks for your reply @stickbreaker 
I had before this URL on the Additional Boards manager URLs:
https://dl.espressif.com/dl/package_esp32_dev_index.json