espressif/arduino-esp32 #1623: SPI.transfer(buffer, size) is missing

PaulStoffregen, 2018-07-11:
ESP32's SPI library is missing overloaded function SPI.transfer(buffer, size).

Here is Arduino's documentation.

https://www.arduino.cc/en/Reference/SPITransfer

Arduino sketches and libraries using this function will not compile for ESP32.  This is a very easy fix, just add this line in SPI.h

      void transfer(uint8_t * data, uint32_t size) { transferBytes(data, data, size); }

me-no-dev, 2018-07-11:
why not make a PR 😄

me-no-dev, 2018-07-11:
don't implement it in the header like that though

iafilius, 2018-12-02:
Hi, just a "me too" message (for libaries 1.0.0 and 1.0.1-rc1).  The quick fix by adding the oneliner to SPI.h worked for me.

PaulStoffregen, 2018-12-02:
While not strictly necessary to get Ethernet to work, you will see slightly improved performance on the W5500 chip if you also add SPI.transfer(const void *txBuffer, void *rxBuffer, size_t count) and define SPI_HAS_TRANSFER_BUF.

Here's where it's actually used by the Ethernet library:
https://github.com/arduino-libraries/Ethernet/blob/master/src/utility/w5100.cpp#L372

If SPI.h does not have this function, Ethernet will revert to using single byte transfers.

iafilius, 2018-12-02:
@me-no-dev  Thanks for accepting accepting the commit

@PaulStoffregen ,

you mean something like:

SPI.h:
class SPIClass
<>
public:
<>
void transfer(const void * txBuffer, const void * rxBuffer, uint32_t size);
<>

SPI.cpp:

// SPI_HAS_TRANSFER_BUF - is defined to signify that this library supports
// a version of transfer which allows you to pass in both TX and RX buffer
// pointers, either of which could be NULL
#define SPI_HAS_TRANSFER_BUF 1
SPI.transfer(const void *txBuffer, void *rxBuffer, size_t count) 
{
	transferBytes(txBuffer, rxBuffer, size);
}

If i see it correctly, no corresponding transferBytes() is available with the right types (void*), and no void* type parameter functions are available yet. What do you have in mind?
Regards

Popcron07, 2018-12-22:
Hi, I ran into a problem which I think might have something to do with your changes.
When I try to compile my code I get this error message:

> C:\...\packages\esp32\hardware\esp32\1.0.0\libraries\SPI\src/SPI.h:69:13: note: candidate: uint8_t SPIClass::transfer(uint8_t)

     uint8_t transfer(uint8_t data);

             ^

> C:\...\packages\esp32\hardware\esp32\1.0.0\libraries\SPI\src/SPI.h:69:13: note:   candidate expects 1 argument, 2 provided

@iafilius wrote it is strange to give the data twice and I think that is why I am not able to compile.
So please tell me, is this the cause for my error or do I have to search for something else?

Also I think this fellow might have the same problem and it really fits because his post is just a week old, too:
[https://stackoverflow.com/questions/53795074/arduino-exit-status-1](url)

Regards, Popcorn

PaulStoffregen, 2018-12-24:
Just to make sure we have a simple & clear description of how to reproduce this problem.

In Arduino 1.8.8, click Tools > Manage Libraries.  Search for "Ethernet" and make sure version 2.0.0 is installed.  Arduino 1.8.8 has Ethernet 2.0.0 by default, so this step is just to make sure you don't have an old version of Ethernet lingering in a location like Documents/Arduino/libraries.

Then open any of the examples, by clicking File > Examples > Ethernet.

Click Verify to reproduce the compile error.  Easy, right?

me-no-dev, 2018-12-31:
Since the original topic is resolved, I am closing this issue. Feel free to continue commenting though :)

closed 2018-12-31 by me-no-dev

ankurbashir, 2019-04-25:
Hi i am working on W5500 with ESP32 but got the error as mentioned below.

Initialize Ethernet with DHCP:
Failed to configure Ethernet using DHCP
Ethernet shield was not found.  Sorry, can't run without hardware. :(

What i am missing here?