espressif/arduino-esp32 #1624: Norton Virus Alert with GIT Download - Adafruit Feather 32 Board Install

Starkshub, 2018-07-11:
Recently, I purchased an Adafruit Huzzah 32 Feather and per their instructions was directed to the GIT site for the board install.

When I tried to download the Windows version of GIT through the site (https://git-scm.com/download/win) Norton Internet Virus Protection alerted me that there is a virus on the GIT download and removed the download from my system. 

The information for the block is as follows:

Norton is showing it as a malware virus released less then 1 week. - "Heur.AdvML.C". 

It shows it as a HIGH RISK virus.

I would like to use the Adafruit unit, but at this time can’t install because of the risk alert.

Please review the file for issues. 

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

beegee-tokyo, 2018-07-11:
No jeed to use Git anymore. You can install Arduino-ESP32 with the board manager of ArduinoIDE now. [Wiki](Add ESP32 to boards manager)

lbernstone, 2018-07-12:
Just as an FYI, this site has nothing to do with git-scm.  It is one of many git clients for windows.  There are others which do not have viruses or other malware included.  Please close this issue.

Starkshub, 2018-07-13:
More Information...

Beegee and Ibernstone.. Thank you for your input.

I have been using the UNO for some time and recently moved to the (Feather) ESP 32. Using the Uno was easy. This new board is not easy to install and I am seeking assistance as I am not an expert with GIT and the repository.

Beegee - I did go to the WIKI link but it seems that it is not from the direct 32 ESP board manufacturer. 

I did contact GIT directly and reported the Norton Virus issue. They responded by requesting I submit a review to Norton which I did complete. 

1) They suggested that I simply download the zip file from the link on the repository, however, I was not informed what to do once that is downloaded and unzipped (again, I am not a GIT expert). Do I copy the files? Where do I copy them to? What next????

Another option:

2) They suggested I download the GIT GUI from (https://desktop.github.com/) which I did and received no virus problems. I have read through the instructions provided on the repository regarding the target location, and am having difficulty finding the target "Hardware" folder. Is this something that I must create? 

Any help or direct instructions regarding both processes would be appreciated.

lbernstone, 2018-07-13:
The board manager link is https://dl.espressif.com/dl/package_esp32_dev_index.json .  That is hosted on the manufacturer's site (even though it is not strictly their code).

Starkshub, 2018-07-13:
Ok all.. I got the board installed.

- I used the link (https://desktop.github.com/) which allowed for the GIT GUI
- The source file was the file provided in the instructions.
- You then need to create two files
1) "hardware" file
2) "espressif" file
- an arduino-esp32 folder will be created
- you then need to enter the folder and execute - get.exe
- Note - when using the Adafruit Feather 32, you need to select that board as opposed to other 32 boards.
There is also a youtube video that helped.
https://www.youtube.com/watch?v=U-9746O78SM

I will close this issue....

I thank all of you for your help.

closed 2018-07-13 by Starkshub