espressif/arduino-esp32 #1626: Weird characters shows randomly in serial

arminsalcin, 2018-07-12:
### Hardware:
Board:							Nodemcu v1.1
Core Installation/update date:                9/jul/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
I use esp32 that runs relays via shift register(74hc595) relays runs AC load , so i made a rtos software , and i have problem that run randomly so in middle or start of program i get randomly characters on serial and software freeze , and characters just keep writing in serial . Only way i can repair it is to press reset button manually but it comes again in some time .

### Sketch:
```cpp

#include <farma.h>

Farma farma ;
TaskHandle_t CheckWaterLevelTask , WaterPumpTask;

void setup() {
  Serial.begin(115200);
  delay(10);
  farma.shiftInit();
  Wire.begin();
  farma.wifiStart();
  delay(500);
  farma.mqttStart();
  delay(500);
  SPIFFS.begin(true);
  delay(500);
  farma.checkConfigFiles();
  delay(500);
  farma.setupPins();
  //set starting states
  farma.turnOffConditioner();
  farma.exe = true;

  delay(300);
  //turn permamently air pump
 // 

  //delay(10);
  //pinMode(levelPin1, INPUT_PULLUP);
  //attachInterrupt(digitalPinToInterrupt(levelPin1), handleInterrupt, FALLING);

  xTaskCreatePinnedToCore(suspendTask, "SuspendTask", 7000, NULL, 1, NULL, 1);
  delay(500);
  xTaskCreatePinnedToCore(mqttLoop, "MqttLoopTask", 10000, NULL, 2, NULL, 1);
  delay(500);
  xTaskCreatePinnedToCore(testLoop, "TestTask", 7000, NULL, 2, NULL, 1);
  delay(500);
  xTaskCreatePinnedToCore(sendReadings, "SendReadingsTask", 10000, NULL, 2, NULL, 1);
 delay(500);
  xTaskCreatePinnedToCore(controlFan, "FanTask", 7000, NULL, 1, NULL, 1);
  delay(500);
  xTaskCreatePinnedToCore(controlWaterLevel, "CheckWaterLevelTask", 5000, NULL, 3, &CheckWaterLevelTask, 1);
  delay(500);
  xTaskCreatePinnedToCore(controlPump, "WaterPumpTask", 7000, NULL, 2, &WaterPumpTask, 1);
  delay(500);
  xTaskCreatePinnedToCore(controlWaterTemp, "WaterTempTask", 5000, NULL, 1, NULL, 1);
  delay(500);
  xTaskCreatePinnedToCore(controlLight, "LightTask", 5000, NULL, 2, NULL, 1);
  delay(500);

}

void loop() {
  farma.klimaLoop();
}

void testLoop( void * parameter )
{
  for (;;) {
  //  Serial.println( farma.readWaterTemp());
    vTaskDelay(5000 / portTICK_PERIOD_MS);
  }
}

void mqttLoop( void * parameter )
{
  for (;;) {
    farma.mqttLoop();
    vTaskDelay(100 / portTICK_PERIOD_MS);
  }
}

void sendReadings( void * parameter )
{
  for (;;) {
    farma.sendReadings();
    vTaskDelay( (120 * 1000) / portTICK_PERIOD_MS);
  }
}

void controlPump( void * parameter )
{
  for (;;) {

    farma.ctrlPort(1, 1);
    Serial.println("on pumpa");
    vTaskDelay((farma.Setpoint.pumpDuration * 1000)  / portTICK_PERIOD_MS); //farma.Setpoint.pumpDuration * 1000
    farma.ctrlPort(1, 0);
    Serial.println("off pumpa");
    vTaskDelay((farma.Setpoint.pumpDelay * 1000) / portTICK_PERIOD_MS); //farma.Setpoint.pumpDelay
  }
}

void controlWaterTemp(void * parameter )
{
  for (;;) {

    farma.waterTempLoop();
    vTaskDelay((60 * 1000) / portTICK_PERIOD_MS);
  }
}

void controlLight( void * parameter )
{
  for (;;) {
    farma.ctrlPort(6, 1);
    Serial.println("on light");
    vTaskDelay((farma.Setpoint.ledTimeOn * 1000)  / portTICK_PERIOD_MS); //farma.Setpoint.ledTimeOn
    farma.ctrlPort(6, 0);
    Serial.println("off light");
    vTaskDelay((farma.Setpoint.ledTimeOff * 1000) / portTICK_PERIOD_MS); //farma.Setpoint.ledTimeOff

  }

}

void controlFan(void * parameter)
{
  for (;;) {
    farma.ctrlPort(2, 1);
    farma.ctrlPort(4, 1);
    Serial.println("on fan");
    vTaskDelay((43200 * 1000)  / portTICK_PERIOD_MS);
    farma.ctrlPort(2, 0);
    Serial.println("off fan");
    vTaskDelay((43200 * 1000) / portTICK_PERIOD_MS);
  }
}

void controlWaterLevel(void * parameter)
{
  for (;;) {
    if (digitalRead(levelPin1) == 1) {
      farma.ctrlPort(7, 1);
      Serial.println("Water is filling");
    }
    vTaskDelay((86400 * 1000) / portTICK_PERIOD_MS);
  }
}

void suspendTask(void * parameter)
{
  for (;;) {
    // Serial.println("suspend task");
    //Serial.println(farma.suspendPump);
    if (farma.suspendPump == 1 && farma.suspendFlag == true) {
      farma.ctrlPort(1, 0);
      vTaskSuspend(WaterPumpTask);
      Serial.println("Task suspended");
      farma.suspendFlag  = false;
    }
    else if (farma.suspendPump == 0 && farma.suspendFlag == false) {
      vTaskResume(WaterPumpTask);
      farma.suspendFlag  = true;
    }

    vTaskDelay(6000 / portTICK_PERIOD_MS);
  }
}

static void handleInterrupt()
{
  farma.ctrlPort(7, 0);
}
```

[image]

me-no-dev, 2018-07-12:
depending on how GPIO0 and EN are connected to the USB-UART chip, sometimes the USB-UART freaks out and switches baudrate. Have seen this with both SiLabs and FTDI chips. In reality if you attach another USB-UART you will see that data is there and baudrate is fine

arminsalcin, 2018-07-12:
but why all freeze , so relays just turn down , and nothing is doing , it seems i get better results as i changed voltage of 74hc595 to 3.3v

arminsalcin, 2018-07-12:
It seems i fixed it changing 74hc595 voltage to 3.3v , i cant resimulate bug now , so if anyone have explanation why it will be good but , we can close this issue . Thanks!

closed 2018-07-12 by arminsalcin

ArTyGill, 2019-01-22:
The same thing happens with constant pickups. This is a hardware problem esp32.

afeser, 2022-11-30:
I have the same problem when Raspberry Pi is a master and ESP32 is a slave,
In my case, it sometimes works. Therefore, I am not sure if this is a problem with ESP32 or with Pi.