espressif/arduino-esp32 #1629: IDF component <-> Makefile based on makeESPArduino

dannybackx, 2018-07-12:
### Hardware:
Board:							MH-ET Live ESP32 MiniKIT v1.1, also TTGO variant
Core Installation/update date:			?11/jul/2017?
IDE name:						IDF component <-> Makefile based on makeESPArduino
Flash Frequency:					N/A
Upload Speed:						115200

### Description:
I have an unfinished development project based to make a touch screen (2.8" OLED) home alarm system. The OLED is controlled via TFT_eSPI. It also queries the weather (wunderground.com).
Switching build from makeESPArduino to esp-idf was a bit of work on trivial stuff, no big deal.

However, the app now doesn't show an image on the OLED, and queries to wunderground appear to hang. Other stuff may be malfunctioning too. Peer to peer communication between the nodes does appear to work.

I would be willing to switch to esp-idf as it appears to be the vendor preference, but how do I go about this ? Or are we only at a point (with esp-idf) where I shouldn't do this yet ?

### Sketch:
https://github.com/dannybackx/espalarm.git

lbernstone, 2018-07-12:
ESP-IDF has a full set of its own APIs, on which arduino-esp32 is based.  It includes none of the C++ code/functions in arduino-esp32, so you would need to write or convert any of those functions you have used. https://esp-idf.readthedocs.io/en/latest/

dannybackx, 2018-07-12:
Thanks for the prompt reply.

Can I use C++ in my application but I need to stick to a limited (C language) API set ?
Or is C++ a no-no completely ?

    Danny

dannybackx, 2018-07-12:
In the code I've ported to esp-idf, I used the instructions for arduino-esp as a component in esp-idf. So I have an application that got compiled completely, no missing APIs.
So I think I'm beyond what you're referring to.
The debug output I've added to my original message above is from the esp-idf compiled application. 

I currently see two differences with code compiled with the "old" build system : 
- OLED lights up (backlight gets controlled by pin 5 as it should) but the display doesn't show what it should, so the code using TFP_eSPI doesn't do its job
- some TCP/IP code works (peer communication) but the queries to wunderground fail (e.g. last line in the debug log above)

Should esp-idf + the arduino-esp32 component be stable enough for a non-trivial application such as this one ?
Are there known issues with porting code that controles multiple devices over SPI (the OLED is basically two devices : a screen and a touch interface) ?
Are there known issues with porting code that uses WiFiClient and prints http to it ?

I tried moving to esp-idf's esp_http_client from within my old build environment, but esp_http_client relies on mbedtls which ... so I ran into trouble there. Is there a good method for porting this part of an application ?

everslick, 2018-07-13:
Please close.