espressif/arduino-esp32 #1633: How to config dhcp IP range in Access Point ?

QuentinGuillet, 2018-07-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:				        40Mhz
Upload Speed:						115200

Hello,

I am using several ESP32 as AP and station mode to have connection in chain. In order to do that, I use static IPs for the ESP, but I also want to connect sometimes to an ESP with my phone to access some information on a webserver. The issue is that the dhcp on the server of the ESP gives me the same IP Address than the ESP which is connected to my Access Point.

To avoid this issue I would like to change the dhcp IP range given to a station mode. I found a way to do it on an ESP8266 but not on an ESP32.

If you have any clue on how to do this, I would be very thankfull.

Thanks anyway
Quentin

me-no-dev, 2018-07-13:
For example to make the network 192.168.5.X

```cpp
WiFi.softAPConfig(IPAddress(192, 168, 5, 1), IPAddress(192, 168, 5, 1), IPAddress(255, 255, 255, 0));
```

closed 2018-07-13 by me-no-dev

QuentinGuillet, 2018-07-13:
Thanks for the answer.
But what I would like, is that when I connect with my phone, I get the IP address 192.168.5.100, and not 192.168.5.2, like for an esp8266. Is it possible ?

Thanks for the reply !

me-no-dev, 2018-07-13:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiAP.cpp#L162

change 1 to 99 and 11 to 110 and you will have it start from 100

QuentinGuillet, 2018-07-13:
Thanks very much !