espressif/arduino-esp32 #1636: Remove unused variables in release build

ciband, 2018-07-13:
### Hardware:
Board:							Adafruit ESP32 Dev
Core Installation/update date:			 Platformio Latest
IDE name:							PlatformIO
Flash Frequency:					PlatformIO default
Upload Speed:						?

### Description:
Variables used only in debug messages show up as unused variables in release build and generate warnings in platformio.

From platformio:

WiFiSTA.cpp:645:30: warning: unused variable 'type' [-Wunused-variable]

WiFiSTA.cpp:656:26: warning: unused variable 'ip' [-Wunused-variable]

```

me-no-dev, 2018-07-13:
those are used when debugging. disregard the warnings ;)

closed 2018-07-13 by me-no-dev

ciband, 2018-07-13:
I understand they are just for debugging.  It is just annoying to have warnings if you want your project to compile clean, especially if you turn on treat warnings as errors.

I would think the code could be modified to only define those variable during debugging.  I know it isn't critical, just nice to have.

me-no-dev, 2018-07-13:
surely at some point it will happen, but as you said not a priority at all :)