espressif/arduino-esp32 #1637: esp32 arduino ide uploading problems

gabovich, 2018-07-13:
### Hardware:
Board:							ESP32 NodeMCU-32S 
Core Installation/update date:		10/jul/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Hello,
I'm a beginner with the esp32 and I'v got couple of things that I don't understand-
1. While I'm uploading a code it's took too much time no matter how long the code is and if this the first uploading.
2. While uploading, I must press for a few seconds on the reset bottom, otherwise I get "A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\r')"
3. Is there any meaning for the Flash Frequency? I tried both 80Mhz and 40Mhz and they worked the same.

Thanks for helping.

me-no-dev, 2018-07-13:
What kind of board is that? can you link me to it?

gabovich, 2018-07-13:
That's was the headline of the product I bought:
ESP-32S ESP32 NodeMCU Development Board 2.4GHz WIFI+Bluetooth Dual Mode

closed 2019-08-15 by stale[bot]