espressif/arduino-esp32 #1638: ESP32-WROOM32: Issue with hardware or software?

victorsanchis, 2018-07-13:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board: esp32wroom32 - Arduino ESP32 Dev Module, yes.
Core Installation/update date:			latest, fresh install
IDE name:							Arduino IDE 1.8.5, fresh install
Flash Frequency:					40Mhz?
Upload Speed:						115200

### Description:
I don't know how this happened but I got a problem with 2 esp32 units -one with the below indicated issue, plus, another one I've just tried after having problems with the first ESP32 unit-. 
I can compile any example or my own sketches but, once the sketch is loaded, it doesn't work and, through the serial monitor I can only get same message, which makes me think if both units have same problem (same day?) because of soft or hardware reasons (?) as "core 0" is continuously mentioned:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
assertion "index >= 0 && index < Nitems" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/compressed_enum_table.hpp", line 38, function: Tenum CompressedEnumTable<Tenum, Nbits, Nitems>::get(size_t) const [with Tenum = nvs::Page::EntryState; unsigned int Nbits = 2u; unsigned int Nitems = 126u; size_t = unsigned int]
abort() was called at PC 0x400d33ab on core 0

Backtrace: 0x400883f8:0x3ffc6fd0 0x400885fb:0x3ffc6ff0 0x400d33ab:0x3ffc7010 0x400d6292:0x3ffc7040 0x400d69b1:0x3ffc7090 0x400d6c37:0x3ffc7100 0x400d6ea5:0x3ffc7160 0x400d5bb6:0x3ffc71c0 0x400d5a11:0x3ffc7210 0x400d5acb:0x3ffc7230 0x400d5b16:0x3ffc7250 0x400d0a24:0x3ffc7270 0x400e569f:0x3ffc7290 0x400d430e:0x3ffc72c0
..............................................................................................................................................
Just in case: 
I have re-installed Arduino IDE and also charged the last latest ESP32 library version (https://github.com/espressif/arduino-esp32). 
I have pushed the "Boot" & "En" buttons for long times... doesn't matter: the sketch is compiled, and uploaded but... the ESP32 shows always same issue/message. 
...............................................................................................................................................
I'm totally newbie with Arduino and programming (0 knowledge!) but I have been able to connect some sensors to both ESP32 units and connect them to a couple of IoT platforms and get those sensor values on the account I opened. However, since yesterday both units show same problem.

### Sketch:
It doesn´t matter the sketch: the unit simply seems to reboot all time... (!)

### Debug Messages:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
assertion "index >= 0 && index < Nitems" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/compressed_enum_table.hpp", line 38, function: Tenum CompressedEnumTable<Tenum, Nbits, Nitems>::get(size_t) const [with Tenum = nvs::Page::EntryState; unsigned int Nbits = 2u; unsigned int Nitems = 126u; size_t = unsigned int]
abort() was called at PC 0x400d33ab on core 0

Backtrace: 0x400883f8:0x3ffc6fd0 0x400885fb:0x3ffc6ff0 0x400d33ab:0x3ffc7010 0x400d6292:0x3ffc7040 0x400d69b1:0x3ffc7090 0x400d6c37:0x3ffc7100 0x400d6ea5:0x3ffc7160 0x400d5bb6:0x3ffc71c0 0x400d5a11:0x3ffc7210 0x400d5acb:0x3ffc7230 0x400d5b16:0x3ffc7250 0x400d0a24:0x3ffc7270 0x400e569f:0x3ffc7290 0x400d430e:0x3ffc72c0

Rebooting...

```
Can anyone help me to fix this? 
Thank you very much in advance for any help!

Victor
P.s.: remember I am totally newbie on this subjects ('though I always try my best :-) )

lbernstone, 2018-07-13:
As mentioned at the very top of the issue report, please run [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to determine what is throwing the exception in your code.  If an example sketch like [WiFiClient](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino) throws this same error, then wipe your device by running `python esptool.py erase_flash` and try again.

thetotalwimmer, 2018-07-19:
Is this in combination with the DeepSleep-Mode? I had issues with that in combination with the Serial.println function.

mmashraf, 2019-07-31:
Hi I buy new ESP32 from Ali express and after first program of blink , it is continuously showing the sw_reset  issue on serial port.   i  also try  python esptool.py erase_flash but it does not work and also following error
raceback (most recent call last):
  File "esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
ets Jun  8 2016 00:22:57

mmashraf, 2019-07-31:
these two commands help me for erase flash. -p with COM4 where ESP32 is connected is required from windows device manager view.
pip install pyserial
python esptool.py -p COM4 erase_flash

after erase of flash following is error. Please advise How it can be fixed to use this eSP32 module. below is continuous error and I was not able to fix it.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

beegee-tokyo, 2019-07-31:
First of all, it is not a good practice to high-jack an issue with something unrelated. You should open a new issue if you cannot find an issue that addresses ```flash read err, 1000```

I had the exact same issue today with a cheap Chinese ESP32 Wrover board.

Problem with ```flash read err, 1000``` was related to a power supply issue.

After supplying from a 2.1A USB power supply the error went away and the ESP32 works as expected.
Supplying from USB port of my PC and the error comes back. 

Try a big capacitor on GND and +5V pin of the ESP32 modules or look for a powered USB hub that gives enough power on USB.

mmashraf, 2019-08-02:
thank for response please suggest capacitor value. Even I connect new power supply with good reading but I have to upload program which is also not happening

beegee-tokyo, 2019-08-02:
Try 470uF or larger.

What could help as well is to try a different USB cable.

For me 2 out of 5 USB cable have problems to supply the ESP32 when connected to the PC.

If you are on Windows, check the settings for the USB ports. Mine was set to "Allow to shutdown USB device for power saving". Disabling that helped as well.

mmashraf, 2019-08-05:
remarks from <lbernstone> are valuable and it looks that it solved the problem.
Sounds like your code upload is failing
See #3036 i.e 
https://github.com/espressif/arduino-esp32/issues/3036

closed 2019-10-18 by stale[bot]

halitekmekcioglu, 2019-11-24:
Same thing on me! I was programmed ESP32-CAM in my office with my laptop and I left it while it's properly working. After that, I came home and I started to connect it again. Only one thing happened as extraordinary, I've connected directly +5v and board had been warmed a lot, and then I realized it and I plugged it to 3.3V. After that, I started to get this script on the serial port. Did anyone fix it yet?

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xcf3f00c7,len:496
ho 0 tail 12 room 4
load:0x191c5e11,len:1494867736
ets Jun  8 2016 00:22:57

rampro, 2019-11-26:
> Same thing on me! I was programmed ESP32-CAM in my office with my laptop and I left it while it's properly working. After that, I came home and I started to connect it again. Only one thing happened [...]

Set Partition scheme to Minimal SPIFFS

halitekmekcioglu, 2019-11-26:
> Set Partition scheme to Minimal SPIFFS

It seem still same.

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

chegewara, 2019-11-26:
> It seem still same.
Brownout detector was triggered

And the answer is still the same. Poor power source or bad quality USB cable.

halitekmekcioglu, 2019-11-26:
> > It seem still same.
> > Brownout detector was triggered
> 
> And the answer is still the same. Poor power source or bad quality USB cable.

You are right, thanks.

sunitasonu, 2020-01-29:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

code is getting upoaded but i am  getting above error plz if anyone reply it will be helpfull thank you

e-labInnovations, 2020-01-30:
I had an ESP32cam working for a few days then started getting that message at boot up. Reloaded the program and still no camera.:disappointed:

### Error:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

Brownout detector was triggered

```
[image]

### Camera label:
HW-297
OV2640

### In program:
```
#define CAMERA_MODEL_AI_THINKER
```

### Board selection:
ESP32 Wrover Module  

[image]

-------

### At the time of working
[image]

sansillusion, 2020-01-30:
Brownout means not enough power/amperage at peaks. Check that your power
source is sufficient and power/usb cable is good.

vimalswaroopj, 2020-04-03:
change the board to DOIT ESP32 DEVKIT V1. It worked for me for the bluetooth program. Hope this will help some one.

avellinosantiago, 2020-04-16:
this happened using an i2c sensor and not initializing correctly mysensor.begin();
i copied some code and removed a check for a bad connection.. if (mysensor.begin() == false ){
and by doing so I removed the initialization of the sensor on the bus...

A4lfr32, 2020-04-24:
> change the board to DOIT ESP32 DEVKIT V1. It worked for me for the bluetooth program. Hope this will help some one.

work for me, thanks

RohitAsegaonkar, 2020-08-06:
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
ets Jun  8 2016 00:22:57

This error is appearing, even though code gets uploaded on board.

josetomas, 2020-08-28:
Hi! I have an Adafruit Huzzah feather ESP32 board, it was working for me, and now I'm receiving the following error: 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

phorenabess, 2020-10-03:
Did you run into similar issue on ESP32-WRROM32D?

antimix, 2020-10-27:
The error:
------------------
load:0x40080400,len:6364
entry 0x400806b8
Brownout detector was triggered
-----------------------------------
Is caused by poor power connections on the board. 
After spending weeks thinking it was a software issue, I took a tester and started to measure power consumption and voltages on the ESP32-CAM. The ESP32 was plugged in an cheap breadboard with several wires.
The board consumption was around 110-200 mA and I was powering it with 5V on the breadboard with wires to the 5V pin and GND. I did not power the ESP board directly, but the breadboard, with some other pin wire cables to do the connections.
I discover than on the power supply wire I had 5.02V, but on the VCC5V pin of the ESP32 I had 4.42V !!!! Impossible but true.
And while I pushed or moved the ESP board on the breadboard, or moved or pushed power cables on the breadboard, the voltage values changed randomly from 4.02 to 5.02V. Then by chance I got the 5.02V on the ESP32 board, and I re-flashed the firmware, very carefully to not move the contacts, and as of sudden all worked fine without any brownout error.
After some minutes I moved by mistake a power cable, and the board dropped on the table, creating false contacts again, and again I had fluctuating voltages and brownout error at boot, until I was able again to have a more or less correct contact.

LESSON LEARNT: NEVER USE breadboard to power such boards, and use a specific shield with power supply plug, or solder the wires directly.

pedrorincon, 2021-01-10:
I solved the problem by placing 3 220uF capacitors between GND and 5v

egeozgul, 2021-06-11:
> I solved the problem by placing 3 220uF capacitors between GND and 5v

This worked for mee too!! Thank you so much. I mounted 2 100uF capacitors between GND and 5v and it worked!

dragon1933, 2022-06-12:
各位好，我是esp32的新手。
我遇到一個奇特的問題，我是用esp 32 devkitc的開發板，接tftlcd(自帶sd卡)，我想要讓它能同時讀sd卡、觸控及顯示。一開始可以，但後來出現這個錯誤後，我就無法再連到sd卡了。我找了很久，仍找不出錯誤在哪，請各位朋友幫忙解決問題。
出現的訊息：
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Card Mount Failed

[image]
這是我的接法

dragon1933, 2022-06-13:
> I had the same issue and it was caused by insufficient power. I changed
> the power supply and it worked

Good night, antimix~
Thank you for the information, it is very valuable to me.
Out of curiosity, the replacement of the power supply refers to the
addition of capacitors. Or any other way?

Sorry, my English is not good, I use google web page translation
software, please forgive me if it caused your misunderstanding.

antimix, 2022-06-14:
Hello,

no I was using somethink like this
*https://elektronicavoorjou.nl/en/product/breadboard-power-supply/*
<https://elektronicavoorjou.nl/en/product/breadboard-power-supply/>
But id did not work well, especially with my ESP32 CAM, probably because
the technical specification are not the real one, and the ESP32 has several
peak consumptions (may reach 900mA) and the power supply could give enough
power.
I replaced it with something like this
https://www.bol.com/nl/nl/p/5v-2-5a-power-supply-voedingsadapter-2500ma-micro-usb-met-eu-stekker-voor-o-a-raspberry-pi-3-en-3b/9200000080763810/
and with 2A it works very well.

regards

dragon1933, 2022-06-15:
Thank you Mr. Dafang for sharing, I will buy more equipment to try.

j-art, 2024-01-12:
I had the same error but it was my code. I had a for without variable initialization

for (int i; i < 8 ; i++) {
  str_id[i] = Msg[i];
}

I initialized i=0 and the problem was solved. Maybe this is not the case, but It can be helpful for some one.

for (int i=0; i < 8 ; i++) {
  str_id[i] = Msg[i];
}

It works for a teensy but when I migrated the code to ESP32 It crashes