espressif/arduino-esp32 #1642: WIFI CONNECTING TO IPHONE HOTSPOT BUT NEVER TO WPA2 PERSONAL

gggabriella, 2018-07-15:
### Hardware:
Board:							        ESP32 ADAFRUIT FEATHER HUZZAH
Core Installation/update date:			14/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					        80Mhz
Upload Speed:						921600

### Description:
I can connect to Iphone's Hot Spot WiFi (the device is able to connect) but it's impossible to connect in my basic WPA2 Personal fiber optic home connection. It's driving me mad, I need this to be reliable. Any suggestions to make it connect?
I tried changing wifi encryption, channels, WPS... the serial port keeps eternally trying to connect and repeating that STA_DISCONNECTED and NO_AP_FOUND and clearly I can see the signal when I do a Wifi Scan in the same device... PLEASE HELP.

[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND

### Sketch:

#include <WiFi.h>
 
const char* ssid = "Orange_Juice";
const char* password =  "vitaminc";
WiFiServer server(80);
 
void setup() {
  Serial.begin(115200);
  delay(10);
  WiFi.enableSTA(true);
  Serial.print("WIFI status = ");
  Serial.println(WiFi.getMode());
  WiFi.disconnect(true);
  WiFi.mode(WIFI_STA);
  Serial.print("WIFI status = ");
  Serial.println(WiFi.getMode());
  WiFi.begin(ssid, password);
  connectWiFi();
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");
 
}
 
void loop() {
 
}

stickbreaker, 2018-07-15:
@gggabriella if you run the `WiFiScan` example, does the ESP32 see your router?  The Esp32 only uses the 2.4ghz band, many of the new routers by default only use 5ghz.

Chuck.

gggabriella, 2018-07-15:
@stickbreaker yes the ESP32 can see my Wifi connection, it is optic fiber and has two simultaneous 2.4ghz and 5ghz signals, the signal it recognize is the 2.4ghz one, which is the one i'm using in the code.

me-no-dev, 2018-07-17:
try
```cpp
#include <WiFi.h>
 
const char* ssid = "Orange_Juice";
const char* password =  "vitaminc";
 
void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
}
 
void loop() {}
```

gggabriella, 2018-07-17:
@me-no-dev tried, but the same, it stays trying to connect forever...
-------------------------
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
Connecting to WiFi..
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
Connecting to WiFi..
Connecting to WiFi..

stickbreaker, 2018-07-17:
@gggabriella run the WiFiScan  example and upload the result. That way we can have more information to diagnose the problem.

Chuck.

gggabriella, 2018-07-18:
@stickbreaker my 2.4ghz signal is Orange_Juice, the optic fiber router has another 5ghz 
-------------------------------
⸮   ⸮ ⸮⸮⸮⸮⸮ ⸮⸮           ⸮   ⸮       ⸮        ⸮ ⸮⸮⸮ ⸮⸮ ⸮⸮  ⸮ ⸮⸮  ⸮   ⸮⸮ ⸮ ⸮⸮ ⸮    ⸮⸮     ⸮  ⸮⸮⸮ ⸮⸮⸮⸮⸮⸮⸮ ⸮  ⸮⸮⸮ ⸮  ⸮⸮⸮⸮  ⸮⸮⸮⸮ ⸮⸮ ⸮⸮⸮⸮ ⸮ ⸮⸮⸮⸮ ⸮  ⸮   ⸮⸮⸮⸮⸮⸮⸮ ⸮⸮ ⸮    ⸮⸮  ⸮     ⸮  [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 1 - SCAN_DONE
scan done
22 networks found
1: Orange_Juice  (-50)*
2: MOVISTAR_4004 (-66)*
3: vodafone0E0B (-73)*
4: MiFibra-26BD (-81)*
5: MOVISTAR_9495 (-83)*
6: MOVISTAR_9D54 (-83)*
7: DIRECT-A0-HP ENVY 4520 series (-87)*
8: Orange-5159 (-88)*
9: MIWIFI_2G_dTXY (-89)*
10: MOVISTAR_BE5E (-90)*
11: margarit38 (-90)*
12: Invitado-8BCB (-91)*
13: JAZZTEL_tTF6 (-91)*
14: JAZZTEL_nz6q (-92)*
15: JAZZTEL_YAkC (-92)*
16: MIWIFI_2G_CzFR (-93)*
17: Orange-05BF (-93)*
18: VANCOMOLOCOS (-94)*
19: MiFibra-2C68 (-94)*
20: MOVISTAR_3F78 (-94)*
21: Orange-DEBF (-95)*
22: MASMOVIL_stv4 (-95)*

me-no-dev, 2018-07-18:
that is rather ridiculous ... WTF? what are the specs of the security of your AP? There must be a reason

capedra, 2018-07-18:
@gggabriella
Maybe it's time to call our nice friend Ivan Grokhotkov (@igrr) and I think he would probably tell you to enable WiFi's debug on Arduino IDE option and copy-paste the results on here.

me-no-dev, 2018-07-18:
@capedra debug is enabled and the log is above ;)

me-no-dev, 2018-07-18:
@gggabriella just out of curiosity... did you try changing the name of the AP?

gggabriella, 2018-07-18:
@me-no-dev attached image of wifi specs, let know if you can't see it

[image]

beegee-tokyo, 2018-07-19:
WiFi Mode **`G+N`** should work, but did you try **`B+G+N`**

gggabriella, 2018-07-19:
@me-no-dev @stickbreaker @capedra @beegee-tokyo @sctv  thank you all, at last I got it fkin working <3!!! 

The WiFi router model is an ARCADYAN PRV3399B-B-LT
I enabled the smart WiFi configuration and the band steering, so both bands got the same SSID. Also changed the SSID (just in case) and restore the security mode back to WPA2 AES
https://tenor.com/es/ver/toddlersandtiaras-victory-dance-gif-4514304

[image]

me-no-dev, 2018-07-19:
@gggabriella great news! I knew it must be some setting that is preventing the connection.

@capedra Ivan has more than enough to deal with ;) there is no need to bring him to Arduino land, unless you think that we are incapable of taking care of things... I talk to him only when I am sure that it's worth wasting his time and this is no way was such a case.

closed 2018-07-19 by me-no-dev

capedra, 2018-07-20:
@gggabriella No problem. It's also great to see that you disabled the vulnerable WPS as well. WPS is a bit like keeping the key to your high-security $10,000 company safe in a $9.99 petty cash box in the top drawer of your desk.
Feel free to open new issues whenever you want since we're here to help everyone around.

@me-no-dev Our Strong Community is very capable of sorting things out, but without him, Espressif Systems wouldn't be the same. So, I thank God that guys like Ivan exist in the world. And I miss talking to him. So, I wish him all the best!

Best Regards, and Thanks for Everyone!