espressif/arduino-esp32 #1645: esp32 mesh

Plissk3n, 2018-07-16:
Hi there!

I  saw that espressif has now released the mesh functionality for the esp32. 
Will this also be ported to the arduino-core here?

I searched and found an esp_mesh.h file but couldn't find an example.
Would be really cool to see.

In any case thank you so much for doing an incredible job with all of this, it's a lot of fun.

lbernstone, 2018-07-16:
It is a bit too fresh to have arduino code made yet.  There are examples in esp-idf. https://github.com/espressif/esp-idf/tree/master/examples/mesh

closed 2019-08-15 by stale[bot]

lbernstone, 2019-08-15:
This really should be converted to use the esp-idf functions, which should be much faster, but it works...
https://github.com/lbernstone/ESP_WiFiMesh