espressif/arduino-esp32 #1648: WifiClient write is slow

HKGK, 2018-07-16:
### Hardware:
Board:							ESP32 Dev Module C
Core Installation/update date:		10/jul/2018
IDE name:						Arduino IDE 1.8.5 IDF from 10 Jul 2018
Flash Frequency:					40Mhz
Upload Speed:					115200

I have been battling to get my WiFi write speed above 8-12MB/sec. My read speed(Seems to recently have improved for also about 10MB/sec...) on same platform is +-180MB/sec which makes me very happy. 

Basically the code looks like this.

#include <WiFi.h>

void setup()
{
  Serial.begin(115200);

  int status = WiFi.status();
 
  while ( status != WL_CONNECTED)
  {  
    status = WiFi.begin("SSID", "password");
    delay(10000);
  }

  Serial.print("WiFi Connected");
}

void loop() 
{
  WiFiClient dclient;

  dclient.connect("server",port); //Connect to someone that will listen

  dclient.setNoDelay(1);

  unsigned char txBuf[1760]={ 0 };
       
  while((WiFi.status() == WL_CONNECTED)&& (dclient.connected())  )
  {
    int txBytes = dclient.write(txBuf, sizeof(txBuf));    
  }
}

Some help/suggestions would be so much appreciated! My hair is running low :)

Regards

Gideon

HKGK, 2018-07-16:
When all is said and done, I need to transfer a big file from SD storage to a server via WiFI using FTP

me-no-dev, 2018-07-17:
You should maybe look into Async (check my repos) if you want high speeds. Currently Async is 20-40 times faster than the regular client.

HKGK, 2018-07-17:
Look promising:)

From a quick look at the ESP32 AsynchTCP class it would seem for sending data, its a simply changing dclient to be of class AsyncClient and not a WiFiClient? I am excited and am going to give it a try later tonight!

Is there a simple ESP32 example?

HKGK, 2018-07-18:
Battling a bit.
void loop()
{
  WiFiClient dclient;
  dclient.connect("server",port); //Connect to someone that will listen
  dclient.setNoDelay(1);
  unsigned char txBuf[1760]={ 0 };

  while((WiFi.status() == WL_CONNECTED)&& (dclient.connected()) )
  {
    int txBytes = dclient.write(txBuf, sizeof(txBuf));
  }
}
With this code and WiFiClient

me-no-dev, 2018-07-18:
Async is not a drop-in replacement. it works based on callbacks and is overall not easy to deal with. It requires a good understanding of how TCP works.

jeroenhouben, 2019-03-25:
@HKGK any progress? I'm facing the same "slowness" . while uploading a file from SD.

ScruffR, 2019-05-29:
This issue still seems unaddressed.
One would expect a lot more throughput on a dual core 240MHz controller than that.

The funny thing is that the client _can_ achieve higher rates (~20KB in ~0.4 sec) but not consistently.

```
void serverStream() // callback for ESP32WebServer (stripped down)
{
  WiFiClient client = server.client();

  String response = "HTTP/1.1 200 OK\r\n";
  response += "Content-Type: multipart/x-mixed-replace; boundary=frame\r\n\r\n";
  server.sendContent(response);

  while (client.connected())
  {
    // acquire data from camera
    // place in 64K buffer

    response = "--frame\r\n";
    response += "Content-Type: image/jpeg\r\n\r\n";
    server.sendContent(response);

    if (client.connected()) {
      client.write(buffer, len);
      // delay(1); // optional for testing
    }
  }
}
```

ScruffR, 2019-08-01:
Why is it unaddressed? I'd think this should be considered imporatant and not hope for this to vanish by itself ;-)

alternadom, 2020-01-08:
Hi,
I'm having the same problem here!
No problem at RX but when I want to TX something there is a lot of lag :/
I mentioned the problem here :
#https://github.com/AlphaLima/ESP32-Serial-Bridge/issues/10#issuecomment-571978644
Hope I will find a solution with you all!

WavRX, 2020-06-14:
Experiencing the same thing, hoping for a solution too.

RafigRzayev, 2020-11-12:
@HKGK  @jeroenhouben @ScruffR  @WavRX @alternadom  Anyone was able to increase the speed? I would appreciate if you could share your experience in case you did.

I am having app. 30kB/sec read speed at the moment, though I need something like 1MB/sec.

HKGK, 2020-11-17:
Sadly I never got to the bottom of this and had to work with the poor throughput.

Please let me know if you have any luck

________________________________

Gideon Krige

Somerset West

South Africa

RafigRzayev, 2020-12-09:
@HKGK  Hi, I was able to increase the speed 10 times. I will post instructions soon at https://github.com/espressif/arduino-esp32/issues/4529