espressif/arduino-esp32 #1651: ESP-DeepSleep-Mode Bug

thetotalwimmer, 2018-07-17:
### Hardware:
Board:							        ESP32 Dev Module (ESP_32_Core_board_V2)
Core Installation/update date:			17/Jul/2018
IDE name:							Arduino IDE

### Description:
When I run the example code for the DeepSleep-Mode of the ESP32 (TimerWakeUp) only the first 4 letters of the Serial.println()-function will be printed. Right after that the ESP directly enters the DeepSleep-Mode.
Interesting fact: when I do the same with a FireBeetle-Board from DFROBOT, the first 6 letters will be printed.

### Sketch:
/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  5        /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case 1  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 2  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 3  : Serial.println("Wakeup caused by timer"); break;
    case 4  : Serial.println("Wakeup caused by touchpad"); break;
    case 5  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  //Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}
```

me-no-dev, 2018-07-17:
yup ;) Serial is not synchronous, so it continues sending after the function returns and you puting to deep sleep stops the serial where it is. Why DFROBOT works differently I have no clue.

thetotalwimmer, 2018-07-17:
Thanks for your reply. I just used this scetch as an example. In our code we are calling a function right after the ESP wakes up. It does not have anything to do with the Serial.println()-function.
There must be some bug in the compiler that is affecting the DeepSleep-Mode.

thetotalwimmer, 2018-07-19:
After bootup only the 4 letters "Boot" are printed on the serial monitor. Then the ESP32 goes into DeepSleep-Mode. I also tried to use a delay before and after the print function but that did not change anything.
I also tried the following script with an LED and in that case the Serial.print worked:

#define LED 5
#define uS_TO_S_FACTOR 1000000
#define TIME_TO_SLEEP  10

RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000); 
  pinMode(LED, OUTPUT);

++bootCount;
Serial.println("Boot number: " + String(bootCount));

for (int i = 0; i < bootCount; i++) {
  digitalWrite(LED, HIGH);
  delay(500);     
  digitalWrite(LED, LOW); 
  delay(500);
}

esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
esp_deep_sleep_start();
}

void loop() {
  //This is not going to be called
}

me-no-dev, 2018-07-19:
As I suspected :) There is nothing wrong with what you see. You call Serial.print(X) and I guess you expect that print will return once all bytes are out of the serial, but that is not the case. The serial port has internal FIFO that we fill and return back to the sketch as soon as we can. You LED thing works because you have `delay` after you set the led, which is blocking and this not going into deep sleep before delay returns.

I hope I made myself clear :)

closed 2018-07-19 by me-no-dev

nerdegutt, 2018-10-14:
I'm a little late to the party here, but I had the same problem. It seems the Arduino Serial library has changed. Try flushing the serial buffer before going to sleep:

```
Serial.flush();
esp_deep_sleep_start();
```

thetotalwimmer, 2018-10-16:
Thank you very much @nerdegutt, it helped. 👍