espressif/arduino-esp32 #1654: Is it possible to get the ESP32 working as a slave device or do reads as master in an SPI communication?

pgrlopes, 2018-07-18:
Hello guys

I'm trying to connect a PSoC to an ESP32 and I've tried I2C with no success so I wanted to try out SPI. I can't seem to find any examples for using the ESP as a slave device though and I'm having trouble understanding how to do so. I need to transmit data from the psoc to the esp so I can send it through BLE/Wi-fi. Would my best bet be setting the esp as a slave or setting it as master and performing reads on it's slave? I cant find examples for any of the cases, would appreciate if someone could point me in the right direction.

Thank you

lbernstone, 2018-07-18:
You must have looked very hard. https://www.startpage.com/do/dsearch?query=esp32+spi+slave+example&cat=web&pl=opensearch&language=english

pgrlopes, 2018-07-19:
Hello sir

Thanks for the help. I've checked all those links but more than half of them are in the ESP-IDF and not on the arduino IDE. I placed my question here because what I would like to know is if there is an example applied to the arduino IDE, cause the rest of my project is done using it, having to change to the IDF would probably take me a lot of time and I don't have that time.

closed 2019-08-15 by stale[bot]