espressif/arduino-esp32 #1656: Install failure

supuflounder, 2018-07-19:
None of the parameters below have anything to do with my problem, so I have not bothered to change them.

I tried to download and install the board definitions for the ESP32 board.  What I got was this:
`Josephs-MacBook-Air:src flounder$ mkdir -p ~/Documents/Arduino/hardware/espressif && \
> cd ~/Documents/Arduino/hardware/espressif && \
> git clone https://github.com/espressif/arduino-esp32.git esp32 && \
> cd esp32 && \
> git submodule update --init --recursive && \
> cd tools && \
> python get.py
Cloning into 'esp32'...
remote: Counting objects: 8617, done.
remote: Compressing objects: 100% (30/30), done.
remote: Total 8617 (delta 11), reused 18 (delta 9), pack-reused 8578
Receiving objects: 100% (8617/8617), 136.38 MiB | 1.09 MiB/s, done.
Resolving deltas: 100% (5027/5027), done.
Submodule 'libraries/AzureIoT' (https://github.com/VSChina/ESP32_AzureIoT_Arduino) registered for path 'libraries/AzureIoT'
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into '/Users/flounder/Documents/Arduino/hardware/espressif/esp32/libraries/AzureIoT'...
Cloning into '/Users/flounder/Documents/Arduino/hardware/espressif/esp32/libraries/BLE'...
Submodule path 'libraries/AzureIoT': checked out '67dfa4f31ef88b0938dd87d955612100dea5562e'
Submodule path 'libraries/BLE': checked out '7951347ed68313d75c367e1f2cce763cb56d1eb2'
System: Darwin, Info: Darwin-17.4.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Downloading xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
Downloading esptool-2.3.1-macos.tar.gz
Done
Extracting esptool-2.3.1-macos.tar.gz
Downloading mkspiffs-0.2.3-arduino-esp32-osx.tar.gz
Traceback (most recent call last):
  File "get.py", line 148, in <module>
    get_tool(tool)
  File "get.py", line 103, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1278, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 352, in wrap_socket
    _context=self)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 579, in __init__
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 808, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)`

I have no idea what this means.  I am using Mac OS 10.13.3 ("High Sierra").  An attempt to re-install it produced
`Josephs-MacBook-Air:tools flounder$ mkdir -p ~/Documents/Arduino/hardware/espressif && cd ~/Documents/Arduino/hardware/espressif && git clone https://github.com/espressif/arduino-esp32.git esp32 && cd esp32 && git submodule update --init --recursive && cd tools && python get.py
fatal: destination path 'esp32' already exists and is not an empty directory.
Josephs-MacBook-Air:espressif flounder$ mkdir -p ~/Documents/Arduino/hardware/espressif && cd ~/Documents/Arduino/hardware/espressif && git clone https://github.com/espressif/arduino-esp32.git esp32 && cd esp32 && git submodule update --init --recursive && cd tools && python get.py
`
So  what do I do now?
       joe
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

todduk, 2018-07-21:
@lbernstone great thanks. Was struggling with `get.exe` never completing but apparently doing nothing. windows.md should probably be updated to reflect the new method.

lbernstone, 2018-07-21:
It is still in pre-release.  The pull request is in just as soon as 1.0.0 goes live.  Please close the issue if your installation completed.

drmpf, 2019-08-10:
Arduino 1.8.9 records error
Error downloading https://dl.espressif.com/dl/package_esp32_index.json
However the release .json link works
https://dl.espressif.com/dl/package_esp32_dev_index.json

closed 2019-10-23 by stale[bot]