espressif/arduino-esp32 #1657: Sensor VEML770 data not change (Wire)

asetyde, 2018-07-19:
### Hardware:
Board: ESP32 Wroom							
Core Installation/update date:			1.0.0_rc3
IDE name:							Arduino ide 1.8.5 and visual micro
Flash Frequency:					        80Mhz
Upload Speed:						115200

### Description:
Sensor VEML770 data not change after commit a59eafbc9dfa3ce818c110f996eebf68d755be24 before, data is updated correctly and sensor work, I've tried to go back with git but too many error with esptool, it's not possibile create a special realese before 27 June ?, too many big commit to find error . 

Sensor VEML770  use wire, i2c;

### Debug Messages:
```none```

stickbreaker, 2018-07-19:
@asetyde there was one significant change to `Wire()` with this last update, A `Wire()` transaction with SENDSTOP=FALSE is queued until a transaction with SENDSTOP=TRUE is send.  This is necessitated because the ESP32 i2c hardware will go into a lockedup state if the transaction is not completed with a STOP within 13.1 ms.
So, this queuing is necessary, the `Wire()` library reports the queued status by returning a ERROR=7 (continue) for all calls that use a ReSTART operation, for example:
```c++
// set i2c device internal register pointer and read 20 bytes from there
Wire.beginTransmission(id);
Wire.write(registerHigh);
Wire.write(registerLow);
uint8_t err = Wire.endTransmission(false);
// since sendStop==false, this write() command was queued until a stop if issued, so the returned status
// is set to 7 (continue) to indicate this.
 
if( err!=7){ // something went wrong
  Serial.printf("Error setting register pointer =%d (%s)",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
}
else { // read from device 
  uint8_t count = Wire.requestFrom(id,20); // sendStop==true, default argument value if not explicitly set
  if(count == 0){ // error condition for requestFrom()
    Serial.printf("Error reading data =%d (%s)",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
  }
else {
 while(Wire.available()){
   Serial.printf("0x%02x ",Wire.read());
}
Serial.println();
}
```
Chuck.

asetyde, 2018-07-19:
[image]
as you can see , light sensor (Sensore Luce) report always same value, there are no debug wire print about strange events.
SHT21 sensor hasn't this issue but code is more recent, I'm not expert of i2c, I not see particular point to change in sensor library 
https://github.com/Ribster/arduino-VEML7700

asetyde, 2018-07-19:
( Also I've tried to restore to 27 June i2c.h/c and wire.cpp/wire.h but came many strange errors on WIFI generic and WIFISTA files .. on complier )
At now I can't ve idea to correct it

stickbreaker, 2018-07-19:
@asetyde I'll look through that library and see if I can find any problem.

Chuck.

asetyde, 2018-07-19:
Many many thanks @stickbreaker !

stickbreaker, 2018-07-19:
@asetyde change this code in the VELM7700.cpp from:
```c++

uint8_t
VEML7700::
receiveData(uint8_t command, uint16_t& data)
{
  Wire.beginTransmission(I2C_ADDRESS);
  if (Wire.write(command) != 1){
    return STATUS_ERROR;
  }
  if (Wire.endTransmission(false)){  // NB: don't send stop here
    return STATUS_ERROR;
  }
  if (Wire.requestFrom(uint8_t(I2C_ADDRESS), uint8_t(2)) != 2){
    return STATUS_ERROR;
  }
  data = Wire.read();
  data |= uint16_t(Wire.read()) << 8;
  return STATUS_OK;
}
```
to:
```c++

uint8_t
VEML7700::
receiveData(uint8_t command, uint16_t& data)
{
  Wire.beginTransmission(I2C_ADDRESS);
  if (Wire.write(command) != 1){
    return STATUS_ERROR;
  }
  if (Wire.endTransmission(false) != 7 ){  // NB: don't send stop here
    return STATUS_ERROR;
  }
  if (Wire.requestFrom(uint8_t(I2C_ADDRESS), uint8_t(2)) != 2){
    return STATUS_ERROR;
  }
  data = Wire.read();
  data |= uint16_t(Wire.read()) << 8;
  return STATUS_OK;
}
```

Chuck.

asetyde, 2018-07-19:
seems not work, value come back 0 , sensor fail

asetyde, 2018-07-19:
Seems worked !!!

asetyde, 2018-07-19:
many many thanks chuck !

stickbreaker, 2018-07-31:
@asetyde I think I found another problem in i2c.

I posted an updated esp32-hal-i2c.c in #1694. Try it.  I think both of these are the same problem.

Chuck.

asetyde, 2018-07-31:
it's fixed for me @stickbreaker !! do you fix on library ?

usmanshahid001, 2018-07-31:
there was an command of Wire.reset() in previous versions it helped me to recover if i2c got struck...but it has been removed...any alternative for that?

asetyde, 2018-07-31:
@usmanshahid001 with Chuck fix it working, about your ask, I can not reply, maybe Chuck !

stickbreaker, 2018-07-31:
@usmanshahid001 I'll see if I can recreate a compatible reset().
@asetyde I'm doing some more testing, I'll create a pr in the next couple of days.

Chuck

asetyde, 2018-07-31:
many thanks !

stickbreaker, 2018-08-01:
@usmanshahid001 the functionality of `Wire.reset()` is incorporated into `Wire.begin()`.  If you need to reset the i2c subsystem just call `Wire.begin()`, it will reset the hardware peripheral, and cycle the bus if necessary.

I'm testing an update that has a couple of changes to `Wire()`:
* `Wire.begin()` is now a `bool` instead of `void`.  It returns `false` if it cannot init the hardware.
* a new function `Wire.busy()` returns the bus state, this will be useful with devices that stretch SCL, or SDA to indicate a 'busy' condition.
* fixes `ReSTART` handling.  Currently it is broken :frowning_face: 

Chuck.

usmanshahid001, 2018-08-02:
@stickbreaker  thanks brother .

stickbreaker, 2018-08-02:
@usmanshahid001 just uploaded these changes into pr #1717. Try it.

Chuck.

copercini, 2018-09-21:
Fixed on https://github.com/espressif/arduino-esp32/pull/1717 =)

closed 2018-09-21 by copercini

asetyde, 2018-10-25:
I reload esp32 library and problem returns .. I think something is not go with fix ?

asetyde, 2018-10-25:
Error reported from sensor 255 , from function -> uint8_t getAutoALSLux(float& lux) -> getAutoXLux    ->

asetyde, 2018-10-25:
WORK with version 1.0.0 rc1 
WORK with version 1.0.0 rc2 
WORK with version 1.0.0 rc3 
NOT WORK with version 1.0.0 rc4 
NOT WORK with version 1.0.0 stable

stickbreaker, 2018-10-25:
@asetyde this is a known problem with Release 1.0.0 it is fixed in the dev branch. See #1962.

Chuck.

asetyde, 2018-10-25:
But I've develop on board manager but I see only 1.0.0 and rc1 rc2 rc3 rc4 .. but no news on  new update library ??

asetyde, 2018-12-05:
@stickbreaker Hi, with 1.01 rc1 Ensor work but I find some freeze of i2C, because I see with microchip cap1296 , touch some times after light sensor work not response for 5 seconds , sometimes for many minutes ! Please do you have an idea about it ?

stickbreaker, 2018-12-05:
@asetyde you can use `Wire.busy()` to see if the i2c bus is hung up.  If it is you can reset it with `Wire.begin()`
```c++
if( Wire.busy() ) {
  bool success = Wire.begin(); // reset i2c hardware, uses previously configured sda,scl pins and frequency.
  if ( !success){
   // if Wire.begin() failed, then the i2c bus will not work. If Wire() could not clear the bus (SDA and SCL high)
// then it cannot configure the i2c peripheral.  until Wire.begin() succeeds, all i2c functionality is dead.
    Serial.printf("i2c Bus reset failed, SCL=%d, SDA=%d \n",digitalRead(SCL),digitalRead(SDA));
    }
```

if you get a failure error code from either `Wire.endTransmission()` or `Wire.requestFrom()` you have to write code to handle the error, either by resetting the bus (`Wire.begin()`) or some other reset function. 

You need to find out what is locking up, the touch sensor, the 'light' sensor or the esp32.  Then either change the circuit to handle the ESD(electro static discharge) and have some method to reset the 'locked up' component.

Chuck.

asetyde, 2018-12-10:
Sensor library VEML770 is killer, but at now I try your code, to investigate

asetyde, 2018-12-16:
Hi , i risolve bug on freezer i2c , ALSAutoLux with algoritm code has a strange bug , freeze i2c . i risolve not using ALS