espressif/arduino-esp32 #1661: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

jmmb392, 2018-07-20:
Dear forum, I am new to Arduino and the ESP32, I am trying to run a very simple program from the book "The Official ESP32 Book". I am running the Arduino IDE 1.8.5 in a Windows 7 machine. The system will remotely control two LEDs connected to an ESP32 DevKitC development board using an HTTP Web Server application. The program is supposed to generate a little webpage with two buttons to turn on/off 2 LEDs that are connected to pins 23 and 22 on the ESP32.  

My setup is the following:

Board: "ESP32 Dev Module"
Flash mode: "DIO"
Flash Frequency: "80 MHz"
Flash Size: "4MB(32Mb)"
PSRAM: "Enabled"
Partition Scheme: "Default"
Upload Speed: "115200"
Core Debug Level: "None"
Port: "COM8"

The program compiles successfully but it does not connect to the network and obviously does not generate the webpage. Surprisingly the same code fully works on a Windows 10 machine. When I check the Serial monitor I get: 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1bdc  PS      : 0x00060a30  A0      : 0x800d1cb4  A1      : 0x3ffb1e60  
A2      : 0x00000000  A3      : 0x3ffb1eaf  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000050  A13     : 0x3ffb9996  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1bdc:0x3ffb1e60 0x400d1cb1:0x3ffb1e80 0x4013ce93:0x3ffb1ea0 0x400d2e90:0x3ffb1ed0 0x400d2ec5:0x3ffb1ef0 0x400d185d:0x3ffb1f10 0x4011fda6:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

```
--------------------------------------

The result that I get from the ESP Exception Decoder is the following:

```
PC: 0x400d1bdc: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiClient.cpp line 100
EXCVADDR: 0x00000008

Decoding stack results
0x400d1bdc: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiClient.cpp line 100
0x400d1cb1: WiFiClient::read(unsigned char*, unsigned int) at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiClient.cpp line 380
0x4013ce93: WiFiClient::read() at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiClient.cpp line 292
0x400d2e90: Stream::timedRead() at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\cores\esp32\Stream.cpp line 36
0x400d2ec5: Stream::readStringUntil(char) at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\cores\esp32\Stream.cpp line 284
0x400d185d: loop() at C:\Users\Marcelo\Documents\Arduino\webserveronoff2/webserveronoff2.ino line 87
0x4011fda6: loopTask(void*) at C:\Users\Marcelo\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 17

```

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32DevKitC
Core Installation/update date:			7/18/2018
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:

The program compiles successfully but it does not connect to the network and obviously does not generate the webpage. Surprisingly the same code fully works on a Windows 10 machine. One difference that I notice from the installation in the Windows 7 and Windows 10, is that the computer with W10, in the tools menu in the Arduino IDE, does not have the option to select anything in PSRAM (enable/disable); whereas the computer with W7 does have the option to select between (enable/disable) in PSRAM. Forum please help me. I am including the sketch code and debug messages.

Thank you.

### Sketch:
```cpp

  /*******************************************************************************************************
 * WEB SERVER TO CONTROL 2 LEDS
 * 
 * This is a simple web server program. 2 LEDs are connected to
 * GPIO ports 23 and 22 of the ESP32 Dev Kit board. The project
 * controls these LEDs (turn ON and OFF) from a web server application.
 * For example, the LEDs can be controlled from any
 * deice that is on the web, for example, a PC, tablet, mobile phone, etc. When 
 * activated a form will appear on the device with buttons
 * and clicking these buttons will control the LEDs. The LEDs are
 * named LED0 and LED1, connected to
 * GPIO ports 23 and 22, respectively.
 * 
 * 
 */

#include "WiFi.h"
#define LED0 23
#define LED1 22

//
// local WiFi name and password
//

const char* ssid = "MyNetwork";
const char* password = "1234AbCd";
WiFiServer server(80);

//
// The HTML code. This code will display two buttons on user's
// device which can be clicked to control the LEDs
//

String html ="<!DOCTYPE html>\
<html>\
<body>\
<center><h1>ESP32 Devkitc LED ON/OFF</h1></center> \
<center><h2>Web Server Example with 2 LEDs</h2></center> \
<form> \
<button name=\"LED0\" button style=\"color:green\" value=\"ON\" type=\"submit\">LED0 ON</button> \
<button name=\"LED0\" button style=\"color:red\" value=\"OFF\" type=\"submit\">LED0 OFF</button> \
<button name=\"LED1\" button style=\"color:green\" value=\"ON\" type=\"submit\">LED1 ON</button> \
<button name=\"LED1\" button style=\"color:red\" value=\"OFF\" type=\"submit\">LED1 OFF</button> \
</form> \
</body> \
</html>";

//
// This function connects the ESP32 to the local WiFi network
// the network name and password are as specified earlier 

void Connect_WiFi()
{
  WiFi.begin(ssid, password);
  while(WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
  }

  
}

void setup() {
  // put your setup code here, to run once:
  // turn OFF LEDs as part of the initiation

  pinMode(LED0, OUTPUT);
  pinMode(LED1, OUTPUT);
  digitalWrite(LED0, LOW);
  digitalWrite(LED1, LOW);
  Connect_WiFi();
  server.begin();
}

void loop() {
  // put your main code here, to run repeatedly:

  WiFiClient client=server.available();
  String request = client.readStringUntil('\r');
  client.flush();

  if(request.indexOf("LED0=ON") != -1)digitalWrite(LED0, HIGH);
  if(request.indexOf("LED0=OFF") != -1)digitalWrite(LED0, LOW); 

  if(request.indexOf("LED1=ON") != -1)digitalWrite(LED1, HIGH);
  if(request.indexOf("LED1=OFF") != -1)digitalWrite(LED1, LOW); 

//
// The HTML page to be displayed on user's device
//

  client.print(html);

}

```

### Debug Messages:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.1.207, MASK: 255.255.255.0, GW: 192.168.1.254
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1d1c  PS      : 0x00060a30  A0      : 0x800d1df7  A1      : 0x3ffb1e60  
A2      : 0x00000000  A3      : 0x3ffb1eaf  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000050  A13     : 0x3ffb9996  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1d1c:0x3ffb1e60 0x400d1df4:0x3ffb1e80 0x4013d55b:0x3ffb1ea0 0x400d3540:0x3ffb1ed0 0x400d3575:0x3ffb1ef0 0x400d196d:0x3ffb1f10 0x4012046e:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

```

lbernstone, 2018-07-20:
You need to test to see if client exists before you try to use any methods on it.  Take a look at the [SimpleWebServer](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/SimpleWiFiServer) example.  Please note, this is a forum for issues with the codebase, not problems with your code.

romankc, 2018-12-17:
> You need to test to see if client exists before you try to use any methods on it. Take a look at the [...]

I find it most interesting that jmmb392, using the identical code, has much different results on two different Windows machines. If this was a problem with jmmb392's code, as you imply, then it would fail in both cases. There is something else going on here.

closed 2019-08-15 by stale[bot]