espressif/arduino-esp32 #1670: WiFi + MCP2515 (CAN over SPI) = nightmare

psykokwak-com, 2018-07-24:
### Hardware:
Board: Custom board based on ESP32 Wrower module
Core Installation/update date:			IDF & Arduino up to date
IDE name:							IDF Component (mingw)
Flash Frequency:						40Mhz
Upload Speed:						115200

### Description:
I try to make the WiFi and my CAN (over SPI) communication working together but I have no succes :/
If I comment the "WiFi.softAP()" line, the CAN link works well.
If I uncomment this line, I have read and write error and after few seconds (random time), the ESP32 crashs.

I use this CAN library https://github.com/coryjfowler/MCP_CAN_lib but I tried with this other one https://github.com/sandeepmistry/arduino-CAN and I had the same results.
I tried to build the sketch with the ArduinoIDE... The same.

I have no more idea. Is it possible to have a conflict between the SPI and WiFi link or a memory leak on the WiFi stack ?

Here the link with the mcp_can library (I changed it a bit) : 
[modified_mcp_can.zip](https://github.com/espressif/arduino-esp32/files/2224514/modified_mcp_can.zip)

No need to have CAN device attached to.

### Sketch:
```cpp

#include <Arduino.h>
#include <SPI.h>
#include <WiFi.h>
#include <WiFiAP.h>

#include "mcp_can.h"

#define VSPI_CLOCK_PIN        4
#define VSPI_MOSI_PIN         15
#define VSPI_MISO_PIN         32
#define VSPI_CS_MCP2515_PIN   25

#define CANBUFFERSIZE         8
#define CAN_INT_MCP2515_PIN   4

#define DAMACCANID            0x00A

#define myTAG                 "SPIandWiFI"

void setup()
{
  esp_log_level_set("*", ESP_LOG_VERBOSE);

  SPIClass mffVSPI = SPIClass(VSPI);
  mffVSPI.begin(VSPI_CLOCK_PIN, VSPI_MISO_PIN, VSPI_MOSI_PIN);
  //pinMode(VSPI_MOSI_PIN, INPUT_PULLUP);
  //pinMode(VSPI_MISO_PIN, INPUT_PULLUP);

  MCP_CAN mffCAN(VSPI_CS_MCP2515_PIN);
  mffCAN.begin(mffVSPI, MCP_ANY, CAN_250KBPS, MCP_8MHZ);
  mffCAN.setMode(MCP_NORMAL);

  byte dataOperationMode[] = { 0x01, 0x00 };
  mffCAN.sendMsgBuf(0x00, 2, dataOperationMode);

  byte dataSetDisplay[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08 };
  mffCAN.sendMsgBuf((0x20 << 4) + DAMACCANID, 8, dataSetDisplay);

  // If I comment this line, the CAN bus works well.
  WiFi.softAP("WIFI Sucks", "fuckingWiFi");

  unsigned long id = 0;
  byte len = CANBUFFERSIZE;
  byte buffer[CANBUFFERSIZE];

[... 185 lines cut ...]

A2      : 0x00008100  A3      : 0x3ffae944  A4      : 0x00000050  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80082e5a  A9      : 0x3ffc04b0
A10     : 0x3ffbe434  A11     : 0x00000004  A12     : 0x3ffc0734  A13     : 0x3ffae944
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00008118  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

Backtrace: 0x40135bb2:0x3ffc07f0 0x400f0301:0x3ffc0810 0x400ea79e:0x3ffc0840 0x400eafa5:0x3ffc0870 0x400ea4f4:0x3ffc0890 0x400e2b0f:0x
3ffc0920
0x40135bb2: spiGetClockDiv at C:/msys32/home/Jyce/esp/arduino/components/arduino/cores/esp32/esp32-hal-spi.c:290

0x400f0301: SPIClass::beginTransaction(SPISettings) at C:/msys32/home/Jyce/esp/arduino/components/arduino/libraries/SPI/src/SPI.cpp:26
5

0x400ea79e: MCP_CAN::mcp2515_readStatus() at C:/msys32/home/Jyce/esp/arduino/main/mcp_can.cpp:998

0x400eafa5: MCP_CAN::checkReceive() at C:/msys32/home/Jyce/esp/arduino/main/mcp_can.cpp:998

0x400ea4f4: setup() at C:/msys32/home/Jyce/esp/arduino/main/SentinelSimpleFirmwareMain.cpp:79

0x400e2b0f: loopTask(void*) at C:/msys32/home/Jyce/esp/arduino/components/arduino/cores/esp32/main.cpp:15

Rebooting...

```

collin80, 2018-07-31:
Here's what I can tell you, it isn't the ESP32 or Arduino-esp32. When I use my own MCP2515 library on the ESP32 I can then connect over wifi and through a TCP/IP port I can saturate the CAN bus at 500k and stream all the traffic to a connected computer without anything crashing. So, the SPI and WiFi libraries work perfectly fine together. It has to be the MCP2515 libraries you're using.

psykokwak-com, 2018-07-31:
Hi Collin80,
Thanks for your answer. I will check on the hardware side of my board.
Is it possible for you to share your mcp2515 library ?
Which IDF and Arduino library version do you use ?
Thanks for your help.

psykokwak-com, 2018-08-02:
Ok, It's an hardware issue. My code works well with a simple ESP32 and an MCP2515 modules.
I will take a look on the hardware side.
Thanks.

closed 2018-08-02 by psykokwak-com

psykokwak-com, 2018-08-14:
Just for information because I find the error : 
I used an MCP2515 in 5V and the ESP32 bus is 3.3V. 
Now I use the MCP2515 in 3.3V with a MCP2562 tranceiver and everything work well!

brandonros, 2019-11-07:
Why did you need both MCP2515 and MCP2562?

dpharris, 2019-11-07:
Why not use the built-in CAN controller?

David

brandonros, 2021-01-05:
I finally got around to getting this working.

* MELIFE 2 Pack ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with ESP32s Antenna RF AMP Filter AP STA for Arduino IDE: https://www.amazon.com/gp/product/B07Q576VWZ

* Comidox 3Pcs MCP2515 CAN Bus Module TJA1050 Receiver SPI Module for Arduino 51 MCU ARM Controller Development Board: https://www.amazon.com/Comidox-MCP2515-Receiver-Controller-Development/dp/B07J9KZ4L4

I included wiring coloring I pulled out of thin air below. I apologize in advance if they don't follow some common spec I should know.

Format: MPC2515 shield -> ESP32 DEVKIT V1 pin
---
* VCC -> 3v3 (red)
* GND -> GND (black)
* INT -> GPIO16 (purple, pick any random GPIO unused; any GPIO can be INT apparently?)
* SI -> GPIO23 (green, VSPI MOSI)
* SO -> GPIO19 (white, VSPI MOSO)
* SCK -> GPIO18 (orange, VSPI CLK)
* CS -> GPIO5 (blue, VSPI CS0)

```c++
#include <mcp_can.h>
#include <mcp_can_dfs.h>
#include <SPI.h>
#include <iso-tp.h>
#include <WiFi.h>
#include <WebSocketsServer.h>

#define MCP_CS 5 // GPIO5 = VSPI CS0
#define MCP_INT 16 // GPIO16

WiFiServer webServer(8080);
WebSocketsServer socketServer = WebSocketsServer(8081);
MCP_CAN CAN0(MCP_CS);
IsoTp isotp(&CAN0, MCP_INT);

struct Message_t rxMsg;

void webSocketEvent(uint8_t num, WStype_t type, uint8_t* payload, size_t length) {
  switch (type) {
    case WStype_DISCONNECTED:
    case WStype_CONNECTED:
    case WStype_TEXT:  
    case WStype_BIN:
    case WStype_ERROR:
    case WStype_FRAGMENT_TEXT_START:
    case WStype_FRAGMENT_BIN_START:
    case WStype_FRAGMENT:
    case WStype_FRAGMENT_FIN:
    break;
  }
}

void setup()
{
  // serial
  Serial.begin(115200);
  // interrupt
  pinMode(MCP_INT, INPUT);
  // CAN
  CAN0.begin(MCP_ANY, CAN_500KBPS, MCP_8MHZ);
  CAN0.setMode(MCP_NORMAL);
  // buffers
  rxMsg.Buffer = (uint8_t *)calloc(MAX_MSGBUF, sizeof(uint8_t));
  // wifi
  WiFi.softAP("ssid", "password");
  // socket server
  socketServer.onEvent(webSocketEvent);
  socketServer.begin();
  // web server
  webServer.begin();
}

void loop()
{
  // socket server
  socketServer.loop();
  // web server
  WiFiClient client = webServer.available();
  if (client) {
    String buffer = "";
    while (client.connected()) {
      while (client.available()) {
        buffer += client.read(); 
      }
      client.println("HTTP/1.1 200 OK");
      client.println("Content-type: text/html");
      client.println();
      client.print("Click <a href=\"/H\">here</a> to turn the LED on pin 5 on.<br>");
      client.println();
      client.stop();
      break;
    }
  }
  // CAN
  if (digitalRead(MCP_INT) == 0) {
    rxMsg.tx_id = 0x7ED;
    rxMsg.rx_id = 0x7E5;
    Serial.println(F("Receive..."));
    isotp.receive(&rxMsg);
    isotp.print_buffer(rxMsg.rx_id, rxMsg.Buffer, rxMsg.len);
  }
}
```

I had to fork and fix the `iso-tp` library here: https://github.com/altelch/iso-tp/pull/11

I `git clone`d that in C:\Users\USERNAME\Documents\Arduino\libraries. I tried Mac but the CP210x driver for the Silicon Labs chip for Mac OS X 10.15.7 Catalina just seems 100% broken.

I used Arduino IDE 1.8.13. I added arduino-esp32 to Board Manager.

psykokwak-com, 2021-01-05:
> Why did you need both MCP2515 and MCP2562?

One is a SPI <> CAN controller. The other one is a CAN transceiver. The two are needed for the operation.

> Why not use the built-in CAN controller? David

Because of my electric wiring that already use the ESP32 pins.

pipehappy1, 2021-05-13:
Come across this, but the very likely reason we need MCP2515 is esp32 has one and only one CAN controller.