espressif/arduino-esp32 #1673: 'String' does not name a type on base64.h

Jeard, 2018-07-25:
Hardware:
Board:							WEMOS LOLIN32
Core Installation/update date:			25/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						512000

Description:
Gives `base64.h:8:12: error: 'String' does not name a type` and `base64.h:9:26: error: 'uint8_t' has not been declared`  when called the `static String base64::encode(uint8_t * data, size_t length)` function.

Including `arduino.h` or `WString.h` and `stdint.h` on `base64.h` fix the error.

lbernstone, 2018-07-25:
Can you give sample code?  This works for me:
```
#include <base64.h>
base64 test;
void setup() {
  test.encode("Blahblahblah");
}
void loop() {}
```

Jeard, 2018-07-25:
@lbernstone : Could please check the below library and its example which I created.
[https://github.com/Jeard/esp32aes128](url)

Jeard, 2018-07-26:
I have tested  the code with `WString.h` and `stdint.h` included on the `ESP32AESB64.h` but still gives the same error.

lbernstone, 2018-07-26:
It works fine for me when I include those.  Are you including them before the library that requires them?

huilgolAni19, 2019-05-25:
@Jeard update the link for the modified version of base64.h and base64.cpp files