espressif/arduino-esp32 #1681: Library import request EM4095 RFID chip

asetyde, 2018-07-26:
The EM4095 is replacement IC of the U2270B Atmel reader chip. EM4095 is a CMOS integrated transceiver circuit for RFID applications working with transponders at a frequency of typically 125 kHz. It integrates a PLL system to achieve self-adaptive carrier frequency to antenna resonant frequency and can communicate with a microprocessor via a simple interface.
New Texas chip use 134,2mhz is difficult to find, specially tag, therefore for best support for all most diffused tag at 125mhz is useful create this library.
I want import ad give to this community this library for EM4095 for esp32, that library came from esp8266 I hope it can be easily imported and integrated.
Someone can be help me to do or evaluate integration ? 

https://github.com/Crypter/EM4095

asetyde, 2018-08-06:
Hi, this is my first working code, it's working on Arduino mega 2560, someone can help me why on esp32 not work ? pin are same

asetyde, 2018-08-06:
#include "Arduino.h"

#define DELAYVAL    320   //384 //standard delay for manchster decode
#define TIMEOUT     1000  //standard timeout for manchester decode

    int demodOut=27;
    int shd=4;
    int mod=5;
    int rdyClk=19;
byte tagData[5]; //Holds the ID numbers from the tag  

//Manchester decode. Supply the function an array to store the tags ID in
bool decodeTag(unsigned char *buf)
{
  unsigned char i = 0;
  unsigned short timeCount;
  unsigned char timeOutFlag = 0;
  unsigned char row, col;
  unsigned char row_parity;
  unsigned char col_parity[5];
  unsigned char dat;
  unsigned char j;
  
  while(1)
  {
    timeCount = 0;
    while(0 == digitalRead(demodOut)) //watch for demodOut to go low
    {

      if(timeCount >= TIMEOUT) //if we pass TIMEOUT milliseconds, break out of the loop
      {
        break;
      }
      else
      {
        timeCount++;
      }
    }

    if (timeCount >= 600)
    {
      return false;
    }
    timeCount = 0;

    delayMicroseconds(DELAYVAL);
    if(digitalRead(demodOut))
    {

		
      for(i = 0; i < 8; i++) // 9 header bits
      {
        timeCount = 0; //restart counting
        while(1 == digitalRead(demodOut)) //while DEMOD out is high
        {
          if(timeCount == TIMEOUT)
          {
            timeOutFlag = 1;
            break;
          }
          else
          {
            timeCount++;
          }
        }

        if(timeOutFlag)
        {
          break;
        }
[... 188 lines cut ...]
      }

    void  loop(){
        
        
          //scan for a tag - if a tag is sucesfully scanned, return a 'true' and proceed
  if(scanForTag(tagData) == true)
  {
    Serial.print("RFID Tag ID:"); //print a header to the Serial port.
    //loop through the byte array
    for(int n=0;n<5;n++)
    {
      Serial.print(tagData[n],DEC);  //print the byte in Decimal format
      if(n<4)//only print the comma on the first 4 nunbers
      {
        Serial.print(",");
      }
    }
    Serial.print("\n\r");//return character for next line
        
        
  }
        
        
      }

asetyde, 2018-08-06:
`

#include "Arduino.h"

#define DELAYVAL    320   //384 //standard delay for manchster decode
#define TIMEOUT     1000  //standard timeout for manchester decode

    int demodOut=27;
    int shd=4;
    int mod=5;
    int rdyClk=19;
byte tagData[5]; //Holds the ID numbers from the tag  

//Manchester decode. Supply the function an array to store the tags ID in
bool decodeTag(unsigned char *buf)
{
  unsigned char i = 0;
  unsigned short timeCount;
  unsigned char timeOutFlag = 0;
  unsigned char row, col;
  unsigned char row_parity;
  unsigned char col_parity[5];
  unsigned char dat;
  unsigned char j;
  
  while(1)
  {
    timeCount = 0;
    while(0 == digitalRead(demodOut)) //watch for demodOut to go low
    {

      if(timeCount >= TIMEOUT) //if we pass TIMEOUT milliseconds, break out of the loop
      {
        break;
      }
      else
      {
        timeCount++;
      }
    }

    if (timeCount >= 600)
    {
      return false;
    }
    timeCount = 0;

    delayMicroseconds(DELAYVAL);
    if(digitalRead(demodOut))
    {

		
      for(i = 0; i < 8; i++) // 9 header bits
      {
        timeCount = 0; //restart counting
        while(1 == digitalRead(demodOut)) //while DEMOD out is high
        {
          if(timeCount == TIMEOUT)
          {
            timeOutFlag = 1;
            break;
          }
          else
          {
            timeCount++;
          }
        }

        if(timeOutFlag)
        {
[... 192 lines cut ...]
    void  loop(){
        
        
          //scan for a tag - if a tag is sucesfully scanned, return a 'true' and proceed
  if(scanForTag(tagData) == true)
  {
    Serial.print("RFID Tag ID:"); //print a header to the Serial port.
    //loop through the byte array
    for(int n=0;n<5;n++)
    {
      Serial.print(tagData[n],DEC);  //print the byte in Decimal format
      if(n<4)//only print the comma on the first 4 nunbers
      {
        Serial.print(",");
      }
    }
    Serial.print("\n\r");//return character for next line
        
        
  }
        
        
      }

`

asetyde, 2018-08-06:
> 

#include "Arduino.h"

#define DELAYVAL    320   //384 //standard delay for manchster decode
#define TIMEOUT     1000  //standard timeout for manchester decode

    int demodOut=27;
    int shd=4;
    int mod=5;
    int rdyClk=19;
byte tagData[5]; //Holds the ID numbers from the tag  

//Manchester decode. Supply the function an array to store the tags ID in
bool decodeTag(unsigned char *buf)
{
  unsigned char i = 0;
  unsigned short timeCount;
  unsigned char timeOutFlag = 0;
  unsigned char row, col;
  unsigned char row_parity;
  unsigned char col_parity[5];
  unsigned char dat;
  unsigned char j;
  
  while(1)
  {
    timeCount = 0;
    while(0 == digitalRead(demodOut)) //watch for demodOut to go low
    {

      if(timeCount >= TIMEOUT) //if we pass TIMEOUT milliseconds, break out of the loop
      {
        break;
      }
      else
      {
        timeCount++;
      }
    }

    if (timeCount >= 600)
    {
      return false;
    }
    timeCount = 0;

    delayMicroseconds(DELAYVAL);
    if(digitalRead(demodOut))
    {

		
      for(i = 0; i < 8; i++) // 9 header bits
      {
        timeCount = 0; //restart counting
        while(1 == digitalRead(demodOut)) //while DEMOD out is high
        {
          if(timeCount == TIMEOUT)
          {
            timeOutFlag = 1;
            break;
          }
          else
          {
            timeCount++;
          }
        }

        if(timeOutFlag)
        {
[... 192 lines cut ...]
    void  loop(){
        
        
          //scan for a tag - if a tag is sucesfully scanned, return a 'true' and proceed
  if(scanForTag(tagData) == true)
  {
    Serial.print("RFID Tag ID:"); //print a header to the Serial port.
    //loop through the byte array
    for(int n=0;n<5;n++)
    {
      Serial.print(tagData[n],DEC);  //print the byte in Decimal format
      if(n<4)//only print the comma on the first 4 nunbers
      {
        Serial.print(",");
      }
    }
    Serial.print("\n\r");//return character for next line
        
        
  }
        
        
      }

>

asetyde, 2018-08-06:
(I try to format but not work !)

asetyde, 2018-08-06:
this is esp8266 adapted , but only work on esp8266 library and 160mhz 

`#include <ESP8266WiFi.h>

static inline int32_t asm_ccount(void) {
    int32_t r; asm volatile ("rsr %0, ccount" : "=r"(r)); return r; }

static inline void asm_nop() {
    asm volatile ("nop"); }

static uint64_t data=0;
static uint8_t  state=0, ostate=0, state_bit=0;

static uint32_t tick=0, tock=0, dtick=0;
static uint32_t first_data=0, second_data=0;
static uint8_t counter=0;

uint8_t getVendor(uint64_t RAW){
  return ((RAW >> 47) & 0xF0) | ((RAW >> 46) & 0xF);
}

uint32_t getID(uint64_t RAW){
  uint32_t result=0;
  RAW = RAW >> 1; //compensate for stop bit
  for (int i=0; i< 10; i++)
  {
    RAW = RAW >> 5;
    result |= (RAW & 0xF) << (i*4);
  }
  return result;
}

void setup() {
  pinMode(D1, INPUT);
  pinMode(BUILTIN_LED, OUTPUT);
  digitalWrite(BUILTIN_LED, HIGH);
//  pinMode(D2, INPUT);
  WiFi.forceSleepBegin();
//  ESP.wdtDisable();
  Serial.begin(9600);
  delay(10);
  Serial.println();
}

void loop() {
  for (;;) {
    if (!counter) yield();
    counter++;
    state = GPI&0b00100000;
    
   if (state!=ostate){
      digitalWrite(BUILTIN_LED, HIGH);

      tick=asm_ccount();
      dtick=tick-tock;
//      Serial.print((uint16_t)state>>5);
//      Serial.print(", ");
//      Serial.print(dtick);
//      Serial.print(", ");
      dtick = (dtick - 20000) / 40000;
//      Serial.println(dtick);

      
      if (dtick) { //+60000
        state_bit = (ostate >> 5);
        data = data<<1;
        data |= state_bit;
      } else if ( state&0b00100000 ) { //-60000
        data = data<<1;
        data |= state_bit;
      }

      first_data = data>>32;
      second_data = data&0xFFFFFFFF;

      if ( ((first_data&0xFF800000) == 0xFF800000) && !(second_data&1) ){
        if (
          !(__builtin_popcountll(data&0x7C000000000000) % 2) &&
          !(__builtin_popcountll(data&0x3E00000000000) % 2) &&
          !(__builtin_popcountll(data&0x1F0000000000) % 2) &&
          !(__builtin_popcountll(data&0xF800000000) % 2) &&
          !(__builtin_popcountll(data&0x7C0000000) % 2) &&
          !(__builtin_popcountll(data&0x3E000000) % 2) &&
          !(__builtin_popcountll(data&0x1F00000) % 2) &&
          !(__builtin_popcountll(data&0xF8000) % 2) &&
          !(__builtin_popcountll(data&0x7C00) % 2) &&
          !(__builtin_popcountll(data&0x3E0) % 2) &&
          
          !(__builtin_popcountll(data&0x42108421084210) % 2) &&
          !(__builtin_popcountll(data&0x21084210842108) % 2) &&
          !(__builtin_popcountll(data&0x10842108421084) % 2) &&
          !(__builtin_popcountll(data&0x8421084210842) % 2)
          ){
            digitalWrite(BUILTIN_LED, LOW);
            Serial.printf("RAW: %08X%08X Ven: %03u UID: ", first_data, second_data, getVendor(data));
            uint32_t uid = getID(data);
            char buffer[100];
            sprintf(buffer, "%04ld", uid/1000000L);
            Serial.print(buffer); 
            sprintf(buffer, "%05ld", uid%1000000L);
            Serial.println(buffer);
          }
      }

      
      ostate=state;
      tock=tick;
    }
  }
//  Serial.printf("%d,%d\r\n", digitalRead(D1), digitalRead(D2));
}`

lbernstone, 2018-08-06:
Post it to your own github repository.  You can ask Crypter to link to your library from his page.  This repository is for the arduino core functionality, extended with esp32 specific features.

asetyde, 2018-08-06:
https://github.com/asetyde/EM4095   create a branch , enjoy it

closed 2018-08-06 by asetyde

csc-techgroup, 2022-07-02:
Use this Library on ESP32 i get an ID, that part is working, but ID is different than the hardware-id given by other readers... I'am using Mikroe 262 readerboard. I know that project is old, but i hope you can still help me :)