espressif/arduino-esp32 #1682: ESP32 resetting

yldzmuhammed, 2018-07-27:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

Guru is not working on my Windows 10 computer.
----------------------------- Remove above -----------------------------

### Hardware:
Board:							NodeMCU - ESP32
Core Installation/update date:		Updated today
IDE name:						Arduino IDE & Platform.io. Same results for both IDE
Flash Frequency:					80Mhz
Upload Speed:					912600

### Description:
When esp32 try to connect a ble peripheral device its rebooting. Below copied the serial port messages.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084

load:0x40080000,len:7944
entry 0x40080310
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ViziSensor, Address: c3:f5:fe:bd:f3:c5, appearance: 0, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb
Found our device!  address: Forming a connection to c3:f5:fe:bd:f3:c5
 - Created client
 - Connected to server
 - Found our service
[E][BLERemoteCharacteristic.cpp:315] retrieveDescriptors():
abort() was called at PC 0x401846df on core 1

Backtrace: 0x40092d50:0x3ffd59b0 0x40092f53:0x3ffd59d0 0x401846df:0x3ffd59f0 0x40184726:0x3ffd5a10 0x401721bf:0x3ffd5a30 0x400d44dd:0x3ffd5a50 0x400d1a12:0x3ffd5ab0 0x400d1bc5:0x3ffd
5b00 0x4018c29c:0x3ffd5b50

Rebooting...

### Sketch:
/**
 * A BLE client example that is rich in capabilities.
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("0000180f-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID(BLEUUID((uint16_t)0x2A6E));

static bool notification = false;
static BLEAddress *pServerAddress;
static bool doConnect = false;
static bool connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;

BLEScan* pBLEScan;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());

    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      //
      Serial.print("Found our device!  address: ");
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      doConnect = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    if(notification == false){
      Serial.println("Turning notification on");
      const uint8_t onPacket[] = {0x01, 0x00};
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2A6E))->writeValue((uint8_t*)onPacket, 2, true);
      notification = false;
    }
  }
  else
    pBLEScan->start(60);

  delay(1000); // Delay a second between loops.
} // End of loop

### Debug Messages:
Debug options not visible on my Arduino IDE.

bouhady, 2018-10-14:
Same problem here.
 my stack trace : 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3e47  PS      : 0x00060730  A0      : 0x800d1530  A1      : 0x3ffd5bd0  
A2      : 0x00000000  A3      : 0x00000014  A4      : 0x00000024  A5      : 0x00000024  
A6      : 0x00000000  A7      : 0x3ffe819c  A8      : 0x800d5bdc  A9      : 0x3ffd5bb0  
A10     : 0x00000002  A11     : 0x3f4015eb  A12     : 0x00000024  A13     : 0x00000001  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000034  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d3e47:0x3ffd5bd0 0x400d152d:0x3ffd5bf0 0x400d16f1:0x3ffd5c60 0x40188108:0x3ffd5cb0

chegewara, 2018-10-14:
You have to remember it is just example. What i am suspecting that in some circumstances peer devices is getting disconnected between getService() and getCharacteristic(). What you can do is to surround getCharacteristic() with `if(pClient->isConnected()`.

Other possible situation is that esp32 cant find characteristic on peer device for few reasons. 

Also it can be related to issue with BLEUUID class, but i would check the other two.

bouhady, 2018-10-14:
Yes, I edited the example for a certain device and connected to it. (it's xiaomi scooter and its making sound on connection.)
I manage to get the list of services and isolated the main one. (exactly according the tree was discovered by "nRF connect" app).
The get Characteristics, or even the specific characteristic by uuid all makes the device to reboot. 😔

chegewara, 2018-10-14:
If you post logs i will try to help, maybe piece of code where you are trying to get characteristic and sceenshot from nRF connect with all listed characteristics that you are interested to connect to.

bouhady, 2018-10-14:
How do I make the internal logs of the library to show in the arduino terminal?

chegewara, 2018-10-14:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#switching-on-debugging

bouhady, 2018-10-15:
thanks, I'll try soon and report.

bouhady, 2018-10-25:
ok, I've tried and still no special info the logs are still crashing immediately when calling "pRemoteService-> getCharacteristics()"
code : 

```
std::map<std::string, BLERemoteService*>* pServices = pClient->getServices();
  Serial.println("Services : " + String(pServices->size()));
  for (std::map<std::string, BLERemoteService*>::iterator it = pServices->begin(); it != pServices->end(); ++it) {
    Serial.println(it->first.c_str());
  }
  BLERemoteService* pRemoteService = pServices->find(serviceUUIDstr)->second;
  Serial.println(" - Found our service");

  std::map<std::string, BLERemoteCharacteristic*>* pCharacteristics = pRemoteService-> getCharacteristics();
  Serial.println(" - Found our BLERemoteCharacteristic");

```
> Services : 4
> 00001800-0000-1000-8000-00805f9b34fb
> 00001801-0000-1000-8000-00805f9b34fb
> 0000fe95-0000-1000-8000-00805f9b34fb
> 6e400001-b5a3-f393-e0a9-e50e24dcca9e
>  - Found our service
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> Core 1 register dump:
> PC      : 0x4000c29b  PS      : 0x00060730  A0      : 0x800d56ee  A1      : 0x3ffd5cf0  
> A2      : 0x3ffd5d01  A3      : 0x0000008f  A4      : 0x00000013  A5      : 0x3ffd5d01  
> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
> A10     : 0x00000041  A11     : 0x00000055  A12     : 0x00000001  A13     : 0x00000000  
> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x0000008f  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x4000c29b:0x3ffd5cf0 0x400d56eb:0x3ffd5d00 0x400d1c89:0x3ffd5d50 0x400d1e4d:0x3ffd5dc0 0x4018bb4c:0x3ffd5e10
> 

[image]
[image]

beegee-tokyo, 2018-10-25:
What is your **`serviceUUIDstr`**? Is it in the list of the found services?
Because in your first post I read **`static BLEUUID serviceUUID("0000180f-0000-1000-8000-00805f9b34fb");`** which is none of the 4 services.

You should check that **`pRemoteService`** is **_NOT_** null before using it to retrieve characteristics.

bouhady, 2018-10-25:
Oh! You are right!!! It was NPE!!! (I made a mistake with the serviceUUIDstr value! and forgot to do null check!
Thanks , Its working now!

epanta, 2019-01-17:
Good morning,
I have a similar problem. When I try to print the value value.c_str().. it reboot

BLE Advertised Device found: Name: BLE Footpod, Address: 00:18:8c:34:10:cd, appearance: 1088, serviceUUID: 00001814-0000-1000-8000-00805f9b34fb, txPower: -2
Found our device!  address: Forming a connection to 00:18:8c:34:10:cd
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1831  PS      : 0x00060130  A0      : 0x8018d437  A1      : 0x3ffca190  
A2      : 0x008c1800  A3      : 0x3ffc5ecc  A4      : 0x0000cd00  A5      : 0x3ffc4ddd  
A6      : 0x3ffc4ddc  A7      : 0x3ffba2c4  A8      : 0x800d182a  A9      : 0x3ffca140  
A10     : 0x3ffc4e88  A11     : 0x4088b990  A12     : 0x2d2a5589  A13     : 0x3ffc5ecc  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x008c1a80  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d1831:0x3ffca190 0x4018d434:0x3ffca1d0 0x40092ddd:0x3ffca1f0

Rebooting...

bouhady, 2019-01-17:
@epanta check that value is probably null.

stiebel, 2019-04-09:
Hi,
I have a similar problem, too:
arduino 1.8.7 on a Ubuntu18.04
BLE Version 1.01 from Neil Kolban

The BLE server is a joystick.
I use the original BLE_client program, and get the serviceUUID, then change line 12 to:
static BLEUUID serviceUUID("00001812-0000-1000-8000-00805f9b34fb");

after rebooting I get a Core 1 panic.

Then I changed line 14 to:
static BLEUUID charUUID(BLEUUID((uint16_t)0x2A4A));

which shoul work as it is one of the UUIDs I get from "nRF connect".

But still I get the Core 1 panic.

He is the lo (debug switched to info)
`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: HZ-2749, Address: ff:ff:70:04:43:5e, appearance: 963, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb
Forming a connection to ff:ff:70:04:43:5e
 - Created client
[I][BLEDevice.cpp:596] addPeerDevice(): add conn_id: 0, GATT role: client
 - Connected to server
[I][BLEDevice.cpp:255] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
 - Found our service
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d5b60  PS      : 0x00060930  A0      : 0x800d1aac  A1      : 0x3ffd5c10  
A2      : 0x00000000  A3      : 0x3ffea8e8  A4      : 0x002a7f4c  A5      : 0x782ddcd2  
A6      : 0x3ffed06c  A7      : 0x0001beb5  A8      : 0x800d5b58  A9      : 0x3ffd5bf0  
A10     : 0x3ffd5c3c  A11     : 0x3ffed06c  A12     : 0x00000024  A13     : 0x782ddcd2  
A14     : 0x3ffd5bf0  A15     : 0x0000002a  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x002a7f4c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d5b60:0x3ffd5c10 0x400d1aa9:0x3ffd5c70 0x400d1b27:0x3ffd5cc0 0x4018a7c4:0x3ffd5d30

Rebooting...

`
Any ideas?

bouhady, 2019-04-09:
Check nulls. Thats what I had (the service was null)

stiebel, 2019-04-09:
Ok, this is a sniplet of the code:
    ```Serial.print(" - Found our service: ");
    Serial.println(serviceUUID.toString().c_str());

    Serial.print(" - Checking for       ");
    Serial.println(charUUID.toString().c_str());
   
    if(pClient->isConnected()) pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
```
and this is the result:
```
...
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: , Address: 7a:db:e0:86:30:16, manufacturer data: 4c0010050318223806
BLE Advertised Device found: Name: HZ-2749, Address: ff:ff:70:04:43:5e, appearance: 963, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb
Forming a connection to ff:ff:70:04:43:5e
 - Created client
[I][BLEDevice.cpp:596] addPeerDevice(): add conn_id: 0, GATT role: client
 - Connected to server
[I][BLEDevice.cpp:255] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
 - Found our service: 00001812-0000-1000-8000-00805f9b34fb
 - Checking for       00002a4e-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d5bb0  PS      : 0x00060930  A0      : 0x800d1b01  A1      : 0x3ffd5ca0  
A2      : 0x3ffead50  A3      : 0x3ffeaf94  A4      : 0x002a7f4c  A5      : 0x2d2ef829  
A6      : 0x3ffed1b0  A7      : 0x00010000  A8      : 0x800d5ba8  A9      : 0x3ffd5c80  
A10     : 0x3ffd5ccc  A11     : 0x3ffed1b0  A12     : 0x00000024  A13     : 0x2d2ef829  
A14     : 0x3ffd5c80  A15     : 0x0000002a  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x002a7f4c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d5bb0:0x3ffd5ca0 0x400d1afe:0x3ffd5d00 0x400d1c27:0x3ffd5d50 0x4018a814:0x3ffd5dc0

Rebooting...
```
Does not look like npe to me...
Please tell me, that I'm wrong.
null would be the easiest to fix I guess.

chegewara, 2019-04-09:
```
Serial.print(" - Found our service: "); 
Serial.println(serviceUUID.toString().c_str()); 
Serial.print(" - Checking for "); 
Serial.println(charUUID.toString().c_str()); 
if(pClient->isConnected()) 
pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID); // <-- pRemoteService can be null
```
You need to check if pRemoteServce is null before you call last line.

stiebel, 2019-04-10:
hi chegewara,
I did not want to confuse with too much code, sorry.
Of course I did not change the original program and it still checks pRemoteService:
```
BLERemoteService*` pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    
    Serial.print(" - Found our service: ");
    Serial.println(serviceUUID.toString().c_str());

    Serial.print(" - Checking for       ");
    Serial.println(charUUID.toString().c_str());
   
    Serial.print(" - Checking for       ");
    Serial.println(charUUID.toString().c_str());
    
    if(pClient->isConnected()) pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
```
anyway, thanks for helping.

Isn't there a chance to have a program checking for characteristics just like checking for services in the first place?

Roland

stiebel, 2019-04-10:
I just switched debug level to verbose and got more information, but I do not really understand how to get the important information out of this...

[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Checking for       00002a4e-0000-1000-8000-00805f9b34fb
 - Checking for       00002a4e-0000-1000-8000-00805f9b34fb
[D][BLERemoteService.cpp:168] retrieveCharacteristics(): >> getCharacteristics() for service: 00001812-0000-1000-8000-00805f9b34fb
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 17, UUID: 00002a4e-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 17 0x17, uuid: 00002a4e-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4e-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 19, UUID: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 19 0x19, uuid: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 20, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 21, UUID: 00002908-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 2 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 23, UUID: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 23 0x23, uuid: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 24, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 25, UUID: 00002908-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 2 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 27, UUID: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 27 0x27, uuid: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 28, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 29, UUID: 00002908-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 2 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 31, UUID: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 31 0x31, uuid: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 32, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 33, UUID: 00002908-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 2 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 35, UUID: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 35 0x35, uuid: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a4d-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 36, UUID: 00002908-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[... cut ...]
A2      : 0x3ffeac58  A3      : 0x3ffeae9c  A4      : 0x002a7f4c  A5      : 0x234fd49e  
A6      : 0x3ffed0b8  A7      : 0x00010000  A8      : 0x800d6e90  A9      : 0x3ffd5c80  
A10     : 0x3ffd5ccc  A11     : 0x3ffed0b8  A12     : 0x00000024  A13     : 0x234fd49e  
A14     : 0x3ffd5c80  A15     : 0x0000002a  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x002a7f4c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d6e98:0x3ffd5ca0 0x400d1d1e:0x3ffd5d00 0x400d1e27:0x3ffd5d50 0x4018c418:0x3ffd5dc0

Rebooting...

chegewara, 2019-04-10:
Ok, from logs we can tell its not null, because characteristic you are looking for present.
What you have to do is to parse backtrace logs:
```
Backtrace: 0x400d6e98:0x3ffd5ca0 0x400d1d1e:0x3ffd5d00 0x400d1e27:0x3ffd5d50 0x4018c418:0x3ffd5dc0
```
https://github.com/me-no-dev/EspExceptionDecoder

stiebel, 2019-04-10:
Ok, here we go:
```
PC: 0x400d59b4: BLERemoteService::getCharacteristic(BLEUUID) at [...]/Arduino/libraries/ESP32_BLE_Arduino/src/BLERemoteService.cpp line 159
EXCVADDR: 0x002a7f4c

Decoding stack results
0x400d59b4: BLERemoteService::getCharacteristic(BLEUUID) at  [...]/Arduino/libraries/ESP32_BLE_Arduino/src/BLERemoteService.cpp line 159
0x400d1a7a: connectToServer() at  [...]/Joystick/ESP_BLE_client_01/ESP_BLE_client_01.ino line 79
0x400d1b27: connectToServer() at  [...]/Joystick/ESP_BLE_client_01/ESP_BLE_client_01.ino line 103
0x4018a5c4: loopTask(void*) at  [...]/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp line 17
```
... but I can't see how this helps.
getCharacteristic(BLEUUID) is the problem, that is what we knew before?
Thanks again,
Roland

chegewara, 2019-04-10:
Hi,
i see where is the issue. Change value in this line to 1:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLERemoteService.cpp#L175

stiebel, 2019-04-11:
Hi chegewara,
great, that was it. When changing to 1 I get a characteristic, but unfortunately I run into the next problem:
Only 0x2A4D seems to work, even if there are lots of other characteristics listed on "nRF connect"
But fine, I will try this first:
Just the next line of the code of the .ino script
```
if(pRemoteCharacteristic->canRead()) {
      Serial.println("can read...");
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }
```
produeces this output:
```
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: HZ-2749, Address: ff:ff:70:04:43:5e, appearance: 963, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb
Forming a connection to ff:ff:70:04:43:5e
 - Created client
 - Connected to server
 - Found our service: 00001812-0000-1000-8000-00805f9b34fb
 - Checking for       00002a4d-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
 - Found our characteristic
can read...
The characteristic value was: 
We are now connected to the BLE Server.
Setting new characteristic value to "Time since boot: 2"
Setting new characteristic value to "Time since boot: 3"
Setting new characteristic value to "Time since boot: 4"
```
I do not get any value as to bee seen at "The characteristic value was: "
How do I get a value now?

Thanks again a million times for your help!

chegewara, 2019-04-11:
`Serial.println(value.c_str());`
Remember this will output only `printable` chars.

stiebel, 2019-04-11:
> PS if its not a problem try ble library from snippets repository with many bug fixes

where do I get this?
I cannot find any ble-arduino stuff here: https://github.com/nkolban/esp32-snippets

stiebel, 2019-04-11:
> PS if its not a problem try ble library from snippets repository with many bug fixes

where do I get this?
I cannot find any ble-arduino stuff here:

stiebel, 2019-04-11:
https://github.com/nkolban/esp32-snippets

stiebel, 2019-04-12:
Ok, I think I came es far as possible:
The service of my joystick is 0x1812 and I am able to connect.
The main charachteristic of my joystick is 0x2a4d and I can initiate reading without errors.
I started notifications with 0x2902 as mentioned by Andreas Spieß in his great video 174.
But...
I still do not get any readings or notifications.
But there is something weird: in his video Andreas shows nrf connect
pressing the notification button he gets readings - one per second.
If I do the same with my joystick I do not get any values at all,
so nrf connect does not get the "real" joystick values either.
Nevertheless the joystick works when bonded the normal way with my android phone and
I can use stick and buttons.
So I guess my (hopefully) last problem is of topic, because it is not a ESP32 problem only.
However, if anyone knows how to deal with this, please tell me.

Regarding my ESP32 problem I think I can call it SOLVED.

Thank you!

Roland

tablatronix, 2019-04-14:
Do you have a PR or diff for these BLE changes you speak of ?

chegewara, 2019-04-14:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteService.cpp#L129

Most reported bugs has been fixed in snippets repository, but didnt merged to arduino.

tablatronix, 2019-04-14:
Thanks I found it and it fixed most of my crashes, have a couple more now and then.

closed 2019-08-15 by stale[bot]

Dhayu-16, 2021-06-22:
Hi friend..
 Can you guys help me, I'm having
 Problem with remoteService,, not responding or null

// Receptor o cliente de un iTag o su emulador con ESP32. Solo necesitamos la direccion MAC del pulsador (itag)

#include "BLEDevice.h"

// The remote service we wish to connect to.

static BLEUUID serviceUUID("0000ffe0-0000-1000-8000-00805f9b34");

// The characteristic of the remote service we are interested in.

static BLEUUID charUUID("0000ffe1-0000-1000-8000-00805f9b34");

static BLEAddress *pServerAddress;

static BLERemoteService* pRemoteService;

static BLERemoteCharacteristic* pRemoteCharacteristic;

const int ledPin = 2;

void setup() {

   Serial.begin(115200);
delay(10000);
   Serial.println("Starting Arduino BLE Client application...");

   BLEDevice::init("");

   delay(10000);

   

   // DIRECCION MAC DEL SERVIDOR ITAG AL QUE NOS CONECTAMOS

   

   pServerAddress = new BLEAddress("FF:FF:20:00:66:3E");

   delay(500);

   Serial.print("Forming a connection to ");

   Serial.println(pServerAddress->toString().c_str());

   BLEClient* pClient = BLEDevice::createClient();

   Serial.println("Created client");

   pClient->connect(*pServerAddress);

   Serial.println("Connected to server");

   delay(500);

     // Obtain a reference to the service we are after in the remote BLE server.

    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);

    if (pRemoteService == nullptr) {

      Serial.print("Failed to find our service UUID: ");

      Serial.println(serviceUUID.toString().c_str());

          }else{

    Serial.println(" - Found our service");}

// Obtain a reference to the characteristic in the service of the remote BLE server.

    BLERemoteCharacteristic* pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID); 
				
    if (pRemoteCharacteristic == nullptr) {

      Serial.print("Failed to find our characteristic UUID: ");

      Serial.println(charUUID.toString().c_str());

    }

    Serial.println(" - Found our characteristic");

    

}

void loop() {

  

    // Read the value of the characteristic.

    uint8_t  rxValue = pRemoteCharacteristic->readUInt8();;

     if (rxValue > 0) { 

      Serial.println(rxValue);

      delay (300);

      pRemoteCharacteristic->writeValue(0);

     }

     

   

  

delay(0);

  

  } // The End

Dhayu-16, 2021-06-22:
[image]