espressif/arduino-esp32 #1685: Esp32 client example resetting when enable the notification

yldzmuhammed, 2018-07-27:
Guru is not working on my Windows 10 computer.
----------------------------- Remove above -----------------------------

### Hardware:
Board:	NodeMCU - ESP32
Core Installation/update date:	Updated today
IDE name:	Arduino IDE & Platform.io. Same results for both IDE
Flash Frequency:	80Mhz
Upload Speed:	912600

### Description:
I am trying to work arduino ble client example. I can connect to the peripheral device, find service, find characteristic.

After that try to enable notify, esp32 crashes.Below details.

Btw my esp32 regularly crashes. I dont know why but keep resetting.

### Error:

The characteristic value was: �

We are now connected to the BLE Server.
Turning notification on
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d402b  PS      : 0x00060530  A0      : 0x800d1cc0  A1      : 0x3ffd5c80
A2      : 0x00000000  A3      : 0x3ffd5cbc  A4      : 0x00000002  A5      : 0x00000001
A6      : 0x3ffeb3a8  A7      : 0x00010000  A8      : 0x800d367d  A9      : 0x3ffd5c30
A10     : 0x3ffd5c7c  A11     : 0x3ffeb3a8  A12     : 0x00000024  A13     : 0xce107853
A14     : 0x3ffd5c30  A15     : 0x3f401f7c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000030  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400d402b:0x3ffd5c80 0x400d1cbd:0x3ffd5cb0 0x4018c29c:0x3ffd5d00

Rebooting...
ets Jun  8 2016 00:22:57

### Sketch:

```
/**
 * A BLE client example that is rich in capabilities.
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("0000181a-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID(BLEUUID((uint16_t)0x2A6E));

static bool notification = false;
static BLEAddress *pServerAddress;
static bool doConnect = false;
static bool connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;

BLEScan* pBLEScan;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());

    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      //
      Serial.print("Found our device!  address: ");
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      doConnect = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    if(notification == false){
      Serial.println("Turning notification on");
      const uint8_t onPacket[] = {0x01, 0x00};
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2A6E))->writeValue((uint8_t*)onPacket, 2, true);
      notification = false;
    }
  }
  else
    pBLEScan->start(60);

  delay(1000); // Delay a second between loops.
} // End of loop
```
Here is difference from original code:

```
Serial.println("Turning notification on");
      const uint8_t onPacket[] = {0x01, 0x00};
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2A6E))->writeValue((uint8_t*)onPacket, 2, true);
      notification = false;
```

JoanMCD, 2018-08-22:
the notification descriptor is the 0x2902, isn't it !?

JoanMCD, 2018-08-22:
the notification descriptor is the 0x2902, isn't it !?

  LOGMSG("Register For Notify / Indication...", LOG_INFORMATIONAL);
  pRemoteCharacteristic_RX->registerForNotify(notifyCallback);

  //const uint8_t bothOff[]        = {0x0, 0x0};
  //const uint8_t notificationOn[] = {0x1, 0x0};
  const uint8_t indicationOn[]   = {0x2, 0x0};
  //const uint8_t bothOn[]         = {0x3, 0x0};
  pRemoteCharacteristic_RX->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn, 2, true);

tablatronix, 2019-04-14:
Any up date for this ?

closed 2019-08-15 by stale[bot]