espressif/arduino-esp32 #1686: wpa2-enterprise does not work in the last arduino cores

piyord, 2018-07-27:
Hi, 

In previous arduino cores, I could connect without problem to wpa2 enterprise radius network, but with last arduino cores fails the connection with same program.

Is the same problem comented in this issue:
https://github.com/espressif/esp-idf/issues/2181

When I call "WiFi.begin(ssid)" the ESP32 try to connect, but a Stack smashing protect failure error occurs.

I tested it on ESP-WROOM-32 and ESP-WROVER default profile with same result.

Program works great until "2018-06-24 14:59:16" arduino core update. From "2018-06-27 09:06:02" update until the last Arduino core update the program fails with same error.

Analizing the Arduino core code, this error is shown when function "esp_wifi_connect()" is executed inside function:
wl_status_t WiFiSTAClass::begin(const char* ssid, const char *passphrase, int32_t channel, const uint8_t* bssid, bool connect)
in "WiFiSTA.cpp" file

This is the error message with Backtrace:

Connecting to network: Uni-Net
Mode: STA
Channel: 1
SSID (7): Uni-Net
Passphrase (0): 
BSSID set: 0
.
Stack smashing protect failure!

abort() was called at PC 0x400ea028 on core 1

Backtrace: 0x4008f6f4:0x3ffbb6e0 0x4008f8f7:0x3ffbb700 0x400ea028:0x3ffbb720 0x400de649:0x3ffbb740 0x400de6f6:0x3ffbb800 0x400db587:0x3ffbb840 0x400db9c2:0x3ffbb880 0x400da7a9:0x3ffbb8a0 0x400da1cc:0x3ffbb8f0 0x400da239:0x3ffbb930 0x400d8c37:0x3ffbb950 0x400d8e39:0x3ffbb990 0x400d7f8d:0x3ffbb9b0 0x400e79aa:0x3ffbba30 0x400e7c36:0x3ffbba60

Rebooting...

This is my program. It is a modified version of "WiFiClientEnterprise.ino" example from Arduino core:

//Sketch edited by: Martin Chlebovec
//Personal website: https://arduino.php5.sk
#include "esp_wpa2.h"
#include <WiFi.h>
#define EAP_IDENTITY "identity@youruniversity.domain" //eduroam login --> identity@youruniversity.domain
#define EAP_PASSWORD "blablablabla" //your Eduroam password
String line; //variable for response
const char* ssid = "Uni-Net"; // Eduroam SSID
const char* host = "arduino.php5.sk"; //external server domain for HTTP connection after authentification
int fault=0;
void setup() {

  fault=128;

  WiFi.mode(WIFI_STA);
  
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.print("Connecting to network: ");
  Serial.println(ssid);
  WiFi.disconnect(true);  //disconnect form wifi to set new wifi connection

  WiFi.printDiag(Serial);
  
  esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide identity
  esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide username
  esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)); //provide password
  esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config to default (fixed for 2018 and Arduino 1.8.5+)
  esp_wifi_sta_wpa2_ent_enable(&config); //set config to enable function (fixed for 2018 and Arduino 1.8.5+)
  WiFi.begin(ssid); //connect to Eduroam function
  WiFi.setHostname("PINGER"); //set Hostname for your device - not neccesary
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  delay(6000);
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address set: "); 
  Serial.println(WiFi.localIP()); //print LAN IP

}
void loop() {
  delay(5000);
  if (WiFi.status() != WL_CONNECTED) { //if we lost connection, retry
    WiFi.begin(ssid);
    delay(500);        
  }
  Serial.print("Connecting to website: ");
  Serial.println(host);
  WiFiClient client;
  if (!client.connect(host, 80)) { // HTTP connection on port 80
    Serial.println("Connection lost! - Failed response");
  }
  
  String url = "/rele/rele1.txt"; //read .txt file    
  Serial.print("Requesting URL: ");
  Serial.println(url);
  client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + host + "\r\n" + "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println("Client timed out! - retry");
    }
  }
  
  while(client.available()) {
    line = client.readStringUntil('\n');
    Serial.println(line);  
  }
  
  Serial.println();
  Serial.println("End connection");
  client.stop();
}

piyord, 2018-07-30:
ok. Thanks.

Decoding stack results
0x4008f6f4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x4008f8f7: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x400ea5a4: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
0x400debc5: x509_parse_tbs_certificate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/x509v3.c line 1438
0x400dec72: x509_certificate_parse at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/x509v3.c line 1547
0x400dbb03: tls_process_certificate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c line 347
0x400dbf3e: tlsv1_client_process_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client_read.c line 958
0x400dad25: tlsv1_client_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tlsv1_client.c line 220
0x400da748: tls_connection_handshake2 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tls_internal.c line 347
0x400da7b5: tls_connection_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/tls/tls_internal.c line 327
0x400d91b3: eap_tls_process_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c line 482
0x400d93b5: eap_peer_tls_process_helper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_tls_common.c line 630
0x400d8509: eap_peap_process at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/wpa2/eap_peer/eap_peap.c line 1097

martinius96, 2018-07-30:
Everyone that tested sketch was able to connect to Eduroam network only or networks under FreeRadius service (for 802.1x - RADIUS server software).
Everybody gets that error on different networks: 
[image]
Sucessful connection on eduroam:
[image]

closed 2019-08-15 by stale[bot]