espressif/arduino-esp32 #1689: setjmp issue

akouz, 2018-07-28:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					        80Mhz
Upload Speed:						921600

### Description:
Describe your problem here

ESP32 crashes when I run a cooperative operating system https://github.com/akouz/a_coos based on the standard "setjmp.h" library.  I can run the sketch "a_coos.ino" on many microcontrollers, including ESP8266, but it does not work on ESP32. 

As far as ESP32 Dev Board does not have built-in LED, to compile the sketch modify line 23 of "a_coos.ino" to

static int led = 5;

### Debug Messages:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
== COOS demo ==
Guru Meditation Error: Core  1 panic'ed (Double exception)
Register dump:
PC      : 0x40086c0d  PS      : 0x00050836  A0      : 0x800d0b54  A1      : 0x3fffffe4  
A2      : 0x40000054  A3      : 0x00000000  A4      : 0x400d0868  A5      : 0x3ffc2428  
A6      : 0x3ffcf240  A7      : 0x3ffcf220  A8      : 0x3ffcf240  A9      : 0x8003d140  
A10     : 0x00000009  A11     : 0x3f400cf1  A12     : 0x09f53009  A13     : 0x00003500  
A14     : 0x00000020  A15     : 0x3ffcd340  SAR     : 0x00000017  EXCCAUSE: 0x00000002  
EXCVADDR: 0x8003d140  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
... etc, keeps repeating from "Guru Meditation Error". Sorry, Exception Decoder decodes nothing out of it.

With some extra prints in the sketch I can see the following:

1) Setup function void Coos::start(void) processed successfully, tasks invoked and return control to the scheduler. Thus, setjmp and longjmp functions work, I can store main context and then restore it. 

2) Function void Coos::run(void) crashes on line 123 of "coos.cpp", eg when it tries to invoke task using stored task context:

longjmp(task_context[task_no], 1);

akouz, 2018-07-28:
It looks like messages are not very consistent. Now I've got the following after reset:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
== COOS demo ==
.Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d0ae1  PS      : 0x00060930  A0      : 0x800d09d5  A1      : 0x3ffcf290  
A2      : 0x800d09d5  A3      : 0x00000000  A4      : 0x80083a78  A5      : 0x3ffcf1c0  
A6      : 0x00000000  A7      : 0x3ffcd380  A8      : 0x800d0bd5  A9      : 0x3ffcf2a0  
A10     : 0x0000028b  A11     : 0x0000028b  A12     : 0x0001e2fb  A13     : 0x3ffc2788  
A14     : 0x3ffcf2b0  A15     : 0x3ffcf280  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d0c39  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d0ae1:0x3ffcf290 0x400d09d2:0x3ffcf2c0 0x400dfe79:0x3ffcf2e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
== COOS demo ==
.Guru Meditation Error: Core  1 panic'ed (Double exception)
Register dump:
PC      : 0x40086b95  PS      : 0x00050836  A0      : 0x800d0adc  A1      : 0x3fffffdc  
A2      : 0x4000004c  A3      : 0x00000000  A4      : 0x3ffc2428  A5      : 0x400d088c  
A6      : 0x3ffcf280  A7      : 0x3ffcf260  A8      : 0x3ffcf280  A9      : 0x09f53009  
A10     : 0x00000077  A11     : 0x3ffc2430  A12     : 0x1009c500  A13     : 0xe52009d5  
A14     : 0x00000020  A15     : 0xff000000  SAR     : 0x00000017  EXCCAUSE: 0x00000002  
EXCVADDR: 0x09f53008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x40086b95Guru Meditation Error of type LoadStoreError occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400867d1  PS      : 0x00060035  A0      : 0x80086910  A1      : 0x3ffffdfc  
A2      : 0x40080b5c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffcf2e0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x40080b5c  A9      : 0x3ffffddc  
A10     : 0x00000035  A11     : 0x3ffcf2c0  A12     : 0x800dfe7c  A13     : 0x3ffcf2c0  
A14     : 0x000237bc  A15     : 0x3ffc2788  SAR     : 0x00000017  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40080b5c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400867d1Guru Meditation Error of type LoadStoreError occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400867d1  PS      : 0x00060033  A0      : 0x80086910  A1      : 0x3ffffc3c  
A2      : 0x40080b5c  A3      : 0x00000000  A4      : 0x80086c75  A5      : 0x3ffffe5c  
A6      : 0x3fffff1c  A7      : 0x00000017  A8      : 0x40080b5c  A9      : 0x3ffffc1c  
A10     : 0x00000031  A11     : 0x3ffffe3c  A12     : 0x80086b76  A13     : 0x3ffffe3c  
A14     : 0x3fffff1c  A15     : 0x3fffffdc  SAR     : 0x0000001e  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40080b5c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  `

... etc

stickbreaker, 2018-07-29:
How is longjump handling the stack? 
The esp32 uses a small (32k-64k) code cache to move code from the external FLASH memory to ICACHE.   So, if the next instruction to be executed is not in ICACHE, program flow halts, a cache refill interrupt is executed.  

This error is usually indicative of a bad pointer address, `LoadProhibited` is saying that an attempt was made to read from an unsupported memory address(usually 0). The memory map of ESP32 is NON-contiguous. 
[image]
 
>Guru Meditation Error of type LoadProhibited occurred on core 1. Exception was unhandled.

I would look for problems with the code, how it switches execution/task.

I don't know why you don't want to use the built in multi-task support?

Chuck.

akouz, 2018-07-29:
I guess the multi-task support you mentioned is specific to ESP-32, while COOS is a very generic cooperative scheduler. It supposed to run on any Arduino board supporting standard "setjmp.h" library. 

If I understand correctly what you said regarding to ESP-32 cash handling and flash memory, ESP-32 implements "setjmp.h" in a non-standard way with unpredictable behaviour.

copercini, 2018-07-29:
Not sure if it help you somehow, but commenting all `COOS_DELAY()` on .ino file, it's running without crashes

akouz, 2018-07-30:
Hi copercini,

In your sketch the coos_task2()  takes control and runs the loop forever. It defeats the whole point of the cooperative OS, where tasks should release control to the scheduler allowing other tasks to run. Control released by the COOS_DELAY() macro where setjmp/longjmp processed:

#define COOS_DELAY(__delay) if (!setjmp(coos.task_context[coos.task_no])) {longjmp(coos.main_context, __delay+1);} else

akouz, 2018-07-30:
I modified Coos::run() as follows:

`void Coos::run(void)
{
  static uchar pr = 0;
  int   res;
  int tmp;
  void (*tsk)(void);
  if (task_delay[task_no] == 0)
  {
    Serial.println("run: setjmp");
    res = setjmp(main_context);           // set return point and get delay value from the task 
    Serial.print(" setjmp exit, res=");
    Serial.println(res);
    if (res == 0)                         // after setting return point
    {
      Serial.println("run: longjmp");
      pr = 1;
      longjmp(task_context[task_no], 1);  // invoke task 
    }
    else                                  // after returning from task 
    {
      Serial.println("run: set delay");
      tmp = --res;              // added for debugging
      Serial.println("run: reload");
      task_delay[task_no] = tmp;        // it crashes here, it is line 137
      Serial.println("run: delay set");
    }
  }
  if (pr)  Serial.println("run: next task");
  task_cnt = (task_cnt > COOS_MAX_TASKS)? COOS_MAX_TASKS : task_cnt;
  if (++task_no > task_cnt)               // next task
  {
    task_no = 0;                          // round-robin
  }
  if (pr) Serial.println("run: update time");
  update_time(); 
}
`
After that I occasionally get the following output:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
== COOS demo ==
.run: setjmp
 setjmp exit, res=0
run: longjmp
 setjmp exit, res=651
run: set delay
run: reload
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d0b0f  PS      : 0x00060930  A0      : 0x800d099d  A1      : 0x3ffcf250  
A2      : 0x3ffc2794  A3      : 0x0000028a  A4      : 0x800d099d  A5      : 0x3ffcf180  
A6      : 0x00000000  A7      : 0x3ffcd340  A8      : 0x800d0b9d  A9      : 0x3ffcf230  
A10     : 0x0000000d  A11     : 0x0000000b  A12     : 0x0000000a  A13     : 0x3ffc278c  
A14     : 0x3ffcf270  A15     : 0x3ffcf240  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d0c01  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d0b0f:0x3ffcf250 0x400d099a:0x3ffcf280 0x400dc82d:0x3ffcf2a0

Rebooting...
`
Stack decoding:
`0x400d0b0f: Coos::run() at C:\Users\user\AppData\Local\Temp\arduino_build_289468\sketch\coos.cpp line 137
0x400d099a: loop() at E:\Proj\18_Arduino\Sketchbook\a_coos/a_coos.ino line 81
0x400dc82d: loopTask(void*) at C:\Users\user\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 18
`
coos.cpp line 137 is:
`task_delay[task_no] = tmp; `

closed 2019-08-15 by stale[bot]

f16falcona46, 2021-02-20:
I believe such a crash is legal behavior for a C or C++ runtime, since longjmp'ing down the stack is undefined behavior.

> [If the function that called setjmp has exited (whether by return or by a different longjmp higher up the stack), the behavior is undefined. In other words, only long jumps up the call stack are allowed.](https://en.cppreference.com/w/c/program/longjmp)

akouz, 2021-03-09:
This would be an issue is someone tries to use setjmp/longjmp in functions. But for COOS it is not the case. There is a clear warning about that:

[COOS_DELAY(x) must be used in the task itself; it cannot be used in functions called from task.](https://github.com/akouz/a_coos)

f16falcona46, 2021-03-09:
The task itself is a function, too. When a task firsts calls COOS_DELAY, it longjmp's back to Coos::start. At this point the task has exited (since it has longjmp'ed up the call stack), so the jmp_buf set by the setjmp inside the task is no longer valid. That means that the longjmp that Coos::run performs to that jmp_buf is also invalid.

akouz, 2021-03-09:
The issue will arise if stack is changed. As soon as stack is the same, setjump/longjump operate correctly. 

If one registers coos_task0() as coos.register_task(coos_task0), it must call COOS_DELAY() from coos_task0() itself. If another function inside coos_task0() is used to call COOS_DELAY(), stack will be changed and system will crash. But it will operate correctly when COOS_DELAY() called from coos_task0() because stack is intact. 

It is very easy to test, and it has been tested many-many times. While stack was not changed, COOS operates correctly. Must not use functions to call COOS_DELAY(), must use it within registered function itself.

f16falcona46, 2021-03-10:
You can't longjmp down the stack in portable code. You called longjmp from within coos_task0, which returned you to Coos::start. That jump was up the stack, and legal--however, you have now exited coos_task0, so any setjmp you made inside that function is invalid and you can't longjmp to it.

When you later try to longjmp into coos_task0 in Coos::run, you are jumping _down the stack_, which is illegal. Since the jmp_buf set inside coos_task0 was invalidated by longjmp'ing out of it, you cannot jump into coos_task0 again.

It works on some other platforms because they are allowed to implement this behavior in any way (or not at all)--the behavior is undefined. (Like reading past the end of allocated memory.) But there's no guarantee those actually work correctly (there can be silent corruption) and they can break at any time in the future.

akouz, 2021-03-10:
You might be confused by COOS tasks look like "functions". In fact COOS tasks newer called as functions and they never return as functions. They do not change stack. COOS tasks are entered  and exited by longjmp/setjmp only.  While processor executes task body, stack stays the same as in the main{). 

Inside COOS task you may call functions as you like, but when functions return to the task body, stack restored to the top level, the same level as in main(). Pairs of longjmp/setjmp are always processed at the top level of stack.

lbernstone, 2021-03-11:
Can I recommend that you guys take this conversation over to https://github.com/espressif/crosstool-NG where the guys that write the interpreter internals can assist.

igrr, 2021-03-11:
> In fact COOS tasks newer called as functions and they never return as functions. They do not change stack. COOS tasks are entered and exited by longjmp/setjmp only.

This doesn't seem to be the case, the first time the task function is invoked, it is invoked by a function call, not by longjmp:

https://github.com/akouz/a_coos/blob/d760d8880b35cc47411293e5384dc77b8ec4fc90/coos.h#L227

So the subsequent longjmps into the task stack are jumps down the stack. This may or may not work, the standard doesn't define the behavior in this case. On Xtensa, this won't work because of register window mechanism.