espressif/arduino-esp32 #1696: Implement calender using esp32

gabovich, 2018-07-30:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 NodeMCU-32S
Core Installation/update date:		11/jul/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Hello,
I need some help with calculate dates and holidays. I saw there are some libraries support the issue.
I used time.h and got the current time and date (except instead of GMT +3 i got GMT +2, why?).
Anyway, is there are any functions to get the sunset/sunrise time on a request day? Or even an expected date of holiday to come in the future (about couple of years).
Basiclaly i need to have a calender on my esp32 board and i wonder how to do this. 

Thank's.

JoanMCD, 2018-08-01:
I would suggest to go online and get this data from an external server. 
For sunlight you need geo location. Holiday can change by goverment decision.

lbernstone, 2018-08-01:
This forum is to present issues with the codebase, not a tutorial.  Please take it to gitter.

closed 2019-08-15 by stale[bot]