espressif/arduino-esp32 #1698: How to make the sdk part of the arduino package?

kapollo, 2018-07-30:
### Hardware:
Board:							mhetesp32devkit
Core Installation/update date:			30/jul/2018
IDE name:							Arduino IDE/Platform.io/IDF component
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I'm having constant problem with getting the ble+ssl mqtt running, there are some guidelines that require running make menuconfig (and thus recompiling the libraries and such).
As far as I understand selected libraries should go to the sdk folder.
Is there a way to automate this?
I know we can use Arduino as a component, but will this automatically use rebuilt libraries?
Is it  possible to generate a package for Platform.io?

lbernstone, 2018-07-30:
Find works pretty well for it. `find . -name "*.a" -exec mv {} $arduino_esp32/tools/sdk/lib`
You don't need a package for platform.io.  Just symlink ~/.platformio/packages/framework-arduinoespressif32 to your $arduino_esp32 directory.

lbernstone, 2018-07-31:
You will need to get the correct headers to match the compiled libraries..  That is some @me-no-dev magic that is not part of the codebase, and that is not trivial.  Maybe he will share.

kapollo, 2018-08-01:
@lbernstone thanks for your input I't seems also linker scripts are required for some changes, so yes it's not trivial, but doable :)

closed 2018-08-01 by kapollo