espressif/arduino-esp32 #1701: ble_server txPower allways -21

eltsov-da, 2018-07-31:
### Hardware:
Board:							TTGO LoRa OLED v1
Core Installation/update date:			30/07/2018
IDE name:							Arduino IDE?
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
ble txPower allways -21
Try to change ble txPower  in  BLE_server.ino example. But in  BLE_scan.ino got only  txPower: -21
Most of all I interested in 
### Sketch:
BLE_server.ino example line 20 
I try:
```
BLEDevice::init("MyESP32");
BLEDevice::setPower(ESP_PWR_LVL_N0); 

```
### Debug
```
Starting BLE work!
[D][BLEDevice.cpp:444] setPower(): >> setPower: 4
[D][BLEDevice.cpp:449] setPower(): << setPower
[D][BLEDevice.cpp:70] createServer(): >> createServer
[D][BLEServer.cpp:305] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:309] registerApp(): << registerApp
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:77] createServer(): << createServer
[D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1716] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 0]]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:211] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
[D][BLEService.cpp:217] setHandle(): << setHandle
[D][BLEService.cpp:90] executeCreate(): << executeCreate
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:94] createService(): << createService
[D][BLEService.cpp:239] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:242] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEService.cpp:254] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=21, data=48656c6c6f20576f726c642073617973204e65696c, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[к.txt](https://github.com/espressif/arduino-esp32/files/2244563/default.txt)

[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLEService.cpp:148] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLECharacteristic.cpp:83] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:94] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1669] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:609] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:611] setHandle(): << setHandle
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:135] executeCreate(): << executeCreate
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1805] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 0x28]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEService.cpp:174] start(): << start()
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, ### customScanResponseData: 0
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEServer.cpp:295] hand**leGATTServerEven**t(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:231] start(): << start
Characteristic defined! Now you can read it in your phone!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!

```
****
and also try:
```
BLEDevice::init("MyESP32");
esp_err_t errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_N0);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_N0);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_N0); 

```
 BLE_scan.ino example I got:
```
Scanning...
Advertised Device: Name: MyESP32, Address: 30:ae:a4:59:21:4e, txPower: -21
Devices found: 1
Scan done!

```
And also it works not more then 1m distance

lbernstone, 2018-07-31:
https://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/controller_vhci.html?highlight=esp_ble_tx_power_set#_CPPv220esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t
indicates this should only be set after a connection is made. Maybe esp_bredr_tx_power_set will do what you are looking for?  No amount of power is going to compensate for a bad antenna.  Do you get decent signal strength with WiFi?

eltsov-da, 2018-07-31:
It's very valid point about esp_ble_tx_power_set and set after a connection is made. But second parameter point that I can set power for advertising  and scanning (enum esp_ble_power_type_t  ESP_BLE_PWR_TYPE_ADV = 9  ESP_BLE_PWR_TYPE_SCAN = 10 ESP_BLE_PWR_TYPE_DEFAULT = 11) and I can't catch where is connection if I need advertising power set.
Antenna also very valid point and I think of it all the time. But I have no problem with WiFi I successfully connect to AP on the other flow in our building.
And thank for esp_bredr_tx_power_set I try it.

ehsanmqn, 2018-09-11:
@eltsov-da Hi, Could you finally find a way to do tx power adjustment? If yes guide me please. I have same problem.

eltsov-da, 2018-09-14:
Hi. It's seems to me it's set of 2 issues :)
1. https://github.com/espressif/esp-idf/issues/2284  - TX power during advertising is hardcoded. But real power is changed.
2. TTGO LoRa OLED v1 hard ware problem. It's very strange for me but 3 days ago I finally got  two Heltec Lora ... modules. And it's seems to me they work much better. I am sure I can got signal in about 5 miters. During week and I'll do some outdoors tests and tell you.

closed 2019-08-15 by stale[bot]