espressif/arduino-esp32 #1705: WiFiEvent SYSTEM_EVENT_STA_DISCONNECTED not fired in master

Viper2000XXL, 2018-07-31:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?31/jul/2018?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
In master the WiFiEvent SYSTEM_EVENT_STA_DISCONNECTED is not fired in Arduino. Switched back to an older WiFiGeneric.cpp and it is working. So the problem seems to be somewhere in WiFiGeneric.cpp file. In Debug-Log in Serial Monitor i can see the STA_DISCONNECTED Event but the event is not pushed to example Arduino Code below. SYSTEM_EVENT_STA_GOT_IP is working as expected but the case SYSTEM_EVENT_STA_DISCONNECTED is never executed.

### Sketch:

void WiFiEvent(WiFiEvent_t event){
    Serial.println(event);
    switch(event) {      
      case SYSTEM_EVENT_STA_GOT_IP:
          //When connected set 
          Serial.print("WiFi connected! IP address: ");
          Serial.println(WiFi.localIP());            
          connected = true;
          break;
      case SYSTEM_EVENT_STA_DISCONNECTED:
          Serial.println("WiFi lost connection");
         /connected = false;
          break;
  }
}

Viper2000XXL, 2018-07-31:
I played a little bit with the code and when i comment out line 377 and 378 which contains following Code:

`WiFi.enableSTA(false);

WiFi.enableSTA(true);`

ist is working and the DISCONNECTED Event is Fired to Arduino Code.

CircuitSetup, 2019-07-25:
> WiFi.enableSTA(false)

Curious what you meant by this. I can't get SYSTEM_EVENT_STA_DISCONNECTED to do anything on the front end either.

closed 2019-10-07 by stale[bot]