espressif/arduino-esp32 #1706: BLE setValue() in loop: GURU error.

remcoNL, 2018-07-31:
### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						115200

### Description:
I get a Guru Meditation error when calling setValue() in the loop, but not before it. 

### Sketch:
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <WiFi.h>
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID         "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID2 "050d3722-93b8-4f0f-b9c8-2be0bb33d559"  

BLECharacteristic *pCharacteristic2;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  WiFi.disconnect();

  BLEDevice::init("MyESP32");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);

  BLECharacteristic *pCharacteristic2 = pService->createCharacteristic(                     
                                         CHARACTERISTIC_UUID2,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  
  pCharacteristic2->setValue("w"); // this does NOT cause a problem

  
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");

}

void loop() {
  
  pCharacteristic2->setValue("w");                // causes GURU meditation problem
delay(10000);
  char w=119;
//pCharacteristic2->setValue(std::string(w,1));   // causes GURU meditation problem
}

```

### Debug Messages:
```

entry 0x40078a90
Starting BLE work!
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:308] registerApp(): << registerApp
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:76] createServer(): << createServer
[D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1716] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 0]]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:161] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
[D][BLEService.cpp:167] setHandle(): << setHandle
[D][BLEService.cpp:90] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:93] createService(): << createService
[D][BLEService.cpp:189] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:192] addCharacteristic(): Adding characteristic: uuid=050d3722-93b8-4f0f-b9c8-2be0bb33d559 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEService.cpp:204] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=1, data=77, characteristic UUID=050d3722-93b8-4f0f-b9c8-2be0bb33d559
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLEService.cpp:126] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLECharacteristic.cpp:83] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:94] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 050d3722-93b8-4f0f-b9c8-2be0bb33d559, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1669] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: 050d3722-93b8-4f0f-b9c8-2be0bb33d559]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:609] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=050d3722-93b8-4f0f-b9c8-2be0bb33d559
[D][BLECharacteristic.cpp:611] setHandle(): << setHandle
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:135] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1805] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 0x28]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEService.cpp:152] start(): << start()
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
Characteristic defined! Now you can read it in your phone!
[DGuru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c29b  PS      : 0x00060c30  A0      : 0x800d406c  A1      : 0x3ffd8b60  
A2      : 0x3ffd8b91  A3      : 0x00000004  A4      : 0x00000013  A5      : 0x3ffd8b91  
A6      : 0x3fff3ca6  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x0000004c  A11     : 0x00000061  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00060a23  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4000c29b:0x3ffd8b60 0x400d4069:0x3ffd8b70 0x400d463d:0x3ffd8be0 0x400d31af:0x3ffd8c00 0x401b1eea:0x3ffd8c40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
Starting BLE work!

lbernstone, 2018-07-31:
pCharacteristic2 depends on objects that are local to setup().

remcoNL, 2018-08-01:
Thank you Iberstone! I didn't even know that was possible. I moved some code out of setup into ... ... nothing? and it works fine now.

closed 2018-08-01 by remcoNL