espressif/arduino-esp32 #1707: Add getSketchSize & getFreeSketchSpace API

JoanMCD, 2018-08-01:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?1/Aug/2018?
IDE name:							?Arduino IDE?

### Description:
- Feature request -
Please implement the well known (ESP8266) API functions  for the ESP32

getSketchSize()
getFreeSketchSpace()

and if it makes sence for the ESP32:
getFlashChipRealSize()
getCoreVersion()
getBootVersion()
getBootMode()

thx, Joan

closed 2019-08-15 by stale[bot]

lbernstone, 2020-09-17:
https://docs.espressif.com/projects/esp-idf/en/v3.3.3/api-reference/system/system.html#chip-version

CriPstian, 2020-09-17:
> https://docs.espressif.com/projects/esp-idf/en/v3.3.3/api-reference/system/system.html#chip-version

This chip version API described at that link does not provide any of the info that was exposed by the functions mentioned above.
Is this a `NO` to my question?
I need a way to obtain the actual size of my chip flash size, not the sum of the existing partitions, like the `getFlashChipRealSize` did before.

lbernstone, 2020-09-17:
It simply means that I am not google.  This information is readily available if you look for it.
https://github.com/espressif/esp-idf/blob/release/v3.3/components/spi_flash/include/esp_spi_flash.h#L54

CriPstian, 2020-09-17:
> It simply means that I am not google. This information is readily available if you look for it.
> https://github.com/espressif/esp-idf/blob/release/v3.3/components/spi_flash/include/esp_spi_flash.h#L54

This is actually what I was saying. There's a clear note above that function that states:
```This value does not necessarily match real flash size.```
But no worries, I'll investigate further, hopefully come back here with an answer or PR.

lbernstone, 2020-09-17:
https://github.com/espressif/esptool can certainly get at that info.  Look there.

VTeselkin, 2021-11-01:
getFlashChipRealSize 
=
`String(spi_flash_get_chip_size())`