espressif/arduino-esp32 #1708: Patition Scheme

JoanMCD, 2018-08-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			1/Aug/2018
IDE name:							Arduino IDE
Files in: \packages\esp32\hardware\esp32\1.0.0\tools\partitions

### Description:
- The config 'default' and 'min_spiffs' contains the same parameters.
- Please add the sizes of data/OTA/SPIFFS into the menu of the Arduino IDE.
- The deluxe version would be to have it dynamic configurable.

It is possible to read back the config (or if exists the OTA) by the sketch ?

Joan

JoanMCD, 2018-08-01:
I have created two configs:

[image]

lbernstone, 2018-08-01:
It would be a nice thing to have a partition editor plugin.  You should start working on that.
You can read the entire table off a device by running `python esptool.py read_flash 0x8000 0xc00 /tmp/parts` from the command line.
Here's a snippet to get the size of a specific partition:
```
#include <esp_partition.h>
void setup() {
  Serial.begin(115200);
    const esp_partition_t *data_partition = NULL;
    data_partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
            ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "spiffs");
    if (data_partition != NULL) {
        Serial.printf("partiton addr: 0x%08x; size: %d; label: %s\n", data_partition->address, data_partition->size, data_partition->label);
    }
}
void loop() {}
```

JoanMCD, 2018-08-01:
Thank you for the feedback and the code example - works great !
Plugin: I'm not the java programmer... I tried to compile a other plugin (ESP32FS) on windows and got many errors. 

The check if the OTA partition exists should integrated into the OTA code. 

extended snippet:
```
#include <esp_partition.h>

  const esp_partition_t *partition = NULL;
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_OTA_0, "app0");
  if (partition != NULL) {
    Serial.printf("APP partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_OTA_1, "app1");
  if (partition != NULL)
  {
    Serial.printf("OTA partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  else
  {
    Serial.printf("No OTA partiton found !\n");
  }
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, (esp_partition_subtype_t)0x99, "eeprom");
  if (partition != NULL) {
    Serial.printf("EEprom partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "spiffs");
  if (partition != NULL) {
    Serial.printf("SPIFFS partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
```

lbernstone, 2018-08-01:
ArduinoOTA is based on Update.h, which is where I lifted that code from.  It does check for a free OTA partition before it runs.

closed 2019-08-15 by stale[bot]