espressif/arduino-esp32 #1709: Failure to upload

karansoi, 2018-08-01:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 LOLIN D32
Core Installation/update date:		MicroPython v1.9.4-417-g21dae8771 on 2018-07-31
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						512000

### Description:
The compiled code does not upload

### Sketch:
```cpp

//Change the code below by your sketch

uint64_t chipid;  

void setup() {
	Serial.begin(115200);
}

void loop() {
	chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
	Serial.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
	Serial.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

	delay(5000);

}

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
Arduino: 1.8.5 (Windows 10), Board: ""WeMos" WiFi&Bluetooth Battery, 80MHz, 512000, Debug"

Archiving built core (caching) in: C:\Users\Karanbir\AppData\Local\Temp\arduino_cache_244977\core\core_espressif_esp32_WeMosBat_FlashFreq_80,UploadSpeed_512000,DebugLevel_debug_6da8d55d0c53c0ebef21ae478b9e06ba.a
Sketch uses 181104 bytes (13%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13980 bytes (4%) of dynamic memory, leaving 313700 bytes for local variables. Maximum is 327680 bytes.
C:\Users\Karanbir\Documents\Arduino\hardware\espressif\esp32/tools/esptool/esptool.exe --chip esp32 --port COM3 --baud 512000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\Karanbir\Documents\Arduino\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\Karanbir\Documents\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 C:\Users\Karanbir\AppData\Local\Temp\arduino_build_588218/GetChipID.ino.bin 0x8000 C:\Users\Karanbir\AppData\Local\Temp\arduino_build_588218/GetChipID.ino.partitions.bin 
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\x08')
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\x08')

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

karansoi, 2018-08-01:
I have changed the usb cables it gives same result. However when i open the serial monitor and press the reset bitton on the board this is what I get

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4596
load:0x40078000,len:0
load:0x40078000,len:12768
entry 0x4007ad68
[0;32mI (387) cpu_start: Pro cpu up.[0m
[0;32mI (387) cpu_start: Single core mode[0m
[0;32mI (387) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (391) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (397) heap_init: At 3FFC57D0 len 0001A830 (106 KiB): DRAM[0m
[0;32mI (403) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (416) heap_init: At 40091B28 len 0000E4D8 (57 KiB): IRAM[0m
[0;32mI (422) cpu_start: Pro cpu start user code[0m
[0;32mI (216) cpu_start: Starting scheduler on PRO CPU.[0m
OSError: [Errno 2] ENOENT
MicroPython v1.9.4-417-g21dae8771 on 2018-07-31; ESP32 module with ESP32
Type "help()" for more information.
>>>

Same result with Putty terminal.

karansoi, 2018-08-01:
when these comands are used in the serial monitor The board responds.
`import` network
sta_if = network.WLAN(network.STA_IF); `sta_if.active(True)`

1469292) wifi: wifi driver task: 3ffcb6ec, prio:23, stack:4096, core=0
I (1469292) wifi: wifi firmware version: ac8d7b4
I (1469292) wifi: config NVS flash: enabled
I (1469302) wifi: config nano formating: disabled
[0;32mI (1469302) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1469312) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (1469332) wifi: Init dynamic tx buffer num: 32
I (1469332) wifi: Init data frame dynamic rx buffer num: 64
I (1469332) wifi: Init management frame dynamic rx buffer num: 64
I (1469342) wifi: Init static rx buffer size: 1600
I (1469342) wifi: Init static rx buffer num: 10
I (1469342) wifi: Init dynamic rx buffer num: 0
[0;33mW (1469352) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration[0m
[0;32mI (1469512) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 2[0m
I (1469522) wifi: mode : sta (b4:e6:2d:96:3f:85)
[0;32mI (1469522) wifi: STA_START[0m
True
`sta_if.scan()`

[0;32mI (1502662) network: event 1[0m
[(b'Mboscodictiasaur', b'P\xc7\xbfP\x8c\xaa', 11, -69, 4, False), (b'5Y8ZJ', b'p\xf1\x96\xa1\x99\xc3', 6, -72, 3, False), (b'HOME-2E28', b'\xb8\x9b\xc9\xeb.(', 11, -72, 4, False), (b'NTGR_VMB_1505788226', b',03\xecI4', 6, -80, 3, False), (b'HOME-1556-2.4', b'T\xbe\xf7\xcb\x96\xb8', 1, -86, 4, False), (b'xfinitywifi', b'^\xbe\xf7\xcb\x96\xb8', 1, -88, 0, False), (b'ANNEPC', b'\xe0F\x9aq\xb2\xd0', 1, -89, 3, False), (b'Field', b'\x00\x12\x0eE\x1c\xa3', 6, -90, 2, False), (b'LinksysVivintfd4', b'$\xf5\xa2\x12\xef\xd6', 11, -90, 3, False)]
>>>

Serial communication is working but upload of sketch via Arduino ide is not working.

lbernstone, 2018-08-01:
Try wiping it.
https://desire.giesecke.tk/index.php/2018/05/04/upload-of-code-over-usb-serial-fails-because-of-erroneous-code/

karansoi, 2018-08-01:
On comand line also erase flash command gives similar error.
`esptool.py -p COM3 -b 230400 erase_flash `
The output is
[image]

stickbreaker, 2018-08-01:
@karansoi if you are using Arduino, open the Serial monitor window, set it to 115200baud.  press the reset on the board.  You should see a block of text on the monitor.  If nothing shows up on the Serial Monitor, you have connectivity issues, or the board is dead.

Chuck.

karansoi, 2018-08-01:
@stickbreaker  i get a response as you said. This is what is displayed:

ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4596
load:0x40078000,len:0
load:0x40078000,len:12768
entry 0x4007ad68
�[0;32mI (387) cpu_start: Pro cpu up.�[0m
�[0;32mI (387) cpu_start: Single core mode�[0m
�[0;32mI (387) heap_init: Initializing. RAM available for dynamic allocation:�[0m
�[0;32mI (391) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM�[0m
�[0;32mI (397) heap_init: At 3FFC57D0 len 0001A830 (106 KiB): DRAM�[0m
�[0;32mI (403) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM�[0m
�[0;32mI (409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM�[0m
�[0;32mI (416) heap_init: At 40091B28 len 0000E4D8 (57 KiB): IRAM�[0m
�[0;32mI (422) cpu_start: Pro cpu start user code�[0m
�[0;32mI (216) cpu_start: Starting scheduler on PRO CPU.�[0m
OSError: [Errno 2] ENOENT
MicroPython v1.9.4-417-g21dae8771 on 2018-07-31; ESP32 module with ESP32
Type "help()" for more information.

stickbreaker, 2018-08-01:
@karansoi so, the board has MicroPython running, do you want to keep it running python or use Arduino c++?

If you want to use Arduino, then install the release core into the Arduino Environment, (boards manager JSON) then create your sketch and upload it.
I use a WeMos Bluetooth & battery without any issues. My download rate is 921600 and 80mhz flash.
when you see:
>Connecting....

Touch the switch marked "Boot"
 
Chuck.

karansoi, 2018-08-02:
I have only one switch "Reset".
How do I install the 'Release core...'
I installed the ESP platform from Github.
The additional boards URL space is blank in the File>>preferences pane

stickbreaker, 2018-08-02:
Here are the instructions:

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

karansoi, 2018-08-02:
@stickbreaker I have installed through the boards manager, but again the upload failed. However when the reset buttone is switched a number of times after the 
` ` Connecting............. 
the code uploads. 
Will I need to do this always? Is there a way around this ?

lbernstone, 2018-08-02:
If you connect pin0 to ground when you want to flash, that is equivalent to pressing the button that should be there.  Painful, but that's what you get for $6.50.

tobozo, 2018-08-02:
LOLIN D-32 Pro: Upload works every time if I manually short D0/GND while the reset button is pushed (then release).

M5Stack, Odroid-Go, now D-32 Pro ... what's this new thing with sellers to remove the flash button, do they believe we all have three hand?

karansoi, 2018-08-07:
Is there a circuit improvisation that can do the pin GPIO0 grounding for code upload? I have raised a dispute for the board not working on aliexpress.

karansoi, 2018-08-09:
@tobozo 
Is this to be done on before upload?
1 ground GPIO0 
2 toggle reset.
3 Remove GPIO0 ground connection

lbernstone, 2018-08-09:
https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf
In the lower left corner you will see the schematic for the automatic upload control.

karansoi, 2018-08-09:
Which board should I buy from Bangood or Alieexpress that is tried by you.
Will appreciate your advice.

lbernstone, 2018-08-09:
I have used TTGO OLED ESP32 boards for development.  Just bought a WROVER from https://www.aliexpress.com/store/product/TTGO-T8-V1-3-ESP32-WiFi-Module-bluetooth-4MB-PSRAM-TF-CARD-3D-ANTENNA-Micropython-Development/2787042_32845435620.html as well.

closed 2018-08-09 by karansoi