espressif/arduino-esp32 #1711: ESP32 halts when repeatedly invoking ble.begin() & ble.end()

muojp, 2018-08-02:
### Hardware:
Board:							DOIT-ESP32-DevBoard
Core Installation/update date:		31/jul/2018
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I tested SimpleBLE library and found it freezes ESP32 board if you periodically call `begin()` and `end()`. On my board, ESP32 halts or reboots at 33rd time of calling `begin()`.

Here are some detail:

 - delay duration (500ms) on the attached sketch is just an example and set for easier issue-repro. I actually tried 500 - 10000 ms and ESP32 crashed in every case.
 - omitting `end()` seem solving the problem (no freeze at 100+ times of invocation) but I don't think it's a desired solution.
 - changing or keeping advertised identifier on each `begin()` doesn't affect the test result.
 - ESP32 sometimes reboots after freezing but sometimes just halts.
 - after digging into the ESP-IDF(ref: https://github.com/espressif/esp-idf/tree/aaf12390e) code, I found it seems `BTA_EnableBluetooth` (bt/bluedroid/bta/dm/bta_dm_api.c) stuck and somehow fails to return `future` call. I'm not sure whether it is an issue of ESP-IDF or ESP32-Arduino, but filing this issue because it reproduces on simple ESP32-Arduino code.

### Sketch:
```cpp
#include "SimpleBLE.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

SimpleBLE ble;

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);
}

void loop() {
  for (int i = 0; i < 64; i++) {
    ble.begin("My BLE node");
    Serial.println(String(i));
    delay(500);
    ble.end();
  }
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
0
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
abort() was called at PC 0x4008a255 on core 0

Backtrace: 0x40090c38:0x3ffdaaf0 0x40090e3b:0x3ffdab10 0x4008a255:0x3ffdab30 0x4008a280:0x3ffdab50 0x4008a3d1:0x3ffdab80 0x4014de6f:0x3ffdaba0 0x40147771:0x3ffdae60 0x401400c1:0x3ffdaeb0 0x40116b73:0x3ffdaee0 0x40111d3a:0x3ffdaf00 0x4012db6e:0x3ffdaf20 0x4010a8b1:0x3ffdaf40 0x40111b36:0x3ffdaf60 0x400ecfe6:0x3ffdaf80

Rebooting...

```

muojp, 2018-08-21:
Workaround: perform `make menuconfig` and disable dynamic memory allocation for BT/BLE

[image]

KorochanOne, 2018-09-20:
I got the same problem with SerialBT. (SerialBT.begin()/SerialBT.end())
I 'd like to try your workaround but I don't know how to use make menuconfig for ESP32-Arduino.
Is there good manual for how to change menucinfig for ESP32-Arduino.

abort() was called at PC 0x4008a255 on core 0

Backtrace: 0x40090c38:0x3ffdba20 0x40090e3b:0x3ffdba40 0x4008a255:0x3ffdba60 0x4008a280:0x3ffdba80 0x4008a3d1:0x3ffdbab0 0x4014e3d7:0x3ffdbad0 0x40147cd5:0x3ffdbd90 0x40140625:0x3ffdbde0 0x400d72ab:0x3ffdbe10 0x400efb3d:0x3ffdbe30 0x400e6ff1:0x3ffdbe50 0x4012b9b3:0x3ffdbe70 0x40114291:0x3ffdbee0 0x40114211:0x3ffdbf00 0x4011436e:0x3ffdbf20 0x400efa96:0x3ffdbf40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
HIT ANY KEY......................

itofficeeu, 2019-01-07:
I can confirm to be able to reproduce this issue with the sketch from @muojp. I am using a Heltec WiFi LoRa 32 with a ESP32 on board. The ESP32 either hangs after 30 times, or crashes and restarts.

I'm not using any OS - and hence do not know how to disable dynamic memory allocation.

I am experiencing similar issues in another program using BLEDevice.h - like in this series of includes:
```
#include <CayenneLPP.h>
#include <lmic.h>
#include <hal/hal.h>
#include <Wire.h>
#include <SPI.h>
#include <SSD1306.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include "BLEDevice.h"
```

Here the Bluetooth client hangs 42 loops after the Bluetooth server is turned of. Between those two events, is the client scanning for devices 42 times - and during scanning, all the time not finding any. If it finds some random Bluetooth units in between, it seems to keep the program running. I will do further studies on my setup regarding the seemingly positive influence of other Bluetooth devices around.

The following are data printed in the terminal. The value, "88.99" is the value gathered by the client (the unit that hangs) from the Bluetooth server, which is another ESP32 "barebone" (ESP-WROOM-32).

After the first semicolon - in each row - the program in the client gets the value from the server, if they are connected. If they aren't connected, the clients starts a scan for 2 seconds. The clients presents whatever it finds in these 2 seconds, and the program continues with the next loop. So all the double semicolons are scans without any connections established and no data - or any other device found.
The text "Now disconnected from BLE unit." comes just after me turning the Bluetooth server off. - I turn it of for the purpose of testing the scans and re-connections.
The final row only has one semicolon - and it is here where the ESP32 hangs in a scan - maybe before or after the real scan.

```
   ...
;88.99;
;88.99;
;88.99;
;88.99;
;;
Now disconnected from BLE unit.
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;;
;
```

The software in the Bluetooth server is a copy from:
[https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_server/BLE_server.ino](url)

The client is (modified slightly) based on this:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_client/BLE_client.ino

I will check if my modifications have had influence, but doubt it, as the problem seems to exist outside my code too. - Otherwise would this thread not exist.

kenken64, 2019-07-15:
Is it possible to get attention to his problem?

wakwak-koba, 2019-07-16:
I found and fixed many bugs in BLECLient, and supports multiple device connection and reconnection after disconnection.
If you are using BLECLient, please try my Fork.
https://github.com/wakwak-koba/arduino-esp32

Please see the difference for details.
https://github.com/espressif/arduino-esp32/compare/master...wakwak-koba:master

kenken64, 2019-07-24:
> I found and fixed many bugs in BLECLient, and supports multiple device connection and reconnection after disconnection.
> If you are using BLECLient, please try my Fork.
> https://github.com/wakwak-koba/arduino-esp32
> 
> Please see the difference for details.
> [master...wakwak-koba:master](https://github.com/espressif/arduino-esp32/compare/master...wakwak-koba:master)

how to install and test from your fork ?

wakwak-koba, 2019-07-25:
If you have already installed the original arduino-esp32, please overwrite the following files into your arduino-esp32 environment.

Required:
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.cpp
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEDevice.cpp
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEDevice.h
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteCharacteristic.cpp
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteCharacteristic.h
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteService.cpp

Not required:
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEScan.cpp
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEScan.h
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertisedDevice.cpp

If necessary:
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/examples/BLE_clients_catch_notifies/BLE_clients_catch_notifies.ino

closed 2019-10-15 by stale[bot]

fronik, 2019-11-06:
> If you have already installed the original arduino-esp32, please overwrite the following files into your arduino-esp32 environment.
> 
> Required:
> https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/...

Hi, thank you very much for your work. 
I use yours BLE fork. And it is much more stable. But after prolonged use (6h - 48h) still hangs. 
I'm a novice, and I can't cope with the problem alone. Can I count on your help?
After work 6h - 48h, hangs on step: `pRemoteCharacteristic->writeValue(buf, 2, true);`
And today in the port monitor at the time of hovering, I saw the following message:
`[E][BLEClient.cpp:302] gattClientEventHandler(): Config mtu failed`

My Project (PlatformIO):
```
#include <Arduino.h>
#include <BLEDevice.h>
#include <WiFi.h>

// WiFi
const char* ssid     = "WiFiName";
const char* password = "WiFiPass";
WiFiServer server(80);
String header;

// Bluetooth
#define FLORA_ADDR "c4:7c:8d:62:11:11" //MAC Xiaomi Flower Care
BLEClient* pClient = BLEDevice::createClient();
// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001204-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID uuid_sensor_data("00001a01-0000-1000-8000-00805f9b34fb");
static BLEUUID uuid_write_mode("00001a00-0000-1000-8000-00805f9b34fb");
static BLEAddress floraAddress(FLORA_ADDR);
static BLERemoteCharacteristic* pRemoteCharacteristic;
int temp;
int moisture;
int light;
int conductivity;

//Other

uint32_t millisControl = 0;         
uint32_t TimerP1 = 30;              
uint32_t TimerP1Start = 0;          
uint32_t TimerP1Stop = TimerP1;     

void getSensorData(BLEAddress pAddress){ 
  btStart();
  Serial.print("=== Forming a connection to Flora device ===");
  Serial.println(pAddress.toString().c_str());
  Serial.println(" - Connection to Flora");
  if (!pClient->connect(pAddress)){
      pClient->disconnect();
      delay(500);
      Serial.println(" - Cannot connect to Flora.");
      return; 
  } else {
      Serial.println(" - Connected to Flora");

      // Obtain a reference to the service we are after in the remote BLE server.
      BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
      if (pRemoteService == nullptr) {
          Serial.print("Failed to find our service UUID: ");
          Serial.println(serviceUUID.toString().c_str());
      } else {
          Serial.println(" - Found our service");
      
          pRemoteCharacteristic = pRemoteService->getCharacteristic(uuid_write_mode);
          Serial.println(" - pRemoteCharacteristic = ");
          uint8_t buf[2] = {0xA0, 0x1F};
[... 158 lines cut ...]
    Serial.println("");
  }

  millisOverControl();
  TimerCheck();
  delay(500);

}

```
platformio.ini 
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

monitor_speed = 115200
upload_speed= 921600
board_build.partitions = no_ota.csv
build_flags =
  -W
```

And sorry for my English, i use google translate.

fronik, 2019-11-06:
And yet, ESP32 continues respond to ping. 
But more than anything is not working (WEB Server or loop)

wakwak-koba, 2019-11-08:
If WiFi is not used, will there be a change in uptime?
I have experience of encountering the phenomenon that BLE disconnects when using with WiFi.

fronik, 2019-11-08:
Ok. i will test without wifi.

fronik, 2019-11-26:
I found a similar problem: https://github.com/nkolban/esp32-snippets/issues/874

I made such corrections:
```
FreeRTOS.cpp

 68: xSemaphoreTake(m_semaphore, 10000UL);
194: rc = ::xSemaphoreTake(m_semaphore, 10000UL) == pdTRUE; 
```

Then started testing two ESP32 boards (2x ESP32 and 2x MiFlora).
Test run: 21.11.2019 - works now. 
Interval get data from MiFlora every 30 seconds (for hard test).

Testing continues. With WiFi.