espressif/arduino-esp32 #1712: BluetoothSerial uses 64% of available Flash

LoRaTracker, 2018-08-02:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			08/jun/2016
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I wrote a simple 'Hello World' program to send out the text via Bluetooth Serial. The program works but uses rather a lot of memory;

Sketch uses 849303 bytes (64%) of program storage space. Maximum is 1310720 bytes.
Global variables use 74520 bytes (22%) of dynamic memory, leaving 253160 bytes for local variables. Maximum is 327680 bytes.

I have been converting a LoRa GPS tracker program across from Arduino Atmel code, adding the Bluetooth serial takes program flash size from around 17% to 70%

Is this expected behaviour ?

Using the libbt.a mod from 'BluetoothSerial silently drops bytes to be sent #1537'

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

JamesGKent, 2018-08-02:
unfortunately the bluetooth protocol is horrifically complex, so the code required to implement it isn't small.
thats not saying this won't/can't be improved in future, but it is currently the state of things that adding in bluetooth functionality consumes large amounts of flash

lbernstone, 2018-08-02:
There are partitioning schemes that support larger apps, and devices with 8MB and even 16MB flash are available.

LoRaTracker, 2018-08-02:
OK thanks for the info.

closed 2018-08-02 by LoRaTracker

Lingaperumal, 2019-05-23:
Hello 
I have a doubt....regarding serial bluetooth..
if the device is not connected to server bluetooth how we can display or to tell that your device is not connected in a serial communication...

thanks in advance