espressif/arduino-esp32 #1713: ESP32 devkit V1  not working!

gabovich, 2018-08-02:
### Hardware:
Board:							ESP32 DevKitV1
Core Installation/update date:		11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:				        40Mhz
Upload Speed:					115200

### Description:
Hey,
Just got my ESP32 devkit V1 and when i plug it in i don't see nothing. Like no COM on the ports and the LED on the esp32 board is off even when iplug it in.
In addition, couple of seconds after i connect the board to my computer, it's warming up really badly. Like it's going to burn or something.
Any suggestions?

I have the esp-32s NODEMCU and it's work fine. So why the ESP32 devkit V1 not working? Maybe got a tackled one?

Thank's.

gabovich, 2018-08-02:
Nothing is shorted. I connect it normally without connect any wire or matrix and except getting really hot nothing happens.

stickbreaker, 2018-08-02:
Failed, Dead board. Not a software problem.

Chuck.

closed 2019-08-15 by stale[bot]