espressif/arduino-esp32 #1714: WiFiClient::connect doesn't work properly

chiararuggeri, 2018-08-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			28/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:

After the last core's update I noticed that connected() function in WiFiClient class doesn't work as before. After some debug I found out that with the last commit, you removed a check on the result of recv() function that you previously made before to check errno variable.
In my sketch I continuously check if client is still connected before to make any operation on it, but now, even if recv returns 0 (and I suppose it means there's no error), errno is checked anyway and sometimes it has a bad value that lets connected function return false.
I don't paste my sketch here since it is really big. You can replicate the issue simply using WiFiClient example that comes with the library, by adding a check for client.connected() before reading data. I'll paste this modified example below.
What happens in this example is that the first line is read, then client get disconnected.

If you also notice this problem, please restore check on result of recv() function in order to avoid check on errno variable if recv is fine.

Thanks.

### Sketch:
```cpp

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        if(client.connected()){
          String line = client.readStringUntil('\r');
          Serial.print(line);
        }
    }

    Serial.println();
    Serial.println("closing connection");
}
```

lbernstone, 2018-08-02:
For a workaround (and maybe a better solution anyhow), use `if (WiFi.status() == WL_CONNECTED)` instead.

Limag, 2019-03-10:
Yes, you should only consider `errno`, if the function call went wrong!

`man recv`: _return the number of bytes received, or -1 if an error occurred;_ 
In the case of the used call `recv(fd(), &dummy, 0, MSG_DONTWAIT);` the expected result is 0 for OK or -1 in case of error.

closed 2019-08-15 by stale[bot]