espressif/arduino-esp32 #1716: Can't upload sketch to ESP32-WROVER board

jwktje, 2018-08-02:
### Hardware:
Board:							 ESP32-WROVER on custom board
Core Installation/update date:			27/7/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
What do we need to do to make our own board compatible with automatic flashing from the Arduino IDE via FTDI?

I want to upload a sketch from the Arduino IDE to a custom board that houses the ESP32-WROVER module. I am using FTDI. The same chip can be flashed when using a Development breakout [something like this](https://nl.aliexpress.com/item/ESP32-test-board-batch-burn-armatuur-voor-ESP32-WROVER-IPEX-Versie-ESP-32-ESP-32S-ESP32/32851571106.html?spm=a2g0z.search0104.3.47.54a941677xzd01&ws_ab_test=searchweb0_0%2Csearchweb201602_4_10152_10151_10065_10344_10068_10342_10343_10340_10341_10696_10084_10083_10618_10304_10307_10820_10821_10301_10843_10059_100031_10103_10624_10623_10622_10621_10620-10152_10151%2Csearchweb201603_46%2CppcSwitch_3&algo_expid=00a1d1bc-0198-47a5-98fe-1d4270dfb21d-7&algo_pvid=00a1d1bc-0198-47a5-98fe-1d4270dfb21d&priceBeautifyAB=0). 

However, when flashing it via FTDI on my custom board, it doesn't work. Testing with someone else we managed to figure out that the chip on the custom board is possible when manually triggering pin states like so:

1. Keep Reset and GPIO connected to ground before upload..
2. When upload start disconnect Reset from ground (It pulls up). Downloading start
3. When uploading completes, disconnect GPIO0 from ground. (It pulls up)
4. Connect RESET with ground, it goes to reset and when released from ground it goes to normal operation

When wanting to use the automatic flash mode/reset via FTDI. It doesn't work. The relevant schematic is shown below. the RESET flag is connected to the EN pin on the ESP32-WROVER. And the 5-volt is regulated to 3.3v for the ESP32. We are also not using FTDI to power the chip during flashing.
[image]

What are we doing wrong? Do we need to add our board to the IDE in a specific way to make the FTDI give the right signals to the ESP?

Any help would be greatly appreciated!

jwktje, 2018-08-02:
Isn't that what my schematic does? Or do you mean pulling up the IO0 to VCC by default is whats wrong? Because in your linked schematic, doesn't that also happen? In the top right "ESP32 Module" section it shows GPIO0 also being pulled up to VDD.

I might be missing something super obvious. Sorry in that case

lbernstone, 2018-08-02:
Sorry, I didn't see the GPIO0 hiding there.  Are you able to communicate with the esp32 at all?  I would suspect power issues.  Try using a different power source for 3V3 to the esp32.

jwktje, 2018-08-02:
Yes, when manually pressing buttons like described in the steps 1 to 4 in my original message, we can flash the chip, and get serial output from it and everything. So, the main issue is with DTR/RTS and automatic flashing.

jwktje, 2018-08-03:
Would we maybe need to get an oscilloscope or something to check if the FTDI chip is actually doing what we think it is? Or is the schematic the issue?

ovaisiq, 2018-08-03:
[WIFI_ESP32.pdf](https://github.com/espressif/arduino-esp32/files/2257606/WIFI_ESP32.pdf)

Hello. I am also working with jwktje on this project. For information complete schematic is presented here.

lbernstone, 2018-08-03:
I think your schematic seems to be the same as all the other boards doing this.  I think a scope is your next step.  A multimeter may even be able to catch the signals.

ovaisiq, 2018-08-03:
https://youtu.be/7vz3XmFQ6bc

ovaisiq, 2018-08-03:
This is something in between Scope and multimeter...  Red LED is connected to DTR and Yellow is connected to RTS. Please let me know if you think the generated sequence is right..

ovaisiq, 2018-08-03:
I think the issue might be value of reset capacitor and/or lack of capacitor with GPIO0.

ovaisiq, 2018-08-03:
When I deleted program and reset button, I removed the GPIO capacitor also.

ovaisiq, 2018-08-04:
Guys. the issue is resolved. I added 2.2uF capacitor to reset pin instead of 100nF. It works like a charm. Tested on two boards.

ovaisiq, 2018-08-04:
At first I added 1uF capacitor but that was not enough...  This high value (2.2uF) may be because of no capacitor present at GPIO0 pin. So this value might be different with a GPIO0 capacitor present.

closed 2019-08-15 by stale[bot]