espressif/arduino-esp32 #1719: espwroom32 dev-kit v2 support the flash encryption

venkat3344, 2018-08-03:
if espwroom32 dev-kit support the flash encryption then support me.

karansoi, 2018-08-03:
@Venkat3344 what support you need. Condition not understood.

venkat3344, 2018-08-03:
need to procedure how to enabled Flash Encryption enabled in espwroom32(espwroom32 dev-kit v2) 
I was tried this steps below to enabled to run flash encryption in espwroom32 dev-kit_v2.
1. Genetate flash encryption key
    C:\python27\Scripts>espsecure.py –port COM3 generate_flash_encryption_key
my_flash_encryption_key.bin 
2. Burn flash encryption key
    C:\python27\Scripts>espefuse.py –port COM3 burn_key flash_encryption my_flash_encryption_key.bin
3.Make menuconfiguration -> Security Features -> Enable Flash Encryption on boot (ENABLE it).
4.Burn efuse flash encrypt cnt
     C:\python27\Scripts>espsecure.py –port COM3 burn_efuse FLASH_ENCRYPT_CNT 
 And shown efuse Summary: FLASH_CRYPT_CNT  Flash encryption mode counter = 1R/W (0x1)
8.Flashing is done Than showeing error repeatedly 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

And Try in another Board: Burn the FLASH_CRYPT_CNT getting error : A fatal error occurred : Efuse FLASH_ENCRYPT_CNT failed to burn. Protected?

Check out the FLASH_CRYPT_CNT with espefuse.py --port "COM3" summary
it should still have the value: 1

copercini, 2018-08-03:
There is a good guide about it here: http://esp-idf.readthedocs.io/en/latest/security/flash-encryption.html

just be careful, it can brick your ESP32

venkat3344, 2018-08-03:
I read flash-encryption.html. and done above mention steps. Is their any different tool or procedure how to enable flash encryption. what the reason Efuse FLASH_ENCRYPT_CNT failed to burn and how to solve this problem.Let me give some useful points in flash-encryption.html

closed 2019-08-15 by stale[bot]