espressif/arduino-esp32 #1720: tone was not declared in this scope

Master811129, 2018-08-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					 80Mhz
Upload Speed:						115200

### Description:
when i use tone function the arduino console says:
"tone was not declared in this scope."
if the esp32 core hasn't tone function please add it.
thanks.

### Sketch:
```cpp

#include <Arduino.h>

void setup()
{
 pinMode(25,OUTPUT);
}

void loop() 
{
  tone(25,200);
  delay(1000);
  tone(25,800);
  delay(1000);
}
```

lbernstone, 2018-08-03:
I think tone is part of the ledc functions in esp32.  I don't have a speaker around to test, but try this:
```
void setup() {
  ledcSetup(0,1E5,12);
  ledcAttachPin(25,0);
}
void loop() {
  ledcWriteTone(0,800);
  delay(1000);
  uint8_t octave = 1;
  ledcWriteNote(0,NOTE_C,octave);  
  delay(1000);
}
```

Master811129, 2018-08-04:
yeah it works! thank you
but why the esp32 developers didn't color the code for easier reading and coding?...

lbernstone, 2018-08-04:
If this is working for you, please close the issue.

Master811129, 2018-08-05:
thanks again!

closed 2018-08-05 by Master811129

KTibow, 2020-05-07:
Just in case anybody finds this, you can make a quick swap like this:
```c++
void tone(byte pin, int freq) {
  ledcSetup(0, 2000, 8); // setup beeper
  ledcAttachPin(pin, 0); // attach beeper
  ledcWriteTone(0, freq); // play tone
}
```
Don't know how to make it so it asynchronously stops playing after an amount of time, though. To stop it, you can do that again with 0 as the frequency, or do something like:
```c++
int playing = 0;
void tone(byte pin, int freq) {
  ledcSetup(0, 2000, 8); // setup beeper
  ledcAttachPin(pin, 0); // attach beeper
  ledcWriteTone(0, freq); // play tone
  playing = pin; // store pin
}
void noTone() {
  tone(playing, 0);
}
```

lbernstone, 2020-05-07:
or my library. https://github.com/lbernstone/Tone

connornishijima, 2020-08-09:
Why was this closed? A tone() abstraction still doesn't seem to exist in this core.

Thomascountz, 2021-02-21:
This functionality is famously unavilable in Espressif's [arduino-esp32](https://github.com/espressif/arduino-esp32) library and members of the community have found various [work-arounds](https://github.com/lbernstone/Tone) such as using the native [LED Control](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html) functions to generate PWM signals.

However, looking more closely at [arduino-esp32](https://github.com/espressif/arduino-esp32) library, not only has Espressif provided a clean API for generating tones, they've provided an interface for generating specific PWM frequencies for specific notes on the chromatic scale in different octaves.

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c#L255-L266

```c
double ledcWriteNote(uint8_t chan, note_t note, uint8_t octave){
  const uint16_t noteFrequencyBase[12] = {
  //   C        C#       D        Eb       E        F       F#        G       G#        A       Bb        B
      4186,    4435,    4699,    4978,    5274,    5588,    5920,    6272,    6645,    7040,    7459,    7902
  };

  if(octave > 8 || note >= NOTE_MAX){
      return 0;
  }
  double noteFreq =  (double)noteFrequencyBase[note] / (double)(1 << (8-octave));
  return ledcWriteTone(chan, noteFreq);
}

```

Although not directly compatible with Arduino's `tone()`, the function allows for a clearer interface by providing named frequencies out of the box via the `note_t` type.

```c
typedef enum {
    NOTE_C, NOTE_Cs, NOTE_D, NOTE_Eb, NOTE_E, NOTE_F, NOTE_Fs, NOTE_G, NOTE_Gs, NOTE_A, NOTE_Bb, NOTE_B, NOTE_MAX
} note_t;
```

## Example Usage
Here is an example of generating PWM frequencies at 50% duty cycle to play the C-major scale.

```c
void playScale(uint8_t pin, uint8_t channel) {
  ledcAttachPin(pin, channel);
  ledcWriteNote(pin, NOTE_C, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_D, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_E, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_F, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_G, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_A, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_B, 4);
  delay(500);
  ledcDetachPin(pin)
}
```

jonfroehlich, 2021-05-13:
Thanks for this uncovering this functionality @Thomascountz! We incorporated it into one of our teaching lessons for our Physical Computing course: https://makeabilitylab.github.io/physcomp/esp32/tone.html.

lbernstone, 2021-05-13:
Tone is now available from in the [ESP32Servo](https://github.com/madhephaestus/ESP32Servo/blob/master/src/ESP32Tone.h) library (which can be found in Arduino library manager).

jonfroehlich, 2021-05-13:
Oh, this is great. Thanks!

limingjie, 2022-03-05:
Verified working code and schematic.

@Thomascountz I was following your comment and it took me quite some time to realize the `ledcWriteNote` should write to the channel. I should have read the function declaration first.

```c++
void playScale(uint8_t pin, uint8_t channel)
{
    ledcAttachPin(pin, channel);
    ledcWriteNote(channel, NOTE_C, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_D, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_E, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_F, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_G, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_A, 4);
    delay(100);
    ledcWriteNote(channel, NOTE_B, 4);
    delay(100);
    ledcDetachPin(pin);
}
```

**Schematic**
[image]

**PCB**
[image]

Pilot5860, 2022-04-26:
After adding the shared function to the code I wrote for esp32, the "notone" error persisted due to the ezbuzzer library I got in the ide interface. Therefore I removed the "ezbuzzer" library and the error was gone. I am hesitant as I am a novice coder. I just wanted to ask if I could have done wrong by removing this library.

The process worked like this:

My installed libraries in the first stage:

#include <Wire.h>
#include <ezBuzzer.h>
#include <LiquidCrystal_I2C.h>
#include <SPI.h>
#include <LIS3MDL.h>
#include <ads1248.h>
#include <ads1256.h>
#include <ads12xx.h>
 
Here are the error codes I got for the tone(buzzer, 500) line:
" 'tone' was not declared in this scope"

I disabled this line.(//),
then you share; I added " void playScale(uint8_t pin, uint8_t channel) " function.

Then I added the following codes:

void setup() {
  ledcSetup(0,1E5,12);
  ledcAttachPin(25,0);
}
void loop() {
  ledcWriteTone(0,800);
  delay(1000);
  uint8_t octave = 1;
  ledcWriteNote(0,NOTE_C,octave);
  delay(1000);
}
and I kept getting this error.
" C:\Users\Hp\Desktop\arduino-1.8.19\libraries\buzzer-main\src\ezBuzzer.cpp:130:22: error: 'noTone' was not declared in this scope noTone(_buzzerPin); "

I thought that the issue might have something to do with the "ezBuzzer" library and I disabled the library(//)
Thus, the errors are over.

What I want to ask is: Now that I have removed the "ezbuzzer" library, should I add another library for the "ledcWriteTone" code written for esp32?
Will these codes work without installing any other library related to "ton"?

dhebbeker, 2023-04-27:
`tone()` and `noTone()` have been implemented meanwhile (see #6402).