espressif/arduino-esp32 #1723: What version to use with ESP-IDF v3.0.2

etcetc, 2018-08-03:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:	custom based upon ESP-WROOM module
Core Installation/update date:			
IDE name:	esp-idf
Flash Frequency:		40Mz
Upload Speed:		115200

### Description:
This is a question / general issue.  For production, I would like to use esp-idf v3.0.2 which is the latest official release.  However, I have not been able to figure out which version of arduino-esp to use with this.  The latest does not compile, so based on [previous response to related question](https://github.com/espressif/arduino-esp32/issues/660#issuecomment-331647581) I went off the commits in the package_esp32_index.template.json file to backtrack to a version that would work.  I found that version a59eafbc on June 27 does not compile, but the immediate commit before that does.  However, a lot changed in version a59eafbc, amongst which was a fix for a bug that triggers stack-smashing.  

I t seems there aren't branches that map to the various IDF versions that are supported.

So the question is: what is the correct way to find the commit to use for the last official esp-idf release?  Would it make sense to have a living branch that compiles with that release and where contributors can submit bug fixes?

lbernstone, 2018-08-03:
https://github.com/espressif/arduino-esp32/issues/1142
tl;dr
`git log | grep -i "update idf" | head -1`

lbernstone, 2018-08-03:
https://github.com/espressif/arduino-esp32/issues/1142
tl;dr
`git log -i --grep "update idf" | head -1`

etcetc, 2018-08-03:
Thanks - neat trick with git for what I did manually.  But:

1.   this works if you guarantee that you will only update IDF once after it is released, so that the version prior to the "update idf" maps to the official released IDF.

2. Still not obvious how to maintain this version, fix bugs, etc.  

Why not have a living branch that maps to latest official esp-idf release for bug fixes?  Seems it would be transparent and more easily maintainable.

lbernstone, 2018-08-03:
Because when the idf changes, it often requires changes to arduino to match.  Unless they are released in tandem, there will always be some gap time where some of the headers & functions don't match up.  If you want to be 100% sure that things are equivalent, checkout the arduino commit from the log and then the matching esp-idf commit (don't head the git log, then git log | grep in idf for the version listed there) and checkout that version of the idf.  It would be nice if @me-no-dev would put something in his build automation that would put the esp-idf commit number into a file, but this method is not too terrible, and there are only a few people that will really need this.

etcetc, 2018-08-05:
You are correct, but my question relates to maintaining an older arduino release that works with the released IDF, because the master branch evolves to match later idf's (which are in beta mode) as you point out.  

I guess there just aren't enough people that care to make it worthwhile.  Thanks

closed 2018-08-05 by etcetc