espressif/arduino-esp32 #1724: (esp-idf) make flash fails due to the overlapping of offset addresses

wtakuo, 2018-08-04:
### Description:
Using esp-idf with arduino-esp32, after successful build, flashing fails due to the overlap of offset addresses.  As the arguments of `python` in the last line of the output of `gmake` shows, `ota_data_initial.bin` and `boot_app0.bin` share the same offset address `0xe000`.  If I remove `0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin` from the arguments, flashing succeeds.

The build configuration (`sdkconfig`) is almost the same as default (generated with `make menuconfig`). In such a case, I think that the `boot_app0.bin` is incorrectly assigned to the offset `0xe000` (according to `default.csv`, the address should be for ota_data).

```
$ gmake
...
esptool.py v2.5.0
To flash all build output, run 'make flash' or:
python /Users/takuo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin 0xe000 /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin 0x1000 /Users/takuo/esp/foo/build/bootloader/bootloader.bin 0x10000 /Users/takuo/esp/foo/build/foo.bin 0x8000 /Users/takuo/esp/foo/build/default.bin

$ gmake flash
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x10000 )...
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin
gmake: *** [/Users/takuo/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2
```

### Configuration
Board: M5Stack, DevKitC
Core Installation/update date: 3, Aug. 2018
IDE: esp-idf (master branch, 31, Jul. 2018, 1c7a8b3b712c4020562551f692d0ced8f7470d2c)
Library: arduino-esp32 (master branch, 30, Jul. 2018, e346f20aa9f455f3147d9d018e09dd7c05818951)
Flash Frequency:	40MHz
Upload Speed: 115200
OS: macOS 10.13.6

Project directory
```
.
|-- Makefile
|-- build
|-- components
|   `-- arduino    (arduino-esp32)
|-- main
|   |-- component.mk
|   `-- main.cpp
`-- sdkconfig
```
`Makefile`
```
PROJECT_NAME := foo
include $(IDF_PATH)/make/project.mk
```
`main.cpp`
```cpp
#include <Arduino.h>
void setup() {}
void loop() {}
```
`component.mk` is empty

`sdkconfig`
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
[... 576 lines cut ...]
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

beegee-tokyo, 2018-08-04:
I think this belongs to [esp-idf issues](https://github.com/espressif/esp-idf/issues), not here. This issues list is for Arduino-ESP32 problems, not for ESP-IDF problems.

wtakuo, 2018-08-04:
Actually I wondered which issues I should post. I posted here because this problem does not occur without arduino-esp32.  Anyway do you think I should close here and repost to esp-idf issue?

beegee-tokyo, 2018-08-04:
Sorry, I missed that you are using Arduino as a component.
You are in the right place.

lbernstone, 2018-08-04:
Do you get the error with other Arduino partition configurations?  In particular, No_OTA?

wtakuo, 2018-08-04:
```
esptool.py v2.5.0
To flash all build output, run 'make flash' or:
python /Users/takuo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin 0xe000 /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin 0x1000 /Users/takuo/esp/foo/build/bootloader/bootloader.bin 0x10000 /Users/takuo/esp/foo/build/foo.bin 0x8000 /Users/takuo/esp/foo/build/no_ota.bin
```

wtakuo, 2018-08-04:
```
$ gmake
esptool.py v2.5.0
To flash all build output, run 'make flash' or:
python /Users/takuo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin 0xe000 /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin 0x1000 /Users/takuo/esp/foo/build/bootloader/bootloader.bin 0x10000 /Users/takuo/esp/foo/build/foo.bin 0x8000 /Users/takuo/esp/foo/build/no_ota.bin
```

wtakuo, 2018-08-04:
```
```
$ gmake
...
esptool.py v2.5.0
To flash all build output, run 'make flash' or:
python /Users/takuo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin 0xe000 /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin 0x1000 /Users/takuo/esp/foo/build/bootloader/bootloader.bin 0x10000 /Users/takuo/esp/foo/build/foo.bin 0x8000 /Users/takuo/esp/foo/build/no_ota.bin
```

wtakuo, 2018-08-04:
Yes. I tried other configurations: Minimal, No OTA, and Minimal SPIFFS. The same error (`ota_data_initial.bin` and `boot_app0.bin` share the same offset `0xe000`) occurs in any case.

```
esptool.py v2.5.0
To flash all build output, run 'make flash' or:
python /Users/takuo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 /Users/takuo/esp/foo/build/ota_data_initial.bin 0xe000 /Users/takuo/esp/foo/components/arduino/tools/partitions/boot_app0.bin 0x1000 /Users/takuo/esp/foo/build/bootloader/bootloader.bin 0x10000 /Users/takuo/esp/foo/build/foo.bin 0x8000 /Users/takuo/esp/foo/build/no_ota.bin
```

wtakuo, 2018-08-04:
Build after `rm -r build` produces the same result, even after the re-installation of `esp-idf` and `arduino-esp32`.

lbernstone, 2018-08-04:
You need to find the source of that bin file and remove it.  I have no such file on my machine.

wtakuo, 2018-08-04:
The file (`ota_data_initial.bin`) is generated in `esp-idf/components/app_update/Makefile.projbuild`. I'm currently trying to find why this makefile works even when using configurations that disable OTA.

If I don't use arduino-esp32, this error does not occur (`ota_data_initial.bin` is not generated when using non-OTA configurations).

lbernstone, 2018-08-04:
Yup, incompatible new code. https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md#compilation-errors

KonstantinKondrashov, 2018-08-06:
Hi all. This issue happens due to the crossed functionality in IDF and Arduino. We added in IDF feature which clear otadate partition by the command `make flash` (only if otadate is in the partition table). For resolving this issue try to comment this line: https://github.com/espressif/arduino-esp32/blob/master/Makefile.projbuild#L9. We will resolve this issue soon. Thanks.

psykokwak-com, 2018-08-06:
@KonstantinKondrashov, Hi. I just tried to comment the line you give but I have no succes.

> esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: C:/msys32/home/Jyce/esp/arduino/components/arduino/tools/partitions/boot_app0.bin
> make: *** [/home/Jyce/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2

KonstantinKondrashov, 2018-08-06:
Sory try to comment only next line: https://github.com/espressif/arduino-esp32/blob/master/Makefile.projbuild#L10

psykokwak-com, 2018-08-06:
Perfect, It's working.
Thanks.

closed 2018-08-16 by me-no-dev