espressif/arduino-esp32 #1729: 40khz Timer setup

folny, 2018-08-05:
Hi

Can you please tell me why this setting is unstable and the oscilloscope output can see bending as otherwise I could set the timer to be stable at 40Khz thank you for your advice.

my code 

#include "driver/timer.h"

static intr_handle_t s_timer_handle;

void init_timer(float timer_period_us)
{
    timer_config_t config = {
            .alarm_en = true,
            .counter_en = false,
            .intr_type = TIMER_INTR_LEVEL,
            .counter_dir = TIMER_COUNT_UP,
            .auto_reload = true,
            .divider = 80 
    };
    
    timer_init(TIMER_GROUP_0, TIMER_1, &config);
    timer_set_counter_value(TIMER_GROUP_0, TIMER_1, 0);
    timer_set_alarm_value(TIMER_GROUP_0, TIMER_1, timer_period_us);
    timer_enable_intr(TIMER_GROUP_0, TIMER_1);
    timer_isr_register(TIMER_GROUP_0, TIMER_1, &timer_isr, NULL, 0, &s_timer_handle);
    timer_start(TIMER_GROUP_0, TIMER_1);
    
}
                                                               
static void timer_isr(void* arg){
TIMERG0.int_clr_timers.t1 = 1;
TIMERG0.hw_timer[1].config.alarm_en = 1;
digitalWrite(23, digitalRead(23) ^ 1);
}

void setup() {
pinMode(23, OUTPUT); 
init_timer(12);
}

void loop() {

}

folny, 2018-08-05:
Thank you for your advice, I did the way you wrote, but the problem still persists

lbernstone, 2018-08-05:
digitalReads and Writes are going to be sent to a queue, so may not give reliable response times (aka, slow as molasses).  Read is not needed in this case either, keep track of a boolean and toggle it instead.

folny, 2018-08-05:
I have tried this option but it does not work right.

char state;

static void IRAM_ATTR timer_func(void* arg)
{
    // your code, runs in the interrupt
    if (state) {
        state = 0;
        gpio_set_level(BLINK_GPIO, 1);
    } else {
        state = 1;
        gpio_set_level(BLINK_GPIO, 0);
    }
}

closed 2019-08-15 by stale[bot]