espressif/arduino-esp32 #1732: BluetoothSerial

michili, 2018-08-06:
@copercini 

sorry for trivial problem

I tried to use the BluetoothSerial lib but failed complteley

ESP32 and Arduino IDE

no Zip ? how to handle ? brought in the complete ESP...

aborted because no esp_bt.h; after fixing (moving it) no because of no esp_app_api.h....

completely confused...

could not yet run the simple "hello world" because I do not get BluetoothSerial.h included...

M.

michili, 2018-08-07:
> It seems that you are using an old version of ESP32 arduino core....
>
> try to just remove the old ESP32 installation, and then follow these steps: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md
>
> BluetoothSerial lib should appears on examples by default

Thank you so much !!! it works.... all the best M.
 

Gesendet: Montag, 06. August 2018 um 17:23 Uhr

copercini, 2018-08-07:
You're welcome =)

closed 2018-08-07 by copercini