espressif/arduino-esp32 #1733: Router connection: 50% failure (50% success)  is back

kas2, 2018-08-06:
Hi

Please refer to https://github.com/espressif/arduino-esp32/issues/234

I just installed the actual core (using v3.2-dev-39-gaaf12390 SDK) and get same failure again

With previous core (v3.1-dev-661-gf586f5e6 SDK), as a workaround, WiFiGeneric.cpp was modified as such:
`
        } else if(reason == WIFI_REASON_AUTH_EXPIRE) {
            if(WiFi.getAutoReconnect()){
                WiFi.begin();
            }

`
This modification is now omitted in the actual core
Please fix

TIA

tomtobback, 2018-10-23:
hi, i have the exact same problem, i think it started after i updated my esp32-arduino core earlier this month. i have read your issues https://github.com/espressif/esp-idf/issues/401 and https://github.com/espressif/arduino-esp32/issues/234 but can't find a solution there.
i'm using a minimal sketch to connect to wifi; it fails every other time after a hard reset (stuck on status 4)
it works every time if i add your 2 lines in the while WiFi.status loop, but i want to use wifimanager (https://github.com/tzapu/WiFiManager) and there the same problem comes up, and i can't figure out where to add those lines to make it work. any suggestions?

njordan77, 2019-08-13:
Hi, i have the same ESP32 (MatrixCore32) issue that every second time it works if hard reset. Very Strange and annoying as this never happened with ESP8266. I found out that its related to my AP, as with a Mobile Phone Hotspot it works great every time.....

Is there already a working fix or any kind of reasonable workarround, even if not so fancy....stability is my major priority.

closed 2019-10-26 by stale[bot]