espressif/arduino-esp32 #1735: ASSERT_PARAM(32 0), in rwble.c at line 222

ArTyGill, 2018-08-06:
### Hardware:
Board:	ESP32 Dev Module
Chip:	ESP32D0WDQ6 (revision 0)
IDE name:	VisualStudio (Arduino IDE)
Flash Frequency:	80Mhz
Upload Speed:	921600
esptool.py : v2.3.1

### Description:
ASSERT_PARAM(32 0), in rwble.c at line 222
error with ble + IRAM_ATTR + random

ArTyGill, 2018-10-04:
Recently, I didn’t notice this problem probably fixing it.

closed 2018-10-04 by ArTyGill

brunofgc, 2022-08-01:
Hi Guys, I'm facing a similar problem. Do you know the contact for rwble.c writer?

The Issue -> assert rwble.c 261, param 00020000 00000000
After it, the processor halts.

System:

Framework: Arduino;
IDE: Visual Studio;
Addon: PlatformIO;

[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
lib_deps = 
    knolleary/PubSubClient@^2.8
    h2zero/NimBLE-Arduino@^1.3.8
    marian-craciunescu/ESP32Ping@^1.7
    plerup/EspSoftwareSerial@^6.16.1
    vshymanskyy/TinyGSM@^0.11.5

LNqueen, 2023-02-27:
> 

me too, have you fixed this issue? and I also used the esp32-c3 chip.