espressif/arduino-esp32 #1739: ESP32 FAT_DATE dos not work...

pcbtborges, 2018-08-08:
### Hardware:
Board:	ESP32 Dev Module
Core Installation/update date: 11/jul/2017
IDE name:	Arduino IDE
Flash Frequency:	80Mhz
Upload Speed:	921600

### Description:
Hi, I am porting a project that runs on a Mega256 where I use the following code to inform the SD routine to add date creation/update properly:
The code below does not work with ESP32's SD.h

Assistance welcome.

### Sketch:
`//function to supply datetime to SD file system
void dateTime(uint16_t* date, uint16_t* time) {
  // return date using FAT_DATE macro to format fields
  *date = FAT_DATE(Clock.getYear() - 48, Clock.getMonth(Century), Clock.getDate());
  // return time using FAT_TIME macro to format fields
  *time = FAT_TIME(Clock.getHour(h12, PM), Clock.getMinute(), Clock.getSecond());
}`

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
exit status 1
'FAT_DATE' was not declared in this scope

closed 2019-08-01 by pcbtborges

mateusz-kusmierz, 2019-12-11:
I have the same problem...

jucaballa, 2020-12-02:
I have the same problem!