espressif/arduino-esp32 #1743: WiFi.SSID returning null or empty using esp_wifi_sta_get_ap_info

megazero1316, 2018-08-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			30/jul/2018
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					92600

### Description:
As far as i know that if esp32 connected to a wifi it store the configuration in the flash but calling WiFi.SSID() return null after restarting esp32, what i notice is that the SSID is called from esp_wifi_sta_get_ap_info while the psk is called esp_wifi_get_config in WiFiSTA.cpp
 the way i solved this problem is i copied the code in WiFi.psk() and added to WiFi.SSID() like this :

```
String WiFiSTAClass::SSID() const
{
    if(WiFiGenericClass::getMode() == WIFI_MODE_NULL){
        return String();
    }
    wifi_ap_record_t info;
    if(!esp_wifi_sta_get_ap_info(&info)) {
        return String(reinterpret_cast<char*>(info.ssid));
    }
    else{
          wifi_config_t conf;
          esp_wifi_get_config(WIFI_IF_STA, &conf);
          return String(reinterpret_cast<char*>(conf.sta.password));    
    }

    return String();
}

```
and everything went good for me , is this consider a good solution for this problem ? as i was looking for others solutions but couldn`t find 

### Sketch:
```
#include <WiFi.h>
/*commented after 1st time connection*/
//const char* ssid     = "your-ssid";
//const char* password = "your-password";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop()
{
    
}

```

javiercuellar73, 2019-01-03:
Great solution!!!! 
This should be add to next version.

lbernstone, 2019-01-03:
You can also extract this directly from nvs.
```
#include <nvs_flash.h>
#include <nvs.h>

const char* get_ssid() {
  nvs_handle hndl;
  nvs_open("nvs.net80211", NVS_READONLY, &hndl);
  size_t ssid_len = 36;
  char ssid[ssid_len];
  esp_err_t err = nvs_get_blob(hndl, "sta.ssid", ssid, &ssid_len);
  String ssid_str;
  if (err == ESP_OK) 
    for(char x : ssid) 
      if (x>32) ssid_str += x;
  return ssid_str.c_str();
}

void setup() {
  Serial.begin(115200);
  nvs_flash_init();
  Serial.println(get_ssid());
}

void loop() {}
```

johnty, 2019-05-16:
just another ping on this: ran into this issue recently. are there any reasons why @megazero1316's solution should not be merged in?

huexpub, 2019-06-13:
yes, is good solution

everslick, 2019-08-15:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_sta_get_ap_infoP16wifi_ap_record_t

looks like a bug in IDF. commenting here to stop stale[bot] from closing this issue.

everslick, 2019-10-14:
@me-no-dev : is `WiFiSTAClass::SSID()` supposed to return the SSID that is configured or the one that  it is connected to?

me-no-dev, 2019-10-14:
it grabs the configuration so I imagine it's the one configured.

everslick, 2019-10-14:
no, the original code does not:

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L547

it grabs `info`.

I think IDF works as designed. it should not return an SSID if not currently connected. Question is, what does a Arduino user expect? Should we return the configured SSID if not connected?

me-no-dev, 2019-10-14:
this is a good question... maybe we should

closed 2019-12-27 by stale[bot]