espressif/arduino-esp32 #1744: Cannot enable WPA2 Enterprise using arduino libraries

daniel-dsouza, 2018-08-09:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?Adafruit Feather ESP32?
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE and Platform.io
Flash Frequency:					?80Mhz?
Upload Speed:						?921600?

### Description:
Fails on `esp_wifi_sta_wpa2_ent_enable(&config);`

### Sketch:
```cpp
#include "esp_wpa2.h"
#include <WiFi.h>

const char* ssid = "network"; // your ssid
#define EAP_USERNAME "usrname"
#define EAP_PASSWORD "hunter123"

void setup() {
    Serial.begin(115200);
    delay(10);

    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    
    // WPA2 enterprise magic starts here
    WiFi.disconnect(true);      
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));

    
    esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
    esp_wifi_sta_wpa2_ent_enable(&config);
    // WPA2 enterprise magic ends here

    WiFi.begin(ssid);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e6f51  PS      : 0x00060530  A0      : 0x800d1ca6  A1      : 0x3ffb1f40  
A2      : 0x3ffb1f74  A3      : 0x3ffc5890  A4      : 0x00000000  A5      : 0x526312a6  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1f40  
A10     : 0x3ffb23cc  A11     : 0x3f4010a5  A12     : 0x0000000a  A13     : 0x3ffb23d6  
A14     : 0x00000079  A15     : 0x00000033  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x400e6f51:0x3ffb1f40 0x400d1ca3:0x3ffb1f70 0x4012ad1f:0x3ffb1fa0

PC: 0x400e6f51
EXCVADDR: 0x00000138

Decoding stack results
0x400d1ca3: setup() at /home/hunter/Documents/Arduino/sketch_aug09a/sketch_aug09a.ino line 24
0x4012ad1f: loopTask(void*) at /home/hunter/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp line 15
```

devyte, 2018-08-09:
Correct, there are issues with wpa2_enterprise in the espressif sdk. This is covered elsewhere.
Closing.

stickbreaker, 2018-08-09:
This code is invalid.
```c++
#define EAP_USERNAME "usrname"
#define EAP_PASSWORD "hunter123"

    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));

```
The define statement is just a substitution:
```c++
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)"usrname", strlen("usrname"));
```
with this type declaration you are telling the compiler to treat the first four letters of "usrn" as the value of a pointer to a cstring of characters.

the correct way would be to define a variable:
```c++
const char* eap_username = "usrname";
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)eap_usernam, strlen(eap_username));
```

Chuck.

daniel-dsouza, 2018-08-09:
@stickbreaker 

Thanks for finding that bug. However the same error persists.

I am trying to follow the sample[WPA2 enterprise sample](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEnterprise/WiFiClientEnterprise.ino)

stickbreaker, 2018-08-09:
@daniel-dsouza 
>Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

`LoadProhibited` is telling you that an invalid memory address was access (READ).  Usually this means that some pointer is loaded with an invalid value,  

```c++
esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
  esp_wifi_sta_wpa2_ent_enable(&config);
```

This is how `WPA2_CONFIG_INIT_DEFAULT()` is defined in `esp32_wpa2.h`
```c++

#include "esp_err.h"
#include "esp_wifi_crypto_types.h"

#ifdef __cplusplus
extern "C" {
#endif

extern const wpa2_crypto_funcs_t g_wifi_default_wpa2_crypto_funcs;

typedef struct {
    const wpa2_crypto_funcs_t *crypto_funcs;
}esp_wpa2_config_t;

#define WPA2_CONFIG_INIT_DEFAULT() { \
    .crypto_funcs = &g_wifi_default_wpa2_crypto_funcs \
}
```

It looks to me that
```
extern const wpa2_crypto_funcs_t g_wifi_default_wpa2_crypto_funcs;
```
May not have been initialized before you use it.  It looks too me that you have skipped a few steps in the initialization sequence.  I have no knowledge how to correctly use WPA2, so I can't be of any more help, Sorry.
 
Chuck.

vitcon-iot, 2018-08-10:
WiFi.mode() is now required in order to init the hardware.

espressif/arduino-esp32#1618
https://github.com/espressif/arduino-esp32/issues/1618#issuecomment-403844871

closed 2019-08-15 by stale[bot]